m-labs/artiq #340: artiq_flash fails, missing board/pipistrello.cfg

jboulder, 2016-03-20:
Flashing pipistrello on Ubuntu 14.04 following [Installing ARTIQ](https://m-labs.hk/artiq/manual/installing.html) instructions. First problem is openocd is neither part of conda deployment nor is user instructed where to obtain openocd.

```$ artiq_flash -t pipistrello -d ~/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/binaries/pipistrello-qc1
Design name: b'top.ncd;UserID=0xFFFFFFFF\x00'
Partname b'6slx45csg324\x00'
Date b'2016/03/20\x00'
Time b'16:44:06\x00'
found binary data length: 1274692
Traceback (most recent call last):
  File "/home/britton/miniconda3/envs/pip2/bin/artiq_flash", line 6, in <module>
    sys.exit(artiq.frontend.artiq_flash.main())
  File "/home/britton/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/frontend/artiq_flash.py", line 125, in main
    "-c", "; ".join(prog),
  File "/home/britton/miniconda3/envs/pip2/lib/python3.5/subprocess.py", line 579, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/home/britton/miniconda3/envs/pip2/lib/python3.5/subprocess.py", line 560, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/home/britton/miniconda3/envs/pip2/lib/python3.5/subprocess.py", line 950, in __init__
    restore_signals, start_new_session)
  File "/home/britton/miniconda3/envs/pip2/lib/python3.5/subprocess.py", line 1544, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'openocd'
```
Install openocd using apt-get. ``$ sudo apt-get install openocd``
Try to run artiq_flash.
```
$ artiq_flash -t pipistrello -d ~/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/binaries/pipistrello-qc1
Design name: b'top.ncd;UserID=0xFFFFFFFF\x00'
Partname b'6slx45csg324\x00'
Date b'2016/03/20\x00'
Time b'16:44:06\x00'
found binary data length: 1274692
Open On-Chip Debugger 0.9.0 (2015-09-02-10:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:60: Error: Can't find board/pipistrello.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
Traceback (most recent call last):
  File "/home/britton/miniconda3/envs/pip2/bin/artiq_flash", line 6, in <module>
    sys.exit(artiq.frontend.artiq_flash.main())
  File "/home/britton/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/frontend/artiq_flash.py", line 125, in main
    "-c", "; ".join(prog),
  File "/home/britton/miniconda3/envs/pip2/lib/python3.5/subprocess.py", line 584, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openocd', '-f', 'board/pipistrello.cfg', '-c', 'init; jtagspi_init 0 /home/britton/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/binaries/pipistrello-qc1/bscan_spi_xc6slx45.bit; jtagspi_program /tmp/tmpkkjg8yj5 0x0; jtagspi_program /home/britton/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/binaries/pipistrello-qc1/bios.bin 0x170000; jtagspi_program /home/britton/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/binaries/pipistrello-qc1/runtime.fbi 0x180000; xc6s_program xc6s.tap; exit']' returned non-zero exit status 1
```
Problem is that artiq_flash assumes location for .bit and .cfg files which is incompatible with the Getting Started documentation.

https://github.com/m-labs/artiq/blob/6dd1eb2e928476fdc4c364fae26b8bc14c0fea54/artiq/frontend/artiq_flash.py#L124

jordens, 2016-03-20:
You need openocd from the git repository. See the manual.

closed 2016-03-20 by jordens

jboulder, 2016-03-20:
Confirmed that building openocd from the git repository works fine. Please apply the following patch so users are not confused. 

```
From aabce5b1f5d6cd1f20c9a0b6db8fd60e2893f5bb Mon Sep 17 00:00:00 2001
From: Joe Britton <joe.britton@gmail.com>
Date: Sun, 20 Mar 2016 11:28:18 -0600
Subject: [PATCH] Fix typo. And in build of openocd on Ubuntu the aclocal
 dependency is provided by automake.

adduser requires root

conda installation requires user to build OpenOCD by hand.
---
 doc/manual/installing.rst | 38 +++++++++++++++++++++-----------------
 1 file changed, 21 insertions(+), 17 deletions(-)

diff --git a/doc/manual/installing.rst b/doc/manual/installing.rst
index 43f7540..452f40d 100644
--- a/doc/manual/installing.rst
+++ b/doc/manual/installing.rst
@@ -40,6 +40,27 @@ If your ``$PATH`` misses reference the ``miniconda3/bin`` or ``anaconda3/bin`` y
 
     $ export PATH=$HOME/miniconda3/bin:$PATH
 
+.. _install-openocd:
+
+Installing OpenOCD
+^^^^^^^^^^^^^^^^^^^
+The following instructions are for Ubuntu.
+
+* Install JTAG tools needed to program the Pipistrello and KC705:
+
+    ::
+
+        $ cd ~/artiq-dev
+        $ git clone https://github.com/ntfreak/openocd.git
+        $ cd openocd
+        $ sudo apt-get install build-essential libtool libusb-1.0-0-dev libftdi-dev automake
+        $ ./bootstrap
+        $ ./configure
+        $ make
+        $ sudo make install
+        $ sudo cp contrib/99-openocd.rules /etc/udev/rules.d
+        $ sudo adduser $USER plugdev
+
 Installing the ARTIQ packages
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 
@@ -189,23 +210,6 @@ These steps are required to generate gateware bitstream (``.bit``) files, build
 .. note::
     The options ``develop`` and ``--user`` are for setup.py to install Migen in ``~/.local/lib/python3.5``.
 
-.. _install-openocd:
-
-* Install JTAG tools needed to program the Pipistrello and KC705:
-
-    ::
-
-        $ cd ~/artiq-dev
-        $ git clone https://github.com/ntfreak/openocd.git
-        $ cd openocd
-        $ sudo apt-get install build-essentials libtool libusb-1.0-0-dev libftdi-dev
-        $ ./bootstrap
-        $ ./configure
-        $ make
-        $ sudo make install
-        $ sudo cp contrib/99-openocd.rules /etc/udev/rules.d
-        $ adduser $USER plugdev
-
 .. _install-flash-proxy:
 
 * Install the required flash proxy gateware bitstreams:
-- 
2.5.0
```

jordens, 2016-03-20:
Sounds good. Please see https://github.com/m-labs/artiq/blob/master/CONTRIBUTING.rst#contributing-code

jordens, 2016-03-20:
The formatting was wrong and the location you move the openocd section to is also bad. I redid the changes. Ok now?

jboulder, 2016-03-20:
Thanks for applying the patch Robert. I've rebuilt the sphinx docs and it
looks good from my end. -Joe