m-labs/artiq #341: pipistrello sometimes fails timing

jordens, 2016-03-20:
Timing failures are intermittent. Potentially triggered by congestion and large filling factor. Failure is usually in the sys_clk domain, in the CPU (IIRC).

It looks like par starts with a particular unlucky guess and then can't find its way out. Overriding the libc rng seed with LD_PRELOAD does not help. Same code, same options can lead to success and failure. Sometimes ten builds in a row succeed followed by five builds failing.

Do we require this being fixed for the release?

sbourdeauducq, 2016-03-21:
Turns out there are still some places in the Python part of the gateware that are still affected by this annoying Python iteration randomization, and the generated Verilog changes from one run to the next, which may in turn affect PAR. Does setting ``PYTHONHASHSEED`` help, as a workaround?

jordens, 2016-03-21:
With `-R` it seems to fail timing consistently now.

closed 2016-03-21 by jordens

mithro, 2016-03-21:
We have found similar problems when using ISE - See https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/209 and https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/72

commit 8f54a1e6 (2016-03-21): pipistrello: sys_clk 83 -> 75 MHz

This should close #341 once migen generates stable output.