m-labs/artiq #342: artiq_master -g not working

dleibrandt, 2016-03-22:
Using v1 prerelease candidate of artiq, installed via conda.  Trying to run `artiq master -g` causes a very long error, which starts with:

```
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:259:34: error: unknown type name ‘git_buf’
 static void _cffi_check__git_buf(git_buf *p)
                                  ^
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c: In function ‘_cffi_layout__git_buf’:
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:270:10: error: unknown type name ‘git_buf’
   struct _cffi_aligncheck { char x; git_buf y; };
          ^
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:272:12: error: ‘git_buf’ undeclared (first use in this function)
     sizeof(git_buf),
            ^
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:272:12: note: each undeclared identifier is reported only once for each function it appears in
In file included from /home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:3:0:
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:274:14: error: expected specifier-qualifier-list before ‘git_buf’
     offsetof(git_buf, ptr),
              ^
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:275:23: error: expected expression before ‘)’ token
     sizeof(((git_buf *)0)->ptr),
                       ^
In file included from /home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:3:0:
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:276:14: error: expected specifier-qualifier-list before ‘git_buf’
     offsetof(git_buf, asize),
              ^
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:277:23: error: expected expression before ‘)’ token
     sizeof(((git_buf *)0)->asize),
                       ^
In file included from /home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:3:0:
/home/rabi/anaconda3/envs/artiq-2016-03-22/lib/python3.5/site-packages/pygit2/__pycache__/pygit2_cffi_90de6ca3x2a59b7ee.c:278:14: error: expected specifier-qualifier-list before ‘git_buf’
     offsetof(git_buf, size),

```

sbourdeauducq, 2016-03-24:
@whitequark Seems the problem is the pygit2 package is attempting to use the (incompatible) system libgit2 instead of the libgit2 from conda.

sbourdeauducq, 2016-03-24:
This is easy to reproduce on the build server.

whitequark, 2016-03-24:
Why on earth is pygit trying to compile stuff when it is _imported_?!

jordens, 2016-03-26:
Probably another libgit2 in the mix. Either on your machine and getting noticed and built against by cffi on the fly or the one that I just uninstalled on the buildbot.

jordens, 2016-03-26:
Should be fixed. @dleibrandt, please `conda install pygit2` (or your usual artiq upgrade routine), verify that it installs 0.22.1-py35_3 and try again.
@whitequark , @sbourdeauducq I uninstalled libgit2-dev from lab. We might either have to dig deeper in the search order for headers and libraries and the recipe or keep it uninstalled.

jordens, 2016-03-26:
Yep. It's the local presence of a system `libgit2-dev` that is problematic. @dleibrandt could you also uninstall that package if it exists?

whitequark, 2016-03-26:
That's wrong, the conda package should ignore the local `libgit2-dev`. If it doesn't it is broken.

jordens, 2016-03-26:
Yes it is.

jordens, 2016-03-26:
Should be fixed with libgit2 and pygit2 0.24.0 now. Could not reproduce it anymore even with system libgit2-dev around.

Please complain here if it is still broken.

closed 2016-03-26 by jordens