m-labs/artiq #343: Running coredevice_examples/photon_histogram.py causes error

dleibrandt, 2016-03-22:
Error message is:

```
root:Terminating with exception (KeyError: "'core_dds'")
  File "/home/rabi/anaconda3/envs/art​iq-2016-03-22/lib/python3.5/si​te-packages/artiq/master/worke​r.py", line 217, in _handle_worker_requests
    data = func(*obj["args"], **obj["kwargs"])
  File "/home/rabi/anaconda3/envs/art​iq-2016-03-22/lib/python3.5/si​te-packages/artiq/master/datab​ases.py", line 27, in get
    return self.data.read[key]
KeyError: "'core_dds'"
```

sbourdeauducq, 2016-03-23:
Is there a ``core_dds`` entry in your ``device_db.pyon``? See the example provided in the ARTIQ repository.

dleibrandt, 2016-03-24:
You're right.  I hadn't updated my `device_db.pyon` since the dds objects changed.  This is working now.  Thanks.

closed 2016-03-24 by dleibrandt