m-labs/artiq #344: windows 64-bit installation

raghavendrasrinivas, 2016-03-23:
I wanted to test out Artiq 1.0 on our windows 64-bit system. It should, run with 64-bit Anaconda, right? I've tried installing it and get this error message when I try to create a new environment with artiq.

C:\Users\rabi>conda create --name artiq-2016-03-22 artiq-kc705-nist_qc2
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata: .......Error: Could not find URL: http://conda.anacon
da.org/m-labs/ch/win-64/

This is after a fresh install of Anaconda with python 3.5 and adding the m-labs channel.

whitequark, 2016-03-23:
>  http://conda.anaconda.org/m-labs/ch/win-64/

Shouldn't this be `/channel/`?

raghavendrasrinivas, 2016-03-23:
I added this channel, 

$ conda config --add channels http://conda.anaconda.org/m-labs/channel/main

so I'm not sure why it's defaulting to m-labs/ch

jordens, 2016-03-23:
Please check, paste, and edit your condarc.

raghavendrasrinivas, 2016-03-24:
Okay, using conda info somehow m-labs/ch was added to the channels list. I removed it and it installed fine. I had uninstalled the previous version of anaconda from this computer so I'm not sure if that's a remnant from the previous installation. 

Seems like it installed fine now, I'll continue with the testing. Thanks for the help and sorry for the trouble.

jordens, 2016-03-24:
np

closed 2016-03-24 by jordens