m-labs/artiq #345: set_dataset from kernel causes error

dleibrandt, 2016-03-24:
Running this experiment:
```
from artiq.experiment import *

class DatasetDemo(EnvExperiment):
    """Dataset demo"""

    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.set_dataset("test_dataset", 5, broadcast=True, save=False)
```

Causes this error:
```
root:Terminating with exception (RPCReturnValueError: type mismatch: cannot serialize <artiq.protocols.sync_struct.N​otifier object at 0x7f63ff6bc4a8> as None (<function HasEnvironment.set_dataset at 0x7f6404a61378> has returned <artiq.protocols.sync_struct.N​otifier object at 0x7f63ff6bc4a8>))
Core Device Traceback (most recent call last):
  File "/home/rabi/artiq-work/dataset​_demo.py", line 22, in artiq_worker_dataset_demo.Data​setDemo.run (RA=0x4b4)
    self.set_dataset("test_dataset​", 5, broadcast=True, save=False)
  File "<artiq>/coredevice/comm_gener​ic.py", line 389, in _send_rpc_value
    lambda: "None")
artiq.coredevice.comm_generic.​RPCReturnValueError(5): type mismatch: cannot serialize <artiq.protocols.sync_struct.N​otifier object at 0x7f63ff6bc4a8> as None (<function HasEnvironment.set_dataset at 0x7f6404a61378> has returned <artiq.protocols.sync_struct.N​otifier object at 0x7f63ff6bc4a8>)
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/art​iq-2016-03-22/lib/python3.5/si​te-packages/artiq/master/worke​r_impl.py", line 231, in main
    exp_inst.run()
  File "/home/rabi/anaconda3/envs/art​iq-2016-03-22/lib/python3.5/si​te-packages/artiq/language/cor​e.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rabi/anaconda3/envs/art​iq-2016-03-22/lib/python3.5/si​te-packages/artiq/coredevice/c​ore.py", line 108, in run
    self.comm.serve(object_map, symbolizer)
  File "/home/rabi/anaconda3/envs/art​iq-2016-03-22/lib/python3.5/si​te-packages/artiq/coredevice/c​omm_generic.py", line 527, in serve
    self._serve_exception(object_m​ap, symbolizer)
  File "/home/rabi/anaconda3/envs/art​iq-2016-03-22/lib/python3.5/si​te-packages/artiq/coredevice/c​omm_generic.py", line 519, in _serve_exception
    raise python_exn
artiq.coredevice.comm_generic.​RPCReturnValueError: type mismatch: cannot serialize <artiq.protocols.sync_struct.N​otifier object at 0x7f63ff6bc4a8> as None (<function HasEnvironment.set_dataset at 0x7f6404a61378> has returned <artiq.protocols.sync_struct.N​otifier object at 0x7f63ff6bc4a8>)
```

This only occurs when run is a kernel AND broadcast=True.

sbourdeauducq, 2016-03-25:
Another thing I guess is problematic is RPCing the ``Notifier`` mutations.

whitequark, 2016-03-27:
@sbourdeauducq This is a bug in set_dataset--it ought to always return None, or at least, always return the same type, in order to be a valid RPC.

sbourdeauducq, 2016-03-27:
Yes, I'll change it.

closed 2016-03-29 by sbourdeauducq

dleibrandt, 2016-03-29:
I was trying to try out this fix, and I ran into a problem.  I'm subscribed to only the main m-labs conda channel.  I created a new conda environment following the instructions in the manual today, but it seems that the git commit id of the artiq version that I got is 10 days old:

```
(artiq-2016-03-29)rabi@688101H116ARTIQ:~/artiq-work$ conda list artiq
# packages in environment at /home/rabi/anaconda3/envs/artiq-2016-03-29:
#
artiq                     1.0rc1          py_0+git1c9b8a1    http://conda.anaconda.org/m-labs/channel/main/noarch/artiq-1.0rc1-py_0+git1c9b8a1.tar.bz2
artiq-kc705-nist_clock    1.0rc1          py_0+git1c9b8a1    http://conda.anaconda.org/m-labs/channel/main/noarch/artiq-kc705-nist_clock-1.0rc1-py_0+git1c9b8a1.tar.bz2
asyncserial (/home/rabi/artiq-dev2/asyncserial) 0.1                       <pip>
llvmlite-artiq            0.5.1                   py35_19    http://conda.anaconda.org/m-labs/channel/main/linux-64/llvmlite-artiq-0.5.1-py35_19.tar.bz2
migen (/home/rabi/artiq-dev2/migen) 0.2                       <pip>
misoc (/home/rabi/artiq-dev2/misoc) 0.1                       <pip>
```

Should I be subscribed to the dev channel also?

sbourdeauducq, 2016-03-30:
The fix will be available in the next release (1.0rc2). The dev channel currently has a number of new and serious bugs.

commit 84d50c4c (2016-03-29): environment,worker_db: mutate datasets from experiments via dedicated method instead of Notifier. Closes #345
commit ac0f6262 (2016-03-29): environment,worker_db: mutate datasets from experiments via dedicated method instead of Notifier. Closes #345