m-labs/artiq #351: Consider enabling -ffast-math flags for ARTIQ Python

whitequark, 2016-03-28:
This is a fairly low hanging fruit; without `nnan` and `ninf`, for example, LLVM will not optimize multiplications by zero into constant zero.

whitequark, 2016-03-28:
The drawbacks are described in http://llvm.org/releases/3.5.0/docs/LangRef.html#fast-math-flags. There is an additional (separate) reassociate pass that performs algebraically equivalent transformations that can nevertheless affect precision.

Should we enable them? This gives a speedup of 180us to 100us on PulseRateDDS when it cannot hoist floating point instructions out of the loop.

@sbourdeauducq, @jordens ?

sbourdeauducq, 2016-03-28:
Fine IMO if clearly documented.

jordens, 2016-03-28:
This can't be enabled on a per function basis, right?
Otherwise LGTM. I have yet to encounter a case in my code where that would matter.

whitequark, 2016-03-28:
This can be enabled on a function basis but it needs yet another piece of syntax, and I don't really know what to use for it.

whitequark, 2016-03-28:
Well, as it happens I already invented the syntax for syscalls, so I made it enabled on per-function basis. I also made it disabled by default because it is a deviation from what Python normally does and e.g. getting different results in a portable function is no good.

sbourdeauducq, 2016-03-29:
No docs?

commit 1038f132 (2016-03-28): compiler: allow specifying per-function "fast-math" flags.

Fixes #351.
commit ee7e648c (2016-04-14): compiler: allow specifying per-function "fast-math" flags.

Fixes #351.
commit 00c0a306 (2016-04-26): doc: Document fast-math flag and kernel invariants.

Fixes #351, #359.
commit 24e24ddc (2016-04-26): doc: Document fast-math flag and kernel invariants.

Fixes #351, #359.
closed 2016-04-26 by whitequark