m-labs/artiq #353: Runtime doesn't seem to be initialising DDS'

r-srinivas, 2016-03-28:
From the description on the manual 

> init()
> 
>     Resets and initializes the DDS channel.
> 
>     The runtime does this for all channels upon core device startup.
> 

and the dds_test.py file in the examples, it seems like I should be able to set the DDS' frequency without initialising it in the experiment? However, I find that I still need a dds,init if not there's no output from just dds.set(). Should that be the case?

closed 2016-03-29 by sbourdeauducq

r-srinivas, 2016-03-29:
Okay, thanks. It definitely took more than 1 ms between initialising dds' though. 10 ms seemed to work for me.

sbourdeauducq, 2016-03-29:
Oh right, the runtime gives 1ms to the 9914 just for the DAC calibration. Somehow I thought that figure was 125us.

commit ce57794e (2016-03-29): doc: fix comment about when and how DDS init should be done. Closes #353
commit 74b71e5f (2016-03-29): doc: fix comment about when and how DDS init should be done. Closes #353