m-labs/artiq #354: RTIO Sequence error with core_dds.batch with dds.init

r-srinivas, 2016-03-28:
I get a RTIO sequence error with the following code,

`        with self.core_dds.batch:
            self.dds0.init()
            self.dds1.init()
            self.dds2.init()
            self.dds3.init()
            self.dds4.init()
            self.dds5.init()
            self.dds6.init()`

But if I do them one by one with a delay between them it's fine. Should one be able to initialise multiple dds' simultaneously?

whitequark, 2016-03-28:
It's not currently possible to initialize DDSes within a batch.

r-srinivas, 2016-03-28:
Why not? It seems like if you can write frequency tuning words to it simultaneously you should be able to write the bits to get it running as well.

sbourdeauducq, 2016-03-29:
I cannot see any reason for doing that other than "it can be done".

closed 2016-03-29 by sbourdeauducq

r-srinivas, 2016-03-29:
I got the GUI to crash by switching between the applets and the log tab on the right. Doing a keyboard-interrupt I get the following message.

```
[artiq-2016-03-24] C:\>artiq_gui
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 260, in _mkste
mp_inner
    fd = _os.open(file, flags, 0o600)
PermissionError: [Errno 13] Permission denied: 'C:\\tmpq51bd0lk'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\gui\state.py"
, line 76, in _do
    self.save()
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\gui\state.py"
, line 70, in save
    pyon.store_file(self.filename, data)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\protocols\pyo
n.py", line 197, in store_file
    with tempfile.NamedTemporaryFile("w", dir=directory, delete=False) as f:
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 549, in NamedT
emporaryFile
    (fd, name) = _mkstemp_inner(dir, prefix, suffix, flags, output_type)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 267, in _mkste
mp_inner
    _os.access(dir, _os.W_OK)):
KeyboardInterrupt

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 260, in _mkste
mp_inner
    fd = _os.open(file, flags, 0o600)
PermissionError: [Errno 13] Permission denied: 'C:\\tmp_i1lib_q'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\Quamash-0.5.4-py3.5
.egg\quamash\__init__.py", line 317, in upon_timeout
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\asyncio\events.py", line 125, in
_run
    self._callback(*self._args)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\asyncio\tasks.py", line 301, in _
wakeup
    self._step()
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\asyncio\tasks.py", line 239, in _
step
    result = coro.send(None)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\gui\state.py"
, line 78, in _do
    self.save()
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\gui\state.py"
, line 70, in save
    pyon.store_file(self.filename, data)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\protocols\pyo
n.py", line 197, in store_file
    with tempfile.NamedTemporaryFile("w", dir=directory, delete=False) as f:
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 549, in NamedT
emporaryFile
    (fd, name) = _mkstemp_inner(dir, prefix, suffix, flags, output_type)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 260, in _mkste
mp_inner
    fd = _os.open(file, flags, 0o600)
KeyboardInterrupt

[artiq-2016-03-24] C:\>
```