m-labs/artiq #355: Error message if DDS channels duplicated

r-srinivas, 2016-03-28:
I think it would be useful if an error message was generated if dds channels were duplicated. For example, I tried these entries in the device_db.pyon file, 

```python
    "dds0": {
        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "AD9914",
        "arguments": {"bus_channel": 58, "channel": 0},
        "comment": "Comments work in DDS panel as well"
    },
    "dds1": {
        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "AD9914",
        "arguments": {"bus_channel": 58, "channel": 0},
    },
```

And the experiment I had ran without any issues.

sbourdeauducq, 2016-03-29:
The more general feature should be a "device DB linter" that issues warnings for this and other potential issues (same problem with TTLs, no core device defined, etc.). The master and artiq_run could run the linter automatically after reading the DDB.

sbourdeauducq, 2016-03-29:
@r-srinivas since the experiment ran without issues, we can close #305? Was frequency monitoring also OK in the GUI?

jordens, 2016-03-29:
A linter can catch a few obvious errors but it will only be a small aid.

r-srinivas, 2016-04-01:
Frequency monitoring meaning? I could run DDS Setter on windows and it seemed to work fine with the one working DDS I have left.

sbourdeauducq, 2016-04-02:
The DDS panel that shows the current frequencies. Anyway, I tested it and it works.

r-srinivas, 2016-04-04:
That DDS panel isn't showing up for me on the Windows GUI, but I can see it on the virtual box.

sbourdeauducq, 2019-02-22:
Done for Kasli systems.

closed 2019-02-22 by sbourdeauducq

sbourdeauducq, 2019-02-22:
Templates are now on Hydra
e.g.
https://nixbld.m-labs.hk/eval/114?