m-labs/artiq #357: flterm not responding

r-srinivas, 2016-03-29:
I was trying to use flterm but it's basically not responding. 

`rabi@68810MAGTRAPVM:~/artiq-dev$ flterm /dev/ttyUSB2`

I've tried the other USB ports and power cycling and resetting the FPGA but I'm not getting any response from it.

whitequark, 2016-03-29:
Can you get any output using `cat /dev/ttyUSBN`?

r-srinivas, 2016-03-29:
I get this if I try that,

```
rabi@68810MAGTRAPVM:~/artiq-dev$ cat /dev/ttyUSB1
rabi@68810MAGTRAPVM:~/artiq-dev$ cat /dev/ttyUSB2
rabi@68810MAGTRAPVM:~/artiq-dev$ cat /dev/ttyUSB3
cat: /dev/ttyUSB3: No such file or directory
rabi@68810MAGTRAPVM:~/artiq-dev$ cat /dev/ttyUSB0
cat: /dev/ttyUSB0: Input/output error
rabi@68810MAGTRAPVM:~/artiq-dev$ 
```

whitequark, 2016-03-29:
Right, I forgot you can't do this to ttys using `cat`. Nevermind what I said.

sbourdeauducq, 2016-03-29:
Works fine here.

sbourdeauducq, 2016-03-29:
Are you sure this isn't just some VM USB bug? This happened regularly when I visited NIST. Try rebooting the VM.

r-srinivas, 2016-03-29:
I tried it briefly on a linux computer that we have and had the same problem but I'll look into it again tomorrow.

jordens, 2016-03-29:
Please first identify which serial device you want to use instead of guessing. E.g. a) use the one that appears if you plug in the relevant cable, b) use `ls /dev/serial/by-id`, c) use `dmesg` etc.
All version numbers (gateware, runtime, artiq) as well, please.

sbourdeauducq, 2016-03-30:
Can you also try with gtkterm and/or the C flterm?

jordens, 2016-03-30:
And also are you actually booting the device when you do this? Do the leds blink?

r-srinivas, 2016-03-30:
Okay, so it works now after I restarted the virtual machine again. Not sure why it didn't when I had restarted it before. Thanks.

closed 2016-03-31 by sbourdeauducq