m-labs/artiq #359: document kernel_constant_attributes

sbourdeauducq, 2016-03-29:


sbourdeauducq, 2016-03-29:
And I think that generally the manual could use a section about ARTIQ-Python. What is supported and what is not, what are the design principles, etc.

jordens, 2016-04-05:
Could we rename them to `kernel_invariants`? A bit shorter and that seemed to be how it was called for llvm. Also, can we have e.g. devices retrieved in build be added automatically to `kernel_invariants`? Those seem to be the heavy hitters.

whitequark, 2016-04-05:
> Could we rename them to kernel_invariants? A bit shorter

Sure if you'd like. I have neither opinion nor insight on how it should be named to make it convenient for ARTIQ users.

> and that seemed to be how it was called for llvm.

I can't see any reason this is relevant.

whitequark, 2016-04-05:
> Also, can we have e.g. devices retrieved in build be added automatically to kernel_invariants? Those seem to be the heavy hitters.

This is a bit hacky because kernel_invariants is strictly a *class* property (it has to be true for all instances to be useful) but `setattr_device` etc are executed on an *instance*.

sbourdeauducq, 2016-04-24:
Can't the compiler retrieve ``kernel_invariants`` from instances, and check that ``kernel_invariants`` is identical for all instances of a given class? This is already what the compiler does for the types of the host variables, isn't it?

whitequark, 2016-04-24:
What happens if they don't match? Silent performance degradation?

sbourdeauducq, 2016-04-24:
Yes with a warning, or an error.

whitequark, 2016-04-24:
Ok.

closed 2016-04-26 by whitequark