m-labs/artiq #360: GUI unstable in Windows

r-srinivas, 2016-03-29:
I've been trying to run some experiments on the GUI. It seems like it's not very stable and about 1 every 2 or 3 times I load the GUI and try running the experiment the GUI stops responding and I have to close the window.

I first load the master,

```
[artiq-2016-03-24] C:\Anaconda3\artiq_test>artiq_master
WARNING:worker(12,led.py):artiq.coredevice.comm_generic:Mismatch between gatewar
e (1.0rc1+18.g590bedb.dirty) and software (1.0rc1) versions
WARNING:worker(13,dds_test.py):artiq.coredevice.comm_generic:Mismatch between ga
teware (1.0rc1+18.g590bedb.dirty) and software (1.0rc1) versions
WARNING:worker(14,led.py):artiq.coredevice.comm_generic:Mismatch between gatewar
e (1.0rc1+18.g590bedb.dirty) and software (1.0rc1) versions
WARNING:worker(15,led.py):artiq.coredevice.comm_generic:Mismatch between gatewar
e (1.0rc1+18.g590bedb.dirty) and software (1.0rc1) versions
WARNING:worker(16,dds_test.py):artiq.coredevice.comm_generic:Mismatch between ga
teware (1.0rc1+18.g590bedb.dirty) and software (1.0rc1) versions
```

Then the gui in a separate command prompt.

`[artiq-2016-03-24] C:\>artiq_gui`

If it crashes I just load the gui again.

The experiments I'm running are,

dds_test.py

```python
from artiq.experiment import *

class DDS_Test(EnvExperiment):
    def build(self):
        self.setattr_device("pca")
        self.setattr_device("core")
        self.setattr_device("core_dds")
        for i in range(12):        
            self.setattr_device("dds%d" % i)
        

    @kernel

    def run(self):
        dt = 10*ms
        self.dds0.init()
        delay(dt)        
        self.dds1.init()
        delay(dt)        
        self.dds2.init()
        delay(dt)        
        self.dds3.init()
        delay(dt)               
        self.dds4.init()
        delay(dt)        
        self.dds5.init()
        delay(dt)        
        self.dds6.init()
        delay(dt)
        self.dds7.init()
        delay(dt)
        self.dds8.init()
        delay(dt)
        self.dds9.init()
        delay(dt)
        self.dds10.init()
        delay(dt)
        self.dds11.init()
        delay(dt)
        with self.core_dds.batch:
            self.dds0.set(100*MHz)
            self.dds1.set(200*MHz)
            self.dds2.set(300*MHz)
            self.dds3.set(400*MHz)
            self.dds4.set(500*MHz)
            self.dds5.set(600*MHz)
            self.dds6.set(700*MHz)
            self.dds7.set(800*MHz)
            self.dds8.set(900*MHz)
            self.dds9.set(1000*MHz)
            self.dds10.set(1100*MHz)
            self.dds11.set(1200*MHz)
```
led.py

```python

from artiq.experiment import *

class LED(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")

    @kernel
    def run(self):
        self.led.on()
        delay(1*s)
        self.led.off()
```

If I run these experiments using artiq_run I have no issues.

The versions I'm using are listed in that warning I got.

sbourdeauducq, 2016-03-29:
What did you do to crash it exactly? Did it print anything? This log trace from the master is not relevant here.

r-srinivas, 2016-03-29:
I just opened one of the experiment files, and clicked submit. I can see it in the queue, but then the gui stops responding and i see the (not responding) in the window. Screenshot looks like this.

http://s27.postimg.org/dz0s7ruw3/artiq_gui_crash.png

sbourdeauducq, 2016-03-29:
Can you press Ctrl-C on the terminal and see if you can get a backtrace that shows where it is stuck?

It works correctly here, so I cannot do anything otherwise.

r-srinivas, 2016-03-29:
I got the GUI to crash by switching between the applets and the log tab on the right. Doing a keyboard-interrupt I get the following message.

```
[artiq-2016-03-24] C:\>artiq_gui
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 260, in _mkste
mp_inner
    fd = _os.open(file, flags, 0o600)
PermissionError: [Errno 13] Permission denied: 'C:\\tmpq51bd0lk'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\gui\state.py"
, line 76, in _do
    self.save()
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\gui\state.py"
, line 70, in save
    pyon.store_file(self.filename, data)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\protocols\pyo
n.py", line 197, in store_file
    with tempfile.NamedTemporaryFile("w", dir=directory, delete=False) as f:
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 549, in NamedT
emporaryFile
    (fd, name) = _mkstemp_inner(dir, prefix, suffix, flags, output_type)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 267, in _mkste
mp_inner
    _os.access(dir, _os.W_OK)):
KeyboardInterrupt

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 260, in _mkste
mp_inner
    fd = _os.open(file, flags, 0o600)
PermissionError: [Errno 13] Permission denied: 'C:\\tmp_i1lib_q'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\Quamash-0.5.4-py3.5
.egg\quamash\__init__.py", line 317, in upon_timeout
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\asyncio\events.py", line 125, in
_run
    self._callback(*self._args)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\asyncio\tasks.py", line 301, in _
wakeup
    self._step()
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\asyncio\tasks.py", line 239, in _
step
    result = coro.send(None)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\gui\state.py"
, line 78, in _do
    self.save()
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\gui\state.py"
, line 70, in save
    pyon.store_file(self.filename, data)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\protocols\pyo
n.py", line 197, in store_file
    with tempfile.NamedTemporaryFile("w", dir=directory, delete=False) as f:
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 549, in NamedT
emporaryFile
    (fd, name) = _mkstemp_inner(dir, prefix, suffix, flags, output_type)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\tempfile.py", line 260, in _mkste
mp_inner
    fd = _os.open(file, flags, 0o600)
KeyboardInterrupt

[artiq-2016-03-24] C:\>
```

sbourdeauducq, 2016-03-29:
Thanks, now that's helpful. I guess you have some security policies that interfere with the way Python's ``tempfile.NamedTemporaryFile`` works when you ask it to create a file on the same filesystem as some target directory. The GUI does that regularly to update ``artiq_gui.pyon``. I guess you don't actually need to do anything to cause the crash: just run the GUI and it will crash on its own after exactly 30 seconds.

sbourdeauducq, 2016-03-29:
In fact, running the GUI in a directory where you have write permission (instead of C:\) may fix the problem.

r-srinivas, 2016-03-29:
Okay, that's a little strange because C: is the local directory and I think the user has admin rights there as far as I know. 

If I do the following,

`[artiq-2016-03-24] c:\Anaconda3\artiq_test>type NUL > 2.txt

[artiq-2016-03-24] c:\Anaconda3\artiq_test>`

It creates the blank text file.

sbourdeauducq, 2016-03-29:
That's not ``C:\``, that's ``C:\Anaconda3\artiq_test``.

Commit above should make the GUI print an error and continue instead of crashing.

r-srinivas, 2016-03-29:
Is there anyway to change the directory in which the gui creates the temporary file? You're right, I don't have access to C:\.

sbourdeauducq, 2016-03-29:
Try running the GUI from a folder where you have write permission. Unless Python messed up their tempfile creation function, that should fix it.

r-srinivas, 2016-03-29:
Okay, that seems to work, thanks!

closed 2016-03-29 by sbourdeauducq