m-labs/artiq #361: GUI can't start due to missing or faulty xcb plugin

r-srinivas, 2016-03-29:
I installed artiq using conda on the virtual machine we had in a new environment, from the main channel. I activated the environment and tried running the gui but got this error message,

```
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~/artiq_test$ artiq_gui
This application failed to start because it could not find or load the Qt platform plugin "xcb".

Available platform plugins are: linuxfb, minimal, offscreen, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)
```

I've tried deleting the environment and reinstalling it but got the same message. Any idea what's wrong?

sbourdeauducq, 2016-03-29:
Is your anaconda up to date?

r-srinivas, 2016-03-29:
I think so.

```
rabi@68810MAGTRAPVM:~$ conda update conda
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata: ......
# All requested packages already installed.
# packages in environment at /home/rabi/anaconda3:
#
conda                     4.0.5                    py35_0  
rabi@68810MAGTRAPVM:~$ 
```

jordens, 2016-03-29:
`conda info` and `conda list` please.

r-srinivas, 2016-03-29:
'''
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~$ conda info
Using Anaconda Cloud api site https://api.anaconda.org
Current conda install:

             platform : linux-64
        conda version : 4.0.5
  conda-build version : 1.19.2
       python version : 3.5.1.final.0
     requests version : 2.9.1
     root environment : /home/rabi/anaconda3  (writable)
  default environment : /home/rabi/anaconda3/envs/artiq-2016-03-29
     envs directories : /home/rabi/anaconda3/envs
        package cache : /home/rabi/anaconda3/pkgs
         channel URLs : http://conda.anaconda.org/m-labs/channel/main/linux-64/
                        http://conda.anaconda.org/m-labs/channel/main/noarch/
                        https://repo.continuum.io/pkgs/free/linux-64/
                        https://repo.continuum.io/pkgs/free/noarch/
                        https://repo.continuum.io/pkgs/pro/linux-64/
                        https://repo.continuum.io/pkgs/pro/noarch/
          config file : /home/rabi/.condarc
    is foreign system : False

(artiq-2016-03-29)rabi@68810MAGTRAPVM:~$ conda list
# packages in environment at /home/rabi/anaconda3/envs/artiq-2016-03-29:
#
aiohttp                   0.17.2                     py_3    http://conda.anaconda.org/m-labs/channel/main/noarch/aiohttp-0.17.2-py_3.tar.bz2
alabaster                 0.7.7                    py35_0    defaults
artiq (/home/rabi/artiq-dev/artiq) 0.1+1212.gfb8a548           <pip>
artiq                     1.0rc1          py_0+git1c9b8a1    http://conda.anaconda.org/m-labs/channel/dev/noarch/artiq-1.0rc1-py_0+git1c9b8a1.tar.bz2
artiq-kc705-nist_qc2      1.0rc1          py_0+git1c9b8a1    http://conda.anaconda.org/m-labs/channel/dev/noarch/artiq-kc705-nist_qc2-1.0rc1-py_0+git1c9b8a1.tar.bz2
asyncserial               0.1                       <pip>
babel                     2.2.0                    py35_0    defaults
binutils-or1k-linux       2.25.1                        0    http://conda.anaconda.org/m-labs/channel/dev/linux-64/binutils-or1k-linux-2.25.1-0.tar.bz2
cffi                      1.5.2                    py35_0    defaults
chardet                   2.3.0                    py35_0    defaults
dateutil                  2.4.2                      py_2    http://conda.anaconda.org/m-labs/channel/dev/noarch/dateutil-2.4.2-py_2.tar.bz2
docutils                  0.12                     py35_0    defaults
fontconfig                2.11.1                        5    defaults
freetype                  2.5.5                         0    defaults
h5py                      2.5.0               np110py35_4    defaults
hdf5                      1.8.15.1                      2    defaults
icu                       56.1                          0    http://conda.anaconda.org/m-labs/channel/main/linux-64/icu-56.1-0.tar.bz2
jinja2                    2.8                      py35_0    defaults
jpeg                      8d                            0    defaults
levenshtein               0.12.0                   py35_1    http://conda.anaconda.org/m-labs/channel/main/linux-64/levenshtein-0.12.0-py35_1.tar.bz2
libffi                    3.0.13                        0    defaults
libgfortran               3.0                           0    defaults
libgit2                   0.24.0                        1    http://conda.anaconda.org/m-labs/channel/main/linux-64/libgit2-0.24.0-1.tar.bz2
libpng                    1.6.17                        0    defaults
libssh2                   1.6.0                         2    http://conda.anaconda.org/m-labs/channel/main/linux-64/libssh2-1.6.0-2.tar.bz2
libxml2                   2.9.2                         0    defaults
lit                       0.4.1                     <pip>
llvmlite-artiq            0.5.1                   py35_19    http://conda.anaconda.org/m-labs/channel/main/linux-64/llvmlite-artiq-0.5.1-py35_19.tar.bz2
markupsafe                0.23                     py35_0    defaults
migen (/home/rabi/artiq-dev/migen) 0.2                       <pip>
misoc (/home/rabi/artiq-dev/misoc) 0.1                       <pip>
mkl                       11.3.1                        0    defaults
numpy                     1.10.4                   py35_1    defaults
openssl                   1.0.2g                        0    defaults
outputcheck               0.4.1                     <pip>
pip                       8.1.1                    py35_1    defaults
prettytable               0.7.2                      py_3    http://conda.anaconda.org/m-labs/channel/dev/noarch/prettytable-0.7.2-py_3.tar.bz2
pycparser                 2.14                     py35_0    defaults
pygit2                    0.24.0                   py35_1    http://conda.anaconda.org/m-labs/channel/main/linux-64/pygit2-0.24.0-py35_1.tar.bz2
pygments                  2.1.1                    py35_0    defaults
pyqt5                     5.5.1                    py35_0    http://conda.anaconda.org/m-labs/channel/main/linux-64/pyqt5-5.5.1-py35_0.tar.bz2
pyqtgraph                 0.9.10          py_112+git7d34e22    http://conda.anaconda.org/m-labs/channel/dev/noarch/pyqtgraph-0.9.10-py_112+git7d34e22.tar.bz2
pyserial                  2.7                      py35_0    defaults
python                    3.5.1                         0    defaults
python-dateutil           2.4.2                     <pip>
python-levenshtein        0.12.0                    <pip>
[... cut ...]
sphinx_rtd_theme          0.1.9                    py35_0    defaults
sqlite                    3.9.2                         0    defaults
system                    5.8                           2    defaults
tk                        8.5.18                        0    defaults
wheel                     0.29.0                   py35_0    defaults
xz                        5.0.5                         1    defaults
zlib                      1.2.8                         0    defaults
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~$ 
'''

jordens, 2016-03-29:
This may or may not be related. But please clean up your installation. Why are you trying to use artiq both from conda and from git?
What does `python -c 'import artiq;print(artiq.__artiq_dir__)'` show?

r-srinivas, 2016-03-29:
```
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~$ python -c 'import artiq;print(artiq.__artiq_dir__)'
/home/rabi/artiq-dev/artiq/artiq
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~$ 
```

We have both the development and release version on this computer.

jordens, 2016-03-29:
How do you think you achieved that?
Is `artiq (/home/rabi/artiq-dev/artiq) 0.1+1212.gfb8a548           <pip>` what you call "development"? How did you install that?
Please clean up your installation.

r-srinivas, 2016-03-29:
```
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~/artiq_test$ cat /home/rabi/anaconda3/envs/artiq-2016-03-29/bin/artiq_gui
#!/home/rabi/anaconda3/envs/artiq-2016-03-29/bin/python
if __name__ == '__main__':
    import sys
    import artiq.frontend.artiq_gui

    sys.exit(artiq.frontend.artiq_gui.main())
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~/artiq_test$ which artiq_gui/home/rabi/anaconda3/envs/artiq-2016-03-29/bin/artiq_gui
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~/artiq_test$ 
```
I'm not the only one using this computer, what's the best way of cleaning it up?

jordens, 2016-03-29:
The command is `which artiq_gui`.

jordens, 2016-03-29:
In a given environment you can only have one artiq "installed". Do both `conda remove artiq` and  `pip uninstall artiq` and then install exactly one again (`conda install artiq` or `python setup.py develop`/`pip install -e .`)

jordens, 2016-03-29:
And please `ldd /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/libqxcb.so`.
Then try installing (with `apt` or `aptitude` or `apt-get`) the packages that contain the `libxcb-*` components (named accordingly on Debian/Ubuntu) that could not be resolved in the ldd output. Alternatively you could install qt5 (`apt`) to pull in the missing libraries. There are also some `xcb` conda packages around but I don't know whether we should rely on them long term.

r-srinivas, 2016-03-29:
```
(artiq-2016-03-29)rabi@68810MAGTRAPVM:~$ which artiq_gui
/home/rabi/anaconda3/envs/artiq-2016-03-29/bin/artiq_gui
```
Could I just create a new environment and install artiq from conda there? That should work, right?

r-srinivas, 2016-03-30:
```
(artiq-2016-03-30)rabi@68810MAGTRAPVM:~$ ldd /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/libqxcb.so
	linux-vdso.so.1 =>  (0x00007ffc70d67000)
	libQt5XcbQpa.so.5 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5 (0x00007faf2bbee000)
	libQt5Gui.so.5 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../../libQt5Gui.so.5 (0x00007faf2b61e000)
	libQt5Core.so.5 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../../libQt5Core.so.5 (0x00007faf2b077000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007faf2ad73000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007faf2a9ae000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007faf2a679000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007faf2a477000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007faf2a267000)
	libxcb-render-util.so.0 => not found
	libxcb-render.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007faf2a05e000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007faf29e3f000)
	libxcb-image.so.0 => not found
	libxcb-icccm.so.4 => not found
	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007faf29c39000)
	libxcb-xfixes.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-xfixes.so.0 (0x00007faf29a32000)
	libxcb-shm.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007faf2982f000)
	libxcb-randr.so.0 => not found
	libxcb-shape.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-shape.so.0 (0x00007faf2962b000)
	libxcb-keysyms.so.1 => not found
	libxkbcommon.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007faf293f1000)
	libfontconfig.so.1 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../.././libfontconfig.so.1 (0x00007faf291b4000)
	libfreetype.so.6 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../.././libfreetype.so.6 (0x00007faf28f20000)
	libQt5DBus.so.5 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../.././libQt5DBus.so.5 (0x00007faf2bb53000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007faf28d02000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007faf289fc000)
	libpng16.so.16 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../.././libpng16.so.16 (0x00007faf287ba000)
	libz.so.1 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../.././libz.so.1 (0x00007faf285a4000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007faf2838e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007faf2818a000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007faf27f82000)
	/lib64/ld-linux-x86-64.so.2 (0x00007faf2bab4000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007faf27d70000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007faf27b6c000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007faf27966000)
	libxml2.so.2 => /home/rabi/anaconda3/envs/artiq-2016-03-29/lib/qt5/plugins/platforms/../../../././libxml2.so.2 (0x00007faf2760c000)
(artiq-2016-03-30)rabi@68810MAGTRAPVM:~$ 
```

Trying to install it I get the following message,

```
(artiq-2016-03-30)rabi@68810MAGTRAPVM:~$ sudo apt-get install libxcb-image.so.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libxcb-image.so.0
E: Couldn't find any package by regex 'libxcb-image.so.0'
(artiq-2016-03-30)rabi@68810MAGTRAPVM:~$ 
```

How should I type in the package name?

r-srinivas, 2016-03-30:
It seems like at some point `pip` was use to install something on the computer, and it automatically installs those files everytime a new environment is created. I'll look into it more tomorrow.

jordens, 2016-03-30:
You probably installed artiq in the conda root environment or in `~/.local/`.

You can use `aptitude` or `synaptic` or the web interface or `apt-file search` to find packages that provide a certain file. In this case it's `libxcb-image0`.

r-srinivas, 2016-03-30:
Okay, i installed the missing xcb packages and the gui loads now. Thanks!

jordens, 2016-03-30:
To be done: document what X libraries qt depends on or package them.

whitequark, 2016-03-31:
We can't package them specifically because of the xkbcommon dependency, which relies on distro-specific files.

jordens, 2016-03-31:
This is about `xcb` which does not depend on `xkbcommon` AFAICT.

whitequark, 2016-03-31:
It does. I don't remember the exact details but I tried building xcb before resorting to pulling it from the host.

commit d561681c (2016-03-31): doc: add note about xcb (closes #361)
commit 822c6545 (2016-03-31): doc: add note about xcb (closes #361)
closed 2016-03-31 by jordens