m-labs/artiq #363: question about @kernel

r-srinivas, 2016-03-30:
I ran into an issue where if I have a blank line after `@kernel` I get an error message with

```
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-03-24\Scripts\artiq_run-script.py", line 5,
 in <module>
    sys.exit(artiq.frontend.artiq_run.main())
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\frontend\arti
q_run.py", line 199, in main
    return run(with_file=True)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\frontend\arti
q_run.py", line 185, in run
    raise exn
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\frontend\arti
q_run.py", line 178, in run
    exp_inst.run()
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\language\core
.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\coredevice\co
re.py", line 99, in run
    object_map, kernel_library, symbolizer = self.compile(function, args, kwargs
, set_result)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\coredevice\co
re.py", line 79, in compile
    stitcher.stitch_call(function, args, kwargs, set_result)
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\compiler\embe
dding.py", line 478, in stitch_call
    synthesizer = self._synthesizer(self._function_loc(function.artiq_embedded.f
unction))
  File "C:\Anaconda3\envs\artiq-2016-03-24\lib\site-packages\artiq\compiler\embe
dding.py", line 611, in _function_loc
    column = re.search("def", source_line).start(0)
AttributeError: 'NoneType' object has no attribute 'start'
```

However, if I have a comment with 

```python
    @kernel
#   def
        
    def run(self):
```
That works fine for the rest of the script. I was just wondering why that was the case and if it was possible to get a more descriptive error message if there's a line between `@kernel` and the function proceeding it.

sbourdeauducq, 2016-03-30:
Reproduced.

whitequark, 2016-03-30:
No, not for 1.0. To solve this properly requires parsing the entire file, which means adding support for async/await to pythonparser.

sbourdeauducq, 2016-03-30:
How does fixing this requires parsing the entire file?

whitequark, 2016-03-30:
Because right now methods are extracted with a hack that doesn't recognize almost all variants of decorators. Python doesn't give you an API to get an entire span of the method, only the first line.

sbourdeauducq, 2016-03-30:
My compiler used ``inspect.getsource`` and ignored the decorators without any problems. Why did you change this?

whitequark, 2016-03-30:
I didn't.

whitequark, 2016-03-30:
Ah, nevermind, this is just a bug in pythonparser. Easy to fix.

sbourdeauducq, 2016-03-30:
Well ``inspect.getsource`` handles spaces after decorators just fine. So why does it break?

```
>>> print(inspect.getsource(test.Test.run))
    @kernel

    def run(self):
        self.core.break_realtime()

```

jordens, 2016-04-08:
@r-srinivas sidenote: proper coding style won't allow you to do that anyway.

whitequark, 2016-04-26:
Actually pythonparser is fine, this was just an embedding problem.

commit aa0882a0 (2016-04-26): embedding: ignore empty lines, like annotations, before kernel functions.

Fixes #363.
commit 9b04778f (2016-04-26): embedding: ignore empty lines, like annotations, before kernel functions.

Fixes #363.
closed 2016-04-26 by whitequark