m-labs/artiq #364: synchronising multiple AD9914s

r-srinivas, 2016-03-30:
We'd like to synchronise multiple AD9914s on our setup following the instructions on page 5 of [http://www.analog.com/media/en/technical-documentation/application-notes/AN-1254.pdf].

It can be done by setting the sync_out bit on all the DDS, not just the master. This basically involves writing to multiple DDS' and then providing an io_update that need not be synchronous. Would the best way of going about this be in modifying the runtime, or would it be better if there were some sort of dds_write and io_update function in dds.py that let the user do it separately? 

This seems to work on two dds' that I've programmed through SPI. The code is,

```python
dds.reset()

# step 3, dac calibration

dds.write(1, np.array([0x03, 0x01, 0x05, 0x21, 0x20]))
dds.io_update()
dds.write(2, np.array([0x03, 0x01, 0x05, 0x21, 0x20]))
dds.io_update()

# step 4, set sync_out on dds'

dds.write(1, np.array([0x01, 0x00, 0x89, 0x0B, 0x00]))
dds.write(2, np.array([0x01, 0x00, 0x89, 0x0B, 0x00]))
dds.io_update()

# step 5, set cal with sync_bit

dds.write(1, np.array([0x1B, 0x00, 0x00, 0x08, 0x40]))
dds.write(2, np.array([0x1B, 0x00, 0x00, 0x08, 0x40]))

# step 7, dac calibration

dds.write(1, np.array([0x03, 0x01, 0x05, 0x21, 0x20]))
dds.io_update()
dds.write(2, np.array([0x03, 0x01, 0x05, 0x21, 0x20]))
dds.io_update()
```

`dds.write(slave_select, array)`

jordens, 2016-03-30:
Is this ARTIQ-spi?
Boils down to the same. If you want to write arbitrary register values and are aware of all the pitfalls (make sure you stay clear of `dds.set()` etc) you need to expose that API. But `rtio_write()` should all you need. No need to change the runtime.
Are you suggesting you do that yourself or is this an issue report?

r-srinivas, 2016-03-30:
No, I had done this using the cheetah USB to SPI adapter we have. I guess this was more a feature request than an issue report.

dhslichter, 2016-03-30:
The code above is not ARTIQ-SPI, this is cheetah SPI code.  The discussions @sbourdeauducq and I have had on this topic boil down to:

- DDS syncing is something that should be part of the DDS init() routine, not exposed to the user as a separate set of methods.  It should be a one-time operation that happens transparently as part of DDS power-up and calibration.  Does not need to be done anew in experiments.

- The DDS syncing will also involve having a CSR that sets a phase delay for the RTIO clock relative to the external clock input to the FPGA, which comes from the SYNC_CLK of the first DDS.  The value of this phase delay should be calibrated to ensure that IO_UPDATE/FUD meets the setup timing criteria for exactly one rising SYNC_CLK edge at the DDs cards.

- DDS syncing does not need to be part of the 1.0 release if there is not enough time to implement it, but should be part of the 1.1 release and should be implemented in the dev channel in the meantime.

r-srinivas, 2016-03-31:
This is somewhat related to #115.

sbourdeauducq, 2016-03-31:
The solution with the current architecture is to add a boolean argument to ``dds_init`` that specifies whether DDS sync should be enabled or not.

And another API, which is not related to the DDS, to control RTIO clock phase, maybe. But I don't know if the hardware permits meeting FUD timing at all DDSes simultaneously, or if some complicated workaround such as changing the FUD timing depending on the selected channel would be necessary.

r-srinivas, 2016-03-31:
I'm not sure if the sync function can be placed in `dds_init`and done for the DDS' one by one. From the application note it seems like one would have to programme them simultaneously.

sbourdeauducq, 2016-03-31:
Why exactly?

r-srinivas, 2016-03-31:
Not sure, some AD black magic probably. I'll try to do it one by one at NIST and call AD tomorrow.

dhslichter, 2016-03-31:
We are planning to test in the very near term to determine the FUD delay introduced by the backplane and DDS loading.  This will inform us if there is a need to do some complex gyrations on FUD timing, but I would imagine it should be ok at 100 MHz RTIO clock.  We will advise of the result of the measurements.

r-srinivas, 2016-04-01:
Okay, I had a chat with AD. Looks like from that application note, step 1-7 can all be done on the DDSs one by one. The only thing that has to be simultaneous is the io-update after the frequency tuning word has been set as that's what gets the phase accumulator going. So what Sébastien suggested about having two options for dds.init() should work.

r-srinivas, 2016-04-01:
I tried testing it out on 2 DDS' we have. Seems like it will work. It seems to be more reliable if you start by initialising the master DDS whose sync_out signal is being distributed to the other DDS' instead of one of the slaves first.

sbourdeauducq, 2016-04-02:
I don't think you even need a simultaneous IO update. When programming DDSes at different times, the ARTIQ runtime should automatically apply the relevant phase correction.

r-srinivas, 2016-04-15:
I'm trying to modify the runtime so the `dds_init` will result in sycing. Aside from dds.c and dds.h, are there any other files I have to modify? Do the following changes make sense? I'm a little unsure about the timing of dds_init. For example, for the 9858, why is it `7*DURATION_WRITE`?

```diff
diff --git a/artiq/runtime/dds.c b/artiq/runtime/dds.c
index 1a7d8f9..779629c 100644
--- a/artiq/runtime/dds.c
+++ b/artiq/runtime/dds.c
@@ -18,7 +18,7 @@
 /* DAC calibration takes max. 1ms as per datasheet */
 #define DURATION_DAC_CAL (147000 << CONFIG_RTIO_FINE_TS_WIDTH)
 /* not counting final FUD */
-#define DURATION_INIT (8*DURATION_WRITE + DURATION_DAC_CAL)
+#define DURATION_INIT (13*DURATION_WRITE + 2*DURATION_DAC_CAL)
 #define DURATION_PROGRAM (6*DURATION_WRITE) /* not counting FUD */
 
 #else
@@ -68,13 +68,18 @@ void dds_init(long long int timestamp, int bus_channel, int 
 #endif
 
 #ifdef CONFIG_DDS_AD9914
+    DDS_WRITE(DDS_CFR4H, 0x0105); /* Enable DAC calibration */
+    DDS_WRITE(DDS_FUD, 0);
+    DDS_WRITE(DDS_CFR4H, 0x0005); /* Disable DAC calibration */
+    DDS_WRITE(DDS_FUD, 0);
     DDS_WRITE(DDS_CFR1H, 0x0000); /* Enable cosine output */
-    DDS_WRITE(DDS_CFR2L, 0x8900); /* Enable matched latency */
+    DDS_WRITE(DDS_CFR2L, 0x8b00); /* Enable matched latency and sync_out*/
     DDS_WRITE(DDS_CFR2H, 0x0080); /* Enable profile mode */
     DDS_WRITE(DDS_ASF, 0x0fff); /* Set amplitude to maximum */
+    DDS_WRITE(DDS_USR0, 0x0840); /* Set cal with sync */
     DDS_WRITE(DDS_CFR4H, 0x0105); /* Enable DAC calibration */
     DDS_WRITE(DDS_FUD, 0);
-    now += DURATION_DAC_CAL;
+    now += 2*DURATION_DAC_CAL;
     DDS_WRITE(DDS_CFR4H, 0x0005); /* Disable DAC calibration */
     DDS_WRITE(DDS_FUD, 0);
 #endif
diff --git a/artiq/runtime/dds.h b/artiq/runtime/dds.h
index 10fa3ff..0b61dd8 100644
--- a/artiq/runtime/dds.h
+++ b/artiq/runtime/dds.h
@@ -38,6 +38,7 @@
 #define DDS_ASF    0x33
 #define DDS_FUD    0x80
 #define DDS_GPIO   0x81
+#define DDS_USR0   0x6d
 #endif
```

sbourdeauducq, 2016-04-15:
> for the 9858, why is it 7*DURATION_WRITE?

Because for the init, there are 7 RTIO writes not counting the one for FUD:
```
    DDS_WRITE(DDS_GPIO, channel);
    DDS_WRITE(DDS_GPIO, channel | 1);
    DDS_WRITE(DDS_GPIO, channel);
    DDS_WRITE(DDS_CFR0, 0x78);
    DDS_WRITE(DDS_CFR1, 0x00);
    DDS_WRITE(DDS_CFR2, 0x00);
    DDS_WRITE(DDS_CFR3, 0x00);
```

r-srinivas, 2016-04-15:
So for the 9914 there are

```
DDS_WRITE(DDS_GPIO, channel);
DDS_WRITE(DDS_CFR1H, 0x0000); /* Enable cosine output */
DDS_WRITE(DDS_CFR2L, 0x8900); /* Enable matched latency */
DDS_WRITE(DDS_CFR2H, 0x0080); /* Enable profile mode */
DDS_WRITE(DDS_ASF, 0x0fff); /* Set amplitude to maximum */
DDS_WRITE(DDS_CFR4H, 0x0105); /* Enable DAC calibration */
DDS_WRITE(DDS_CFR4H, 0x0005); /* Disable DAC calibration */
```

Excluding the FUD, init is  8*DURATION_WRITE? What am I missing?

sbourdeauducq, 2016-04-15:
Nothing, it is 8.
https://github.com/m-labs/artiq/blob/master/artiq/runtime/dds.c#L21

r-srinivas, 2016-04-15:
So you include the FUD in between the DAC cals in that count but exclude the final FUD? So for the code I posted above it should be fine? There shouldn't be any other files I need to modify?

sbourdeauducq, 2016-04-15:
Yes. Looks good. I would make the sync optional (via a function argument, this also needs updating the driver). And keep ``DDS_USR0`` sorted.

r-srinivas, 2016-04-16:
Is there any other file I have to update? I somehow get an underflow error trying to run this experiment,

```Python
from artiq.experiment import *

class DDS0_Start(EnvExperiment):
    def build(self):
     
        self.setattr_device("core")
        self.setattr_device("core_dds")
        self.setattr_device("dds0")

    @kernel
    def run(self):
        self.dds0.init()
        delay(10*s)
        self.dds0.set(100*MHz)
```

```
ERROR:worker(3301,dds0_start.py):root:Terminating with exception (RTIOUnderflow: RTIO underflow at 117154209664 mu, channel 50, slack -189256 mu)
Core Device Traceback (most recent call last):
  File "/home/rabi/artiq-master/repository/dds0_start.py", line 20, in artiq_worker_dds0_start.DDS0_Start.run (RA=0x115c)
    self.dds0.init()
  File "<artiq>/coredevice/dds.py", line 140, in artiq.coredevice.dds._DDSGeneric.init (inlined)
    dds_init(now_mu(), self.bus_channel, self.channel)
  File "/home/rabi/artiq-dev/artiq/artiq/runtime/rtio.c", line 28, in rtio_process_exceptional_status
    timestamp, channel, timestamp - rtio_get_counter());
artiq.coredevice.exceptions.RTIOUnderflow(0): RTIO underflow at 117154209664 mu, channel 50, slack -189256 mu
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 231, in main
    exp_inst.run()
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/coredevice/core.py", line 108, in run
    self.comm.serve(object_map, symbolizer)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 527, in serve
    self._serve_exception(object_map, symbolizer)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 519, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.RTIOUnderflow: RTIO underflow at 117154209664 mu, channel 50, slack -189256 mu
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 3301
```

sbourdeauducq, 2016-04-17:
Try adding a small delay before the init.

r-srinivas, 2016-04-26:
That seemed to work, thanks. Looks like they seem to sync but not super deterministically. We may need to play around with the sync_delay. Would it be better to have this still under one `dds.init` function that takes two arguments, one whether to sync or not, and one with the delay or to have this as a separate function altogether, like `dds.init_sync`?

sbourdeauducq, 2016-04-26:
What is ``sync_delay``?

The reason you need to add a delay at the beginning is chiefly because you are doing another DAC calibration that takes a lot of time, so you need to budget for it.

r-srinivas, 2016-04-26:
Why don't you need to budget for it with just the normal init function? I guess I don't understand why the delay needs to be before you call init. 

Sorry, I meant the SYNC_IN delay ADJ as specified http://www.analog.com/media/en/technical-documentation/data-sheets/AD9915.pdf. I've tried to use identical SMA cables from the clock distribution we have for the SYNC_OUT signal but somehow it doesn't seem to be 100% deterministic on startup.

sbourdeauducq, 2016-04-26:
``dds_init`` is a point-in-time function (like SPI writes, DDS program, ...) which does not advance the timeline. It completes DDS initialization at ``now`` and therefore needs to perform operations before ``now``. You need a sufficient time buffer to perform all those operations, and with the long DAC calibration you have added, the default time buffer that you get upon entering a kernel is not enough.

r-srinivas, 2016-07-09:
I wrote a new function for syncing that takes the sync_delay as the argument. I rebuilt the bitstream, flashed the board and tested it. I can adjust the phases of the sync_clks so seems like it works.

```diff
diff --git a/artiq/coredevice/dds.py b/artiq/coredevice/dds.py
index fc99529..9332ac3 100644
--- a/artiq/coredevice/dds.py
+++ b/artiq/coredevice/dds.py
@@ -15,6 +15,11 @@ def dds_init(time_mu: TInt64, bus_channel: TInt32, channel: TInt32) -> TNone:
     raise NotImplementedError("syscall not simulated")
 
 @syscall(flags={"nowrite"})
+def dds_init_sync(time_mu: TInt64, bus_channel: TInt32,
+                  channel: TInt32, sync_delay: TInt32) -> TNone:
+    raise NotImplementedError("syscall not simulated")
+
+@syscall(flags={"nowrite"})
 def dds_set(time_mu: TInt64, bus_channel: TInt32, channel: TInt32, ftw: TInt32,
             pow: TInt32, phase_mode: TInt32, amplitude: TInt32) -> TNone:
     raise NotImplementedError("syscall not simulated")
@@ -152,6 +157,24 @@ class _DDSGeneric:
         dds_init(now_mu(), self.bus_channel, self.channel)
 
     @kernel
+    def init_sync(self, sync_delay=0):
+        """Resets and initializes the DDS channel as well as configures
+        the AD9914 DDS for synchronisation. The synchronisation procedure
+        follows the steps outlined in the AN-1254 application note.
+
+        This needs to be done for each DDS channel before it can be used, and
+        it is recommended to use the startup kernel for this.
+
+        This function cannot be used in a batch; the correct way of
+        initializing multiple DDS channels is to call this function
+        sequentially with a delay between the calls. 10ms provides a good
+        timing margin.
+
+        :param sync_delay: integer from 0 to 0x3f that sets value of
+        SYNC_OUT (bits 3-5) and SYNC_IN (bits 0-2) delay ADJ bits."""
+        dds_init_sync(now_mu(), self.bus_channel, self.channel, sync_delay)
+
+    @kernel
     def set_phase_mode(self, phase_mode):
         """Sets the phase mode of the DDS channel. Supported phase modes are:
```
```diff
diff --git a/artiq/runtime/dds.c b/artiq/runtime/dds.c
index 1a7d8f9..5c3574e 100644
--- a/artiq/runtime/dds.c
+++ b/artiq/runtime/dds.c
@@ -19,6 +19,7 @@
 #define DURATION_DAC_CAL (147000 << CONFIG_RTIO_FINE_TS_WIDTH)
 /* not counting final FUD */
 #define DURATION_INIT (8*DURATION_WRITE + DURATION_DAC_CAL)
+#define DURATION_INIT_SYNC (16*DURATION_WRITE + 2*DURATION_DAC_CAL)
 #define DURATION_PROGRAM (6*DURATION_WRITE) /* not counting FUD */
 
 #else
@@ -80,6 +81,40 @@ void dds_init(long long int timestamp, int bus_channel, int channel)
 #endif
 }
 
+void dds_init_sync(long long int timestamp, int bus_channel, int channel, int sync_delay)
+{
+    long long int now;
+
+    now = timestamp - DURATION_INIT_SYNC;
+
+#ifdef CONFIG_DDS_ONEHOT_SEL
+    channel = 1 << channel;
+#endif
+    channel <<= 1;
+    DDS_WRITE(DDS_GPIO, channel);
+
+#ifdef CONFIG_DDS_AD9914
+    DDS_WRITE(DDS_CFR4H, 0x0105); /* Enable DAC calibration */
+    DDS_WRITE(DDS_FUD, 0);
+    DDS_WRITE(DDS_CFR4H, 0x0005); /* Disable DAC calibration */
+    DDS_WRITE(DDS_FUD, 0);
+    DDS_WRITE(DDS_CFR2L, 0x8b00); /* Enable matched latency and sync_out*/
+    DDS_WRITE(DDS_FUD, 0);
+    /* Set cal with sync and set sync_out and sync_in delay */
+    DDS_WRITE(DDS_USR0, 0x0840 | (sync_delay & 0x3f));
+    DDS_WRITE(DDS_FUD, 0);
+    DDS_WRITE(DDS_CFR4H, 0x0105); /* Enable DAC calibration */
+    DDS_WRITE(DDS_FUD, 0);
+    now += 2*DURATION_DAC_CAL;
+    DDS_WRITE(DDS_CFR4H, 0x0005); /* Disable DAC calibration */
+    DDS_WRITE(DDS_FUD, 0);
+    DDS_WRITE(DDS_CFR1H, 0x0000); /* Enable cosine output */
+    DDS_WRITE(DDS_CFR2H, 0x0080); /* Enable profile mode */
+    DDS_WRITE(DDS_ASF, 0x0fff); /* Set amplitude to maximum */
+    DDS_WRITE(DDS_FUD, 0);
+#endif
+}
+
 /* Compensation to keep phase continuity when switching from absolute or tracking
  * to continuous phase mode. */
 static unsigned int continuous_phase_comp[CONFIG_RTIO_DDS_COUNT][CONFIG_DDS_CHANNELS_PER_BUS];
```
```diff
diff --git a/artiq/runtime/dds.h b/artiq/runtime/dds.h
index 10fa3ff..5b058c7 100644
--- a/artiq/runtime/dds.h
+++ b/artiq/runtime/dds.h
@@ -36,6 +36,7 @@
 #define DDS_FTWH   0x2f
 #define DDS_POW    0x31
 #define DDS_ASF    0x33
+#define DDS_USR0   0x6d
 #define DDS_FUD    0x80
 #define DDS_GPIO   0x81
 #endif
@@ -55,6 +56,8 @@ enum {
 };
 
 void dds_init(long long int timestamp, int bus_channel, int channel);
+void dds_init_sync(long long int timestamp, int bus_channel, int channel,
+    int sync_delay);
 void dds_batch_enter(long long int timestamp);
 void dds_batch_exit(void);
 void dds_set(long long int timestamp, int bus_channel, int channel,
```
```diff
diff --git a/artiq/runtime/ksupport.c b/artiq/runtime/ksupport.c
index 3fcedfb..1be23c2 100644
--- a/artiq/runtime/ksupport.c
+++ b/artiq/runtime/ksupport.c
@@ -118,6 +118,7 @@ static const struct symbol runtime_exports[] = {
     {"rtio_input_data", &rtio_input_data},
 
     {"dds_init", &dds_init},
+    {"dds_init_sync", &dds_init_sync},
     {"dds_batch_enter", &dds_batch_enter},
     {"dds_batch_exit", &dds_batch_exit},
     {"dds_set", &dds_set},
```

jordens, 2016-07-20:
Do you want this to be merged?
A patch to the mailing list or in a forked repository would be a lot easier to discuss.

r-srinivas, 2016-07-25:
I think it would be useful if it could be merged. Sébastien and I had discussed doing that briefly while he was here. Okay, I'll look into making a patch.

sbourdeauducq, 2016-08-15:
Done.

closed 2016-08-15 by sbourdeauducq