m-labs/artiq #365: Can't write to dds' using flterm

r-srinivas, 2016-03-31:
I'm trying to write to AD9914 registers using flterm, I've first initialised the DDS' using an experiment file.

I read `0x05` from DDS 0, which has been set to `0x89` after the `self.dds0.init()` from the experiment.

```
test> ddssel 0
test> ddsr 0x05
0x8900
```

I reset the dds, 

```
test> ddsreset
test> ddssel 0
test> ddsr 0x05
0x0900
```

And see that the register value has been reset to the default of `0x0900`.

now if I try to right to that register and FUD, it doesn't update.

```
est> ddssel 0
test> ddsw 0x05 0x8900
test> ddsfud
test> ddssel 0
test> ddsr 0x05
0x0900
```

Even init from flterm doesn't work.

```
test> ddssel 0
test> ddsinit
test> ddssel 0 
test> ddsr 0x05
0x0900
```

I tried setting the bus,

```
test> ddsbus 58
test> ddssel 0
test> ddsw 0x05 0x8900
test> ddsfud
test> ddssel 0
test> ddsr 0x05
0x0900
test> ddsinit 
test> ddssel 0
test> ddsr 0x05
0x0900
```
ddstest fails,

```
test> ddstest 1 0
readback fail on DDS 0, 0x00000000 != 0xaaaaaaaa
readback fail on DDS 0, 0x00000000 != 0x55555555
readback fail on DDS 0, 0x00000000 != 0xa5a5a5a5
readback fail on DDS 0, 0x00000000 != 0x5a5a5a5a
readback fail on DDS 0, 0x00000000 != 0xffffffff
readback fail on DDS 0, 0x00000000 != 0x12345678
readback fail on DDS 0, 0x00000000 != 0x87654321
readback fail on DDS 0, 0x00000000 != 0x0000ffff
readback fail on DDS 0, 0x00000000 != 0xffff0000
readback fail on DDS 0, 0x00000000 != 0x00ff00ff
readback fail on DDS 0, 0x00000000 != 0xff00ff00
```

Also, I think in the help file, 

```
ddstest <n> <c> - perform test sequence on DDS
```

Should be ddstest <c> <n>.

sbourdeauducq, 2016-03-31:
Toggling the reset pin, which is what ``ddsreset`` does, tickles various AD9914 silicon bugs (the runtime does not reset them for this reason). Are you sure this isn't what you are seeing? Do you still have problems if you power-cycle the DDS cards while making sure that reset is not toggled?

r-srinivas, 2016-03-31:
I tried turning off the DDS' and fpga, restarting them loading flterm and starting with `ddstest 1 0`. Still get this problem. 

Looking on the scope, ddsreset pulses the reset for about 500 ns (100 MHz clock signal to FPGA). ddsfud sends a pulse of about 5 ns. Does that sound right?

Looking at ddstest, there's a reset about 75 us before the ddsfud, with the bits being written in between.

r-srinivas, 2016-03-31:
Okay, looking at the write pin, which is pin 56 according to the schematic I have, all `ddsw` seems to do is pull it low for about 5 ns, regardless what I try to write.

r-srinivas, 2016-03-31:
Strangely, the dds on channel 3 can be written to but not completely. 

```
test> ddstest 1 0
readback fail on DDS 0, 0x00000000 != 0xaaaaaaaa
readback fail on DDS 0, 0x00000000 != 0x55555555
readback fail on DDS 0, 0x00000000 != 0xa5a5a5a5
readback fail on DDS 0, 0x00000000 != 0x5a5a5a5a
readback fail on DDS 0, 0x00000000 != 0xffffffff
readback fail on DDS 0, 0x00000000 != 0x12345678
readback fail on DDS 0, 0x00000000 != 0x87654321
readback fail on DDS 0, 0x00000000 != 0x0000ffff
readback fail on DDS 0, 0x00000000 != 0xffff0000
readback fail on DDS 0, 0x00000000 != 0x00ff00ff
readback fail on DDS 0, 0x00000000 != 0xff00ff00
test> ddstest 1 1
readback fail on DDS 1, 0x00000000 != 0xaaaaaaaa
readback fail on DDS 1, 0x00000000 != 0x55555555
readback fail on DDS 1, 0x00000000 != 0xa5a5a5a5
readback fail on DDS 1, 0x00000000 != 0x5a5a5a5a
readback fail on DDS 1, 0x00000000 != 0xffffffff
readback fail on DDS 1, 0x00000000 != 0x12345678
readback fail on DDS 1, 0x00000000 != 0x87654321
readback fail on DDS 1, 0x00000000 != 0x0000ffff
readback fail on DDS 1, 0x00000000 != 0xffff0000
readback fail on DDS 1, 0x00000000 != 0x00ff00ff
readback fail on DDS 1, 0x00000000 != 0xff00ff00
test> ddstest 1 2
readback fail on DDS 2, 0x00000000 != 0xaaaaaaaa
readback fail on DDS 2, 0x00000000 != 0x55555555
readback fail on DDS 2, 0x00000000 != 0xa5a5a5a5
readback fail on DDS 2, 0x00000000 != 0x5a5a5a5a
readback fail on DDS 2, 0x00000000 != 0xffffffff
readback fail on DDS 2, 0x00000000 != 0x12345678
readback fail on DDS 2, 0x00000000 != 0x87654321
readback fail on DDS 2, 0x00000000 != 0x0000ffff
readback fail on DDS 2, 0x00000000 != 0xffff0000
readback fail on DDS 2, 0x00000000 != 0x00ff00ff
readback fail on DDS 2, 0x00000000 != 0xff00ff00
test> ddstest 1 3
readback fail on DDS 3, 0xaaa8aaa8 != 0xaaaaaaaa
readback fail on DDS 3, 0x5a585a58 != 0x5a5a5a5a
readback fail on DDS 3, 0xfffdfffd != 0xffffffff
readback fail on DDS 3, 0x0000fffd != 0x0000ffff
readback fail on DDS 3, 0xfffd0000 != 0xffff0000
readback fail on DDS 3, 0x00fd00fd != 0x00ff00ff
test> 
```

dhslichter, 2016-03-31:
This is correct behavior according the datasheet for the AD9914; when you are doing a write, the "action" is all happening in the parallel address and data buses.

r-srinivas, 2016-03-31:
if I move dds3 to channel 5, it's behaviour is reproduced. 

```
test> ddstest 1 5
readback fail on DDS 5, 0xaaa8aaa8 != 0xaaaaaaaa
readback fail on DDS 5, 0x5a585a58 != 0x5a5a5a5a
readback fail on DDS 5, 0xfffdfffd != 0xffffffff
readback fail on DDS 5, 0x0000fffd != 0x0000ffff
readback fail on DDS 5, 0xfffd0000 != 0xffff0000
readback fail on DDS 5, 0x00fd00fd != 0x00ff00ff
test> 
```

r-srinivas, 2016-04-01:
Okay, looks like it's a hardware error. I power cycled them and replugged them in. I was using a test file that programmed the DDS' to output integer multiples of 100 MHz so I thought they were programming okay. Looks like some of the data lines aren't connected properly. 

```
test> ddstest 1 0
test> ddstest 1 1
readback fail on DDS 1, 0xaa88aa88 != 0xaaaaaaaa
readback fail on DDS 1, 0x55545554 != 0x55555555
readback fail on DDS 1, 0xa584a584 != 0xa5a5a5a5
readback fail on DDS 1, 0x5a585a58 != 0x5a5a5a5a
readback fail on DDS 1, 0xffdcffdc != 0xffffffff
readback fail on DDS 1, 0x12145658 != 0x12345678
readback fail on DDS 1, 0x87444300 != 0x87654321
readback fail on DDS 1, 0x0000ffdc != 0x0000ffff
readback fail on DDS 1, 0xffdc0000 != 0xffff0000
readback fail on DDS 1, 0x00dc00dc != 0x00ff00ff
test> ddstest 1 2
readback fail on DDS 2, 0x00000000 != 0xaaaaaaaa
readback fail on DDS 2, 0x00000000 != 0x55555555
readback fail on DDS 2, 0x00000000 != 0xa5a5a5a5
readback fail on DDS 2, 0x1a521a52 != 0x5a5a5a5a
readback fail on DDS 2, 0x00000000 != 0xffffffff
readback fail on DDS 2, 0x12301670 != 0x12345678
readback fail on DDS 2, 0x07600320 != 0x87654321
readback fail on DDS 2, 0x00003ff2 != 0x0000ffff
readback fail on DDS 2, 0x3ff20000 != 0xffff0000
readback fail on DDS 2, 0x3ff20000 != 0x00ff00ff
readback fail on DDS 2, 0x3f003f00 != 0xff00ff00
test> ddstest 1 5
readback fail on DDS 5, 0xaaa8aaa8 != 0xaaaaaaaa
readback fail on DDS 5, 0x5a585a58 != 0x5a5a5a5a
readback fail on DDS 5, 0xfffdfffd != 0xffffffff
readback fail on DDS 5, 0x0000fffd != 0x0000ffff
readback fail on DDS 5, 0xfffd0000 != 0xffff0000
readback fail on DDS 5, 0x00fd00fd != 0x00ff00ff
test> 
```

Sorry for the trouble.

dhslichter, 2016-04-01:
This looks to me like a single data bit is not connected or is broken on this chip.  f going to d and a going to 8 means that D1 is always low no matter what.  I would imagine that a broken address bit is responsible for the DDS where you can't read all the registers.

sbourdeauducq, 2016-04-01:
> Looking on the scope, ddsreset pulses the reset for about 500 ns

Again, don't reset AD9914s. They are buggy.

r-srinivas, 2016-04-05:
We had a look at the chips using a probe station and looks like some of the data lines aren't soldered properly. flterm was very useful in diagnosing these problems. From #185, flterm was removed from the conda packages. I'm wondering if it would be useful to have some sort of dds.test function in artiq that has the same functionality as the ddstest in flterm does.

sbourdeauducq, 2016-04-05:
flterm is still available in conda, but it is part of the misoc package.

closed 2016-04-05 by sbourdeauducq