m-labs/artiq #366: time.sleep should be RPC

sbourdeauducq, 2016-03-31:
```
repository/coredevice_examples/simple/blink_forever.py:18:17-18:27: error: cannot call this expression of type <instance builtins.builtin_function_or_method {
                __objectid__: int(width=32)
        }>
                time.sleep(0.3)
                ^^^^^^
```

sbourdeauducq, 2016-03-31:
Probably related to #260 ...

whitequark, 2016-03-31:
Yeah, sort of. Builtin functions don't work with `inspect`.

jordens, 2016-04-08:
`import sleep from time` does not help here. But wrapping does. OK to push to 2.0 as with #384?

```python
from artiq.experiment import *
import time

def sleep(t):
    time.sleep(t)        

class Foo(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        sleep(.1)
```

whitequark, 2016-04-08:
then we should do the same with #260.

jordens, 2016-04-08:
then no

whitequark, 2016-04-26:
The issue with modules tripping up the typechecker is tracked separately in https://github.com/m-labs/artiq/issues/408.

commit a88425b6 (2016-04-26): compiler: allow RPCing builtin functions.

Fixes #366.
commit 0f6f6846 (2016-04-26): compiler: allow RPCing builtin functions.

Fixes #366.
closed 2016-04-26 by whitequark