m-labs/artiq #367: reimplement ARTIQ in C/C++

dhslichter, 2016-04-01:
NIST IT security has just pushed a new policy that any and all software developed by outside contractors must be in C or C++ only.  Apparently they are going to run some sort of scanning software on the source to look for malicious code and this will only work on code written in C or C++.  All other programming languages are banned.  So we will need to reimplement the entirety of ARTIQ in C or C++.

whitequark, 2016-04-01:
I agree. This will make using Qt much easier.

sbourdeauducq, 2016-04-02:
It's already done because Oxford requested a MS-DOS port. Will release the sources soon.
[image]

closed 2016-04-02 by sbourdeauducq