m-labs/artiq #368: No way to initialize gui argument widget to a value from the dataset database

dleibrandt, 2016-04-01:
There are many parameters that we want to share between experiments, such as the frequencies and durations of various types of pulses.  As I understand the way artiq is intended to be used, we should put these parameters into the dataset database.

We'd also like to be able to modify the values of these parameters from the gui.  As far as I can tell, there is currently no way to create a gui argument widget and set it's value to be equal to that of an entry in the dataset database.  This means that for each parameter for which we have a gui argument widget, we will have to manually set the value every time we open the experiment, even if we just want to use the value that's already in the dataset database.  This is not practical.

What I'd like is to be able to do something like the following in the build() method:
```
self.setattr_argument("detection_pulse_t", NumberValue(self.get_dataset("detection_pulse_t", default=200*us)))
```
And then I want the detection_pulse_t widget to be set to the value from the dataset database when I open the experiment.

Is there currently a way to do this that I'm missing?  Or is there some sort of workaround?  If not, how can it be added?

sbourdeauducq, 2016-04-02:
Click the argument recompute button and it should work. One would not want to lose all the values set by the user in the GUI when the experiment is closed/opened, so they are not recomputed at that time.

dleibrandt, 2016-04-02:
Nope.  `self.get_dataset("detection_pulse_t", default=200*us)` returns `None` when you click the argument recompute button, but it returns the actual value from the dataset database when you click the submit button.

Perhaps there should be an example experiment demonstrating how to do this.

sbourdeauducq, 2016-04-02:
Will be fixed in 1.0rc3.

commit 774eccea (2016-04-02): master/worker_impl: use ParentDatasetDB in examine mode. Closes #368
commit b095c949 (2016-04-02): master/worker_impl: use ParentDatasetDB in examine mode. Closes #368
closed 2016-04-02 by sbourdeauducq