m-labs/artiq #371: Accessing file name in GUI

r-srinivas, 2016-04-04:
Is there a convenient way of accessing the file name in the GUI? I can see that the file name is listed if running the experiment. Is there any other way of finding it? It could be useful to have the file name displayed at the top of the experiment window, for example.

sbourdeauducq, 2016-04-05:
Experiment windows reference experiment names in the repository (=explorer), and so displaying filenames there is somewhat complicated as they depend on the contents of the repository (and would need to be updated when the repository is rescanned, etc.).

What about a tooltip in the explorer that shows the filename when the cursor is over an experiment name?

r-srinivas, 2016-04-05:
Like a mouseover? That would be handy. Basically someway of finding the filename on the GUI without actually running the experiment woudl be useful.

commit 2dfc1a8d (2016-04-05): gui/explorer: display filenames and classes in tooltips. Closes #371
closed 2016-04-05 by sbourdeauducq