m-labs/artiq #373: How to pass an argument to the build method of a class derived from HasEnvironment?

dleibrandt, 2016-04-04:
Is this possible?  Does anyone have example code?

sbourdeauducq, 2016-04-05:
Can you give more details? What are you trying to do in the end?

dleibrandt, 2016-04-05:
I have several types of laser pulses, which each require different values of the same arguments.

Example class is below.  I'd like `name` to be an argument, so that I can instantiate several of these in one `EnvExperiment`, with different names for the different types of pulses.

```
class DopplerLaserPulse(HasEnvironment):
    """397pi & 866 pulse"""
    
    def build(self):
        name = "Precool pulse"
        
        self.setattr_argument("t", NumberValue(1000*us, unit="us", scale=1e-6, min=0), name)
        self.setattr_argument("f_397pi", NumberValue(200*MHz, unit="MHz", scale=1e6, min=0), name)
        self.setattr_argument("a_397pi", NumberValue(4095, step=1, min=0, max=4095, ndecimals=0), name)
        self.setattr_argument("f_866", NumberValue(200*MHz, unit="MHz", scale=1e6, min=0), name)
        self.setattr_argument("a_866", NumberValue(4095, step=1, min=0, max=4095, ndecimals=0), name)
        
    @kernel
    def pulse(self):
        with self.core_dds.batch:
            self.dds_397pi.set(self.f_397pi)
            self.dds_866.set(self.f_866)
        with parallel:
            self.ttl_397pi.pulse(self.t)
            self.ttl_866.pulse(self.t)

```

sbourdeauducq, 2016-04-05:
In the top-level experiment:
```
self.pulse = DopplerLaserPulse(parent=self, name="pulse1")
```

In ``DopplerLaserPulse.build``:
```
name = self.get_argument("name")
```

Arguments passed as keywords when instantiating experiments fully override the regular mechanism that goes through the GUI etc.

sbourdeauducq, 2016-04-05:
There may be some conflict with the argument names though (t, f_397pi, etc.)

dleibrandt, 2016-04-05:
Thanks.  That's just what I was looking for.

You're right about about the argument name conflict.  Here is my hack solution:
```
class DopplerLaserPulse(HasEnvironment):
    """397pi & 866 pulse"""
    
    def build(self):
        self.name = self.get_argument("name")
        self.parent = self._HasEnvironment__parent
        
        self.setattr_argument(self.name + "_t", NumberValue(1000*us, unit="us", scale=1e-6, min=0), self.name)
        self.t = getattr(self, self.name + "_t")

        self.setattr_argument(self.name + "_f_397pi", NumberValue(200*MHz, unit="MHz", scale=1e6, min=0), self.name)
        self.f_397pi = getattr(self, self.name + "_f_397pi")
        
        self.setattr_argument(self.name + "_a_397pi", NumberValue(4095, step=1, min=0, max=4095, ndecimals=0), self.name)
        self.a_397pi = getattr(self, self.name + "_a_397pi")
        
        self.setattr_argument(self.name + "_f_866", NumberValue(200*MHz, unit="MHz", scale=1e6, min=0), self.name)
        self.f_866 = getattr(self, self.name + "_f_866")
        
        self.setattr_argument(self.name + "_a_866", NumberValue(4095, step=1, min=0, max=4095, ndecimals=0), self.name)
        self.a_866 = getattr(self, self.name + "_a_866")
        
    @kernel
    def pulse(self):
        with self.parent.core_dds.batch:
            self.parent.dds_397pi.set(self.f_397pi)
            self.parent.dds_866.set(self.f_866)
        with parallel:
            self.parent.ttl_397pi.pulse(self.t)
            self.parent.ttl_866.pulse(self.t)
```
Is there a better way to avoid the argument name conflict?

Also, what are your thoughts on my use of `self._HasEnvironment__parent` to get the devices that I need to make the pulses?  Is there a better way to do this?

sbourdeauducq, 2016-04-06:
> Is there a better way to avoid the argument name conflict?

Instead of ``setattr_argument``, use ``get_argument`` directly, e.g.:

```
self.a_397pi = self.get_argument(self.name + "_a_397pi" ...
```

Better than that - that's not developed at the moment, but would be nice to have.

> Also, what are your thoughts on my use of self._HasEnvironment__parent to get the devices that I need to make the pulses? Is there a better way to do this?

Requesting the same device twice is no problem (you get the previous driver on the second call), and ``setattr_device`` calls are automatically forwarded to the parent. So just use e.g. ``self.setattr_device("dds_397pi")`` in ``DopplerLaserPulse.build`` as if it were a standalone experiment. You should not use ``self._HasEnvironment__parent`` in your code.

dleibrandt, 2016-04-06:
Both of the above suggestions work.  Thanks.

closed 2016-04-06 by dleibrandt