m-labs/artiq #374: improve coredevice tracebacks

jordens, 2016-04-05:
Tracebacks and exceptions from within ARTIQ-Python or the runtime could be presented much better. The exception itself is printed twice (in between the two parts and at the end) and the frames around the exception servicing are at least confusing.

Even though tracebacks are very internal objects, there is a lot that could be done to improve this, e.g. https://github.com/pallets/jinja/blob/master/jinja2/debug.py. This may become cleaner and easier with the new `TracebackException`.

whitequark, 2016-04-05:
That's really broken. They're poking into the undocumented, unstable guts of the CPython runtime, and I'm not going to implement that gross hack. What we can do instead is to grab the traceback out of the CPython exception, merge it with ours, and then display the entire traceback ourselves.

jordens, 2016-04-06:
Yes. I wouldn't want to do what they do either. Parse the CPython exception with TracebackException and colleagues and then mix in our parts.