m-labs/artiq #375: Reload applet in last position

r-srinivas, 2016-04-05:
I've been playing around with the flopping simulation on windows and get it to plot the results on an applet while the simulation is running. That works great. However, if I close the applet and start it again by clicking enable, it reloads into its default position. It could be nice to have the applet reload into it's last position. 

What I'm imagining for example is having a graph maybe on another monitor. If accidentally close the applet and click enable again, it should reload on that other monitor instead of having it reload into it's default position on the GUI and then have to resize it and reposition it. 

Strangely if I click restart applet it seems to reload in its last position instead of starting from default again.

closed 2016-04-06 by sbourdeauducq

r-srinivas, 2016-04-26:
Is this fix in 1.0rc3? I'm running that on Windows with the following experiment.

```python
from artiq.experiment import *
import numpy as np

class IonMonitor2(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl1")
        self.setattr_device("ttl2")
        self.setattr_device("ttl6")
        self.setattr_device("scheduler")        
        self.points = 1
        self.counts = []

    def run(self):
        self.set_dataset("pmt_counts",
                         np.zeros((self.points,)),
                         broadcast=True, save=True)
        while True:
            self.core.comm.close()
            self.scheduler.pause()
            self.input = np.random.randint(0, 20, self.points)
            self.pulse()
            self.mutate_dataset("pmt_counts", 0, self.input[0])
            
    @kernel
    def pulse(self):
        self.core.break_realtime()
        for i in range(1):
            self.ttl2.pulse(10*ms)
            delay(10*ms)
```

I'm using the bignumber applet to display the PMT counts.

```{python} -m artiq.applets.big_number --embed {ipc_address} pmt_counts```

If I move the applet, then disable and enable it again it doesn't go back to its previous position. It loads to the default position. This is while the experiment, IonMonitor2 is running.If I terminate the experiment, move the applet, then disable and click enable again it loads to the default position.

sbourdeauducq, 2016-04-26:
No, it's in 2.0. The window embedding part of Qt (used for applets) is particularly full of bugs so I'm careful about touching it in 1.x.

commit 1d94cfd8 (2016-04-06): gui: save/restore geometry of applets when they are disabled/enabled. Closes #375