m-labs/artiq #376: Display experiment time/name and results of fit in applet

r-srinivas, 2016-04-05:
How do I go about getting the applet to display various data? For example, it would be useful if the time when the experiment ran or its RID was displayed in the applet title or somewhere. The results of a fit that is done is also useful to have. In the context of the flopping simulation, the resonant frequency f0 for example would be useful to display on the applet.

jordens, 2016-04-06:
You have to write that applet.

sbourdeauducq, 2016-04-06:
With the next commits:
```
$python -m artiq.applets.plot_xy --embed $ipc_address --title "Freq: {flopping_freq}" flopping_f_brightness
```
But yes, you should dig into the applet source and look into how to customize them. Not all options can be put into the command lines.

RID is another issue.

commit 356afb04 (2016-04-06): applets: support title for histogram and XY. Closes #376
closed 2016-04-06 by sbourdeauducq