m-labs/artiq #378: position and size of floating applets is not saved upon GUI restart

sbourdeauducq, 2016-04-06:
When an applet is floating, it gets back its default size upon GUI restart. When an applet is docked, Qt correctly restores its size (minus the bug described in #287).

Log docks (which are also dynamically created by the GUI) do not have this problem.

Commenting out ``self.resize`` in ``_AppletDock.__init__`` does not solve this problem.

``objectName`` appears to be correctly set.

The sequence of function calls to Qt for docking and restoring state appears to be identical between log and applet docks.

sbourdeauducq, 2016-04-06:
@jordens can you reproduce that? Might be another WM-related bug, as this seems to be assorted with a window focus issue.

sbourdeauducq, 2016-04-06:
Problem not present on Windows 7 (also tested with 1d94cfd8ed4684ecb10c3601659c8b53d8bcb482). Looks like another Qt bug.

jordens, 2016-04-06:
Works fine here. Floating applets are restored to position and size.

sbourdeauducq, 2016-09-19:
Can't reproduce this anymore (Qt 5.7). Closing for now.

closed 2016-09-19 by sbourdeauducq