m-labs/artiq #379: Memory leak in artiq_gui

cjbe, 2016-04-06:
Using Artiq 1.0rc1 on Ubuntu 15.10, with the examples/master device_db, and the kc705-clock gateware, artiq_gui leaks memory. 

Over 1000 minutes, the RSS linearly increased from 87 MB to 1500 MB.

sbourdeauducq, 2016-04-06:
Were you running something in particular or is the memory consumption increasing while nothing is happening?

cjbe, 2016-04-06:
I was not running anything. I just started the master, ctrlmgr, and gui, and monitored the memory usage

sbourdeauducq, 2016-04-06:
Were there any logging messages emitted by e.g. ctlmgr?

cjbe, 2016-04-06:
No logging messages were emitted

jordens, 2016-04-06:
Reproduced. Comes in chunks of about 6MB and valgrind doesn't see it. It's probably not strictly a "leak" then.

sbourdeauducq, 2016-04-06:
Python has some functions to print the state of the GC (number of live objects, etc.). What do they say if you call them regularly with a timer?

jordens, 2016-04-06:
http://mg.pov.lt/objgraph.py

jordens, 2016-04-06:
10 Qtimers, 10 functions, 10 tuples, and 30 cells per second.

jordens, 2016-04-06:
probably https://github.com/harvimt/quamash/issues/62

jordens, 2016-04-06:
was quamash.
@sbourdeauducq or @whitequark could you please copy the package over?

closed 2016-04-06 by jordens

sbourdeauducq, 2016-04-06:
Already there.

jordens, 2016-04-06:
There is a bunch of `Exception ignored in: Exception ignored in: ` pairs with the new quamash one for each applet. https://bugs.python.org/issue22836