m-labs/artiq #382: Square root in kernel terminates with error

dleibrandt, 2016-04-06:
The following code:
```
import numpy as np
from artiq.experiment import *

class TestKernelMath(EnvExperiment):
    """Test kernel math"""

    def build(self):
        self.setattr_device("core")
        self.setattr_argument("x", NumberValue(3.1))
        
        print("build:")
        print("x = ", self.x)
        print("x**0.5 = ", self.x**0.5)
        
    @kernel
    def run(self):
        print("kernel:")
        print("x = ")
        print(self.x)
        print("x**0.5 = ")
        print(self.x**0.5)
```

Produces this error:
```
root:Terminating with exception (OSError: Incorrect reply from device: _D2HMsgType.LOAD_FAILED (expected _D2HMsgType.LOAD_COMPLETED))
```

The error goes away if I comment out the last line.

(I'm running 1.0rc2.)

whitequark, 2016-04-06:
LLVM has transformed `pow(x, 0.5)` into `sqrt` and looks like we don't import that from libm.

whitequark, 2016-04-06:
We don't actually use libm in ksupport... we probably should enable LTO and simply link in libm.

commit bb064c67 (2016-04-06): runtime: add kernel-accessible sqrt.

Fixes #382.
closed 2016-04-06 by whitequark