m-labs/artiq #384: random number generation in kernel

dleibrandt, 2016-04-07:
I'd like to generate a random number in the kernel.  I'm trying the following code:
```
import numpy as np
import random

from artiq.experiment import *

class TestKernelMathRandom(EnvExperiment):
    """Test kernel math"""

    def build(self):
        self.setattr_device("core")
        
        print("build:")
        print(random.randint(0, 10))
        
    @kernel
    def run(self):
        print("kernel:")
        print(random.randint(0, 10))
```
It spits out an error when random is called in the kernel (but not in build):
```
root:Terminating with exception (AttributeError: type object 'module' has no attribute 'randint')
```

Of course, I could pre-generate all the random numbers I need in the build method, but that would be inconvenient.  Is there a way to do it in the kernel?

More generally, what all is available in the kernel?  Is this spelled out somewhere in the documentation?

sbourdeauducq, 2016-04-08:
The error probably has something to do with #366 and #260.

Note that this would not generate random numbers on the device, but RPC the ``randint`` function for execution on the host. If you want to actually generate on the device, e.g. for speed reasons, you have to implement your own PRNG that uses kernel or portable functions.

sbourdeauducq, 2016-04-08:
@dleibrandt actually, does the RPC work if you use ``randint`` like this:

```
from random import randint

...
    @kernel
    ...
        print(randint(0, 10))
```

whitequark, 2016-04-08:
Yes, modules are not currently supported since they do not fit well into the type system.

sbourdeauducq, 2016-04-08:
Can they be easily supported without integrating them into the type system (i.e. any module reference must be in the global namespace; putting modules in local variables, function arguments, etc. is not supported)

whitequark, 2016-04-08:
No, that's actually harder than doing it properly.

whitequark, 2016-04-08:
(Where by "properly" I mean faking an unique type for every instance of the module, instead of using the `type(module)` like it would normally happen.)

sbourdeauducq, 2016-04-08:
In this particular case, there will also be the problem of resolving the RPC at compile time.

whitequark, 2016-04-08:
I don't understand. It works if you import `randint` directly. It will also work if you do something like:
```python
class fakerandomcls:
  randint = random.randint
random = fakerandomcls()
```
which is what modules would effectively translate into under my propsal.

whitequark, 2016-04-08:
Ah wait, module functions don't receive `self`. That's more problematic then.

sbourdeauducq, 2016-04-08:
Okay, if the problem is understood and RPC works with the alternative import, we can defer this to 2.0 and say modules shall not be referenced in kernels.

dleibrandt, 2016-04-08:
@sbourdeauducq - Your alternative import statement also generates an error, but a different one:
```
root:Terminating with exception (RPCReturnValueError: type mismatch: cannot serialize 5 as None (<bound method Random.randint of <random.Random object at 0xa34c28>> has returned 5))
```

whitequark, 2016-04-09:
Oh yeah, I remember now why I didn't implement embedding modules in the first place--it is not very useful as most of the time you have to add annotations.

sbourdeauducq, 2016-04-09:
@dleibrandt 
```
def annotated_randint(...) -> TInt:
  return random.randint(...)
```

then use ``annotated_randint`` in kernels.

@whitequark can you add documentation for all these things?

sbourdeauducq, 2016-04-27:
Moved to #410 and #408.

closed 2016-04-27 by sbourdeauducq