m-labs/artiq #386: artiq_master -g crashes on windows

r-srinivas, 2016-04-08:
I was trying to go through the steps of git integration on windows. I've done the following,

- copied my repository folder into another folder called artiq_work
- made a new folder called repository
- initialised an empty repository there

```
rabi@68810NEWASSY MINGW64 /c/Anaconda3/artiq_test/repository
$ git init --bare
Initialized empty Git repository in C:/Anaconda3/artiq_test/repository/
```

I tried running `artiq_master -g` from ` /c/Anaconda3/artiq_test/` and I get an error message saying Python has stopped working.

sbourdeauducq, 2016-04-08:
Probably pygit2 crashing when interfacing with libgit.

sbourdeauducq, 2016-04-10:
This is sufficient to cause the crash:
```python
import pygit2
x = pygit2.Repository("some_repos")
```

sbourdeauducq, 2016-04-10:
The crash happens in ``msvcr120.dll`` called by ``git2.dll``.

whitequark, 2016-04-10:
What's the exact backtrace?

sbourdeauducq, 2016-04-10:
I don't know, visual studio DRM bullshit prevents me from seeing it.

whitequark, 2016-04-29:
To make VS work, set the date to Feb 1.

This is how it crashes:
[image]

Looks like a libgit2 bug...

sbourdeauducq, 2016-04-29:
I suppose enabling debugging symbols in libgit2 would give more info.

whitequark, 2016-05-03:
It was an ABI mismatch; libssh2, libgit2 and pygit2 were built with VC12 but at some point conda upgraded Python to VC14 (or something like this).

It is fixed now. Note that I ripped out openssl from libssh2 because openssl is built with old VC, no idea whether this materially affects ARTIQ.

closed 2016-05-03 by whitequark

jordens, 2016-05-03:
Other conda packages seem to tie down the msvc_runtime abi of their python.

whitequark, 2016-05-03:
I think it does something mostly right when Python is a dependency but in our case it isn't, and CMake needs explicit specification of the toolset.

jordens, 2016-05-03:
pygit2 should have python as a dependency, right?

whitequark, 2016-05-03:
I guess.