m-labs/artiq #387: Unable to set up git integration

r-srinivas, 2016-04-08:
I was trying to set up the git integration following the instructions on the manual but a `_pygit2.GitError: Reference 'refs/heads/master' not found` error. I've listed the commands I used below.

```
rabi@68810MAGTRAPVM:~/artiq-master$ source activate artiq-2016-04-04
discarding /home/rabi/anaconda3/bin from PATH
prepending /home/rabi/anaconda3/envs/artiq-2016-04-04/bin to PATH
(artiq-2016-04-04)rabi@68810MAGTRAPVM:~/artiq-master$ mv repository ~/artiq-work(artiq-2016-04-04)rabi@68810MAGTRAPVM:~/artiq-master$ mkdir repository
(artiq-2016-04-04)rabi@68810MAGTRAPVM:~/artiq-master$ cd repository
(artiq-2016-04-04)rabi@68810MAGTRAPVM:~/artiq-master/repository$ git init --bare
Initialized empty Git repository in /home/rabi/artiq-master/repository/
(artiq-2016-04-04)rabi@68810MAGTRAPVM:~/artiq-master/repository$ cd ~/artiq-master
(artiq-2016-04-04)rabi@68810MAGTRAPVM:~/artiq-master$ artiq_master -g
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-2016-04-04/bin/artiq_master", line 6, in <module>
    sys.exit(artiq.frontend.artiq_master.main())
  File "/home/rabi/anaconda3/envs/artiq-2016-04-04/lib/python3.5/site-packages/artiq/frontend/artiq_master.py", line 71, in main
    experiment_db = ExperimentDB(repo_backend, device_db.get_device_db)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-04/lib/python3.5/site-packages/artiq/master/experiments.py", line 84, in __init__
    self.cur_rev = self.repo_backend.get_head_rev()
  File "/home/rabi/anaconda3/envs/artiq-2016-04-04/lib/python3.5/site-packages/artiq/master/experiments.py", line 190, in get_head_rev
    return str(self.git.head.target)
_pygit2.GitError: Reference 'refs/heads/master' not found
(artiq-2016-04-04)rabi@68810MAGTRAPVM:~/artiq-master$ 
```

jordens, 2016-04-08:
Make a commit in the repository.

r-srinivas, 2016-04-08:
It seems like there's an issue with creating a bare repository as stated in the instructions. I can't make a commit either and get the message,

`fatal: This operation must be run in a work tree`

if I have a repository where I just do `git init` it seems okay.

jordens, 2016-04-08:
Sorry, More precisely: make a commit and push it to that repository.

sbourdeauducq, 2016-04-08:
If you're running everything on a single machine, you may also give the non-bare repository to ARTIQ and use a post-commit instead of a post-receive hook.

r-srinivas, 2016-04-08:
It may be obvious then but it seems like there's a missing step in the documentation. If I follow the instructions listed here, 

https://m-labs.hk/artiq/manual-release-1/getting_started_mgmt.html#setting-up-git-integration

I get the error, which can be resolved using your suggestions but I think the documentation should be updated to include those.

commit 7d2fa951 (2016-04-13): manual/git: commit needed before starting master, refresh. Closes #387
commit 317e6ea3 (2016-04-13): manual/git: commit needed before starting master, refresh. Closes #387
closed 2016-04-13 by sbourdeauducq