m-labs/artiq #388: GUI crashes on windows if I rename experiment that's open

r-srinivas, 2016-04-08:
I have this experiment here,

```python
from artiq.experiment import *

class DDS0_Start(EnvExperiment):
    def build(self):
     
        self.setattr_device("core")
        self.setattr_device("core_dds")
        self.setattr_device("dds0")

    @kernel
    def run(self):
        self.dds0.init()
        delay(10*ms)
        self.dds0.set(100*MHz)
```

I did the following,

-open experiment
-change experiment name from `DDS0_Start` to `DDS0_Start2`
-GUI crashes

There's some sort of traceback on the command prompt but it's hard to capture before it closes. 

This sequence also crashes the GUI,

-open experiment
-close experiment
-change experiment name from `DDS0_Start` to `DDS0_Start2`
-GUI crashes

The master is running fine in the meantime. This doesn't happen to me on the virtual machine.

sbourdeauducq, 2016-04-09:
If you run the GUI in a ``cmd`` window, not with ``start artiq_gui``, the traceback should stay. We do need tracebacks in most cases to debug something.

sbourdeauducq, 2016-04-09:
How did you change the name? Did you also run a repository re-scan?

r-srinivas, 2016-04-09:
I changed the name in the python script then scanned the repository through
the GUI.

r-srinivas, 2016-04-09:
Sorry, I should have specified that it's the scanning of the repository
that triggers the crash. I've been having the artiq_gui open in a cmd
window but it closed before I could see the traceback.

sbourdeauducq, 2016-04-09:
``cmd`` windows should not close when the application terminates. How exactly did you start the GUI?

r-srinivas, 2016-04-09:
I had a .cmd file that starts the gui.

Something like,

```
cd c:\anaconda3\artiq_test
activate artiq-2016-04-04
artiq_gui
```

I guess I should have just done it manually for the traceback. Maybe I cmd
/k at the end would help.

jordens, 2016-04-09:
Yes. The effect of that `.cmd` could be achieved much easier with just a shortcut.

sbourdeauducq, 2016-04-09:
@r-srinivas On an unrelated note, you should probably call ``dds.init`` once in the startup kernel and not in every experiment.

sbourdeauducq, 2016-04-10:
@r-srinivas I cannot reproduce the problem so please post your traceback.

r-srinivas, 2016-04-10:
So just to clarify, this is the sequence of events.

-double click on `DDS0_Start` in GUI to open experiment panel
-change name of experiment to `DDS0_Start2` in Spyder
-right click on experiment explorer and click `Scan repository HEAD`
-GUI Closes with the following traceback
```
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\gui\models.py
", line 258, in index
    parent_item.children_by_row[row])
IndexError: list index out of range
```

sbourdeauducq, 2016-04-11:
Are you sure opening the experiment is necessary? This should not touch any models.

sbourdeauducq, 2016-04-11:
Is your experiment at the root of the repository or in a subfolder? Can you post the full contents of your repository?

r-srinivas, 2016-04-11:
It's in a subfolder. For the file it's in repository/dds/dds0_start.py Yes, if I don't open the experiment then scanning the repository just updates the file name without crasing the gui.

sbourdeauducq, 2016-04-11:
Reproduced. The actual conditions are:

  1. the renamed experiment must be in a subfolder
  2. the renamed experiment must be selected in the explorer (opening is not necessary)

sbourdeauducq, 2016-04-11:
This is also independent of Windows/Linux.

r-srinivas, 2016-04-11:
Probably the reason why I couldn't reproduce it on the windows machine was because the experiment wasn't in a subfolder. I didn't think that would have been an issue.

sbourdeauducq, 2016-04-11:
Qt trees are very complicated.

commit 3134106a (2016-04-11): gui/models: handle Qt calling DictSyncTreeSepModel.index with garbage inputs. Closes #388
commit 93619003 (2016-04-11): gui/models: handle Qt calling DictSyncTreeSepModel.index with garbage inputs. Closes #388
closed 2016-04-11 by sbourdeauducq