m-labs/artiq #390: `self.scheduler.pause()` doesn't work on coredevice

r-srinivas, 2016-04-08:
I'm trying to test out the `pause` function to see if I can make it turn off an experiment at a particular state. I'm running the following experiment which I want to pause by running another at a higher priority

```python
from artiq.experiment import *

class DDS0_Start(EnvExperiment):
    def build(self):
     
        self.setattr_device("core")
        self.setattr_device("core_dds")
        self.setattr_device("dds0")
        self.setattr_device("scheduler")

    @kernel
    def run(self):
        self.dds0.init()
        delay(10*ms)
        self.dds0.set(200*MHz)
        delay(10*s)
        self.scheduler.pause()        
        self.dds0.set(100*MHz)
```

I get the following error,

```
ERROR:worker(1009,dds0_start.py):root:While compiling <repository>\dds\dds0_star
t.py
<artiq>\master\worker_impl.py:87:5: fatal: this function cannot be called as an
RPC
    def pause(self):
    ^
<repository>\dds\dds0_start.py:20:5: note: in system call here
    def run(self):
    ^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1009
```

It seems like I can't implement `self.scheduler.pause()` using @kernel? How should I write this instead?

sbourdeauducq, 2016-04-09:
Break the kernel in two parts and call them from a host function (run does not need to be a kernel) with pause in between.

sbourdeauducq, 2016-04-09:
And by the way - not only doesn't pause work on coredevice, but you also should disconnect from it before calling pause (otherwise, if another experiment preempts and connects to the device, you'll have problems). ``self.core.comm.close()``, reconnection is automatic with the next kernel.

closed 2016-04-09 by sbourdeauducq

whitequark, 2016-04-09:
Why doesn't `pause()` call `comm.close()`?

sbourdeauducq, 2016-04-12:
It now does in master/2.0.