m-labs/artiq #391: Experiment removed from schedule before it's over

r-srinivas, 2016-04-09:
I'm running the following experiment,

```python
from artiq.experiment import *

class DDS0_Scan_Test(EnvExperiment):
    def build(self):
     
        self.setattr_device("core")
        self.setattr_device("core_dds")
        self.setattr_device("dds0")

        self.setattr_argument("minimum", NumberValue(100, min=10, max=200))
        self.setattr_argument("maximum", NumberValue(200, min=100, max=500))
        self.setattr_argument("step_size", NumberValue(10, min=1, max=100))

    @kernel
    def run(self):
        self.dds0.init()
        delay(10*ms)
        for i in range(self.minimum, self.maximum, self.step_size):
            self.dds0.set(i*1e6)
            delay(1*s)
```

Looking at the signal on the scope, it seems to work. However, it removes itself from the scheduler before the experiment is complete, usually when the DDS is at 130 MHz. If I then send in another experiment at a lower priority,

```python
from artiq.experiment import *

class DDS0_Start(EnvExperiment):
    def build(self):
     
        self.setattr_device("core")
        self.setattr_device("core_dds")
        self.setattr_device("dds0")

    @kernel
    def run(self):
        self.dds0.init()
        delay(10*ms)
        self.dds0.set(100*MHz)
```

It basically interrupts the frequency ramp before it's done and sets the frequency back to 100 MHz. This was done on windows.

sbourdeauducq, 2016-04-09:
Yes, the first experiment puts all the commands into the RTIO FIFO and then returns. How should this be handled?

jordens, 2016-04-09:
What about an FAQ?

sbourdeauducq, 2016-04-09:
Or auto-synching of the core device upon disconnection? ``comm.close`` could run something like ``while core.get_rtio_counter() < now_mu(): pass``.

whitequark, 2016-04-09:
Auto-sync sounds good.

jordens, 2016-04-09:
Please don't do that. That breaks seamless handover.

jordens, 2016-04-09:
Raghu could just `sync()` on some channel, preferrably in the experiment that breaks others and needs the exclusive access, `DDS0_Start`.

sbourdeauducq, 2016-04-09:
There are some issues with seamless handover between different experiments (it works well only between kernels of the same experiment). When a new experiment starts it doesn't reset the RTIO timecounter (so time is not lost), but it does reset other parts of the RTIO core.

sbourdeauducq, 2016-04-09:
If there was no such reset, Raghu would have gotten a ``SequenceError`` upon startup of the second experiment.

jordens, 2016-04-09:
But wouldn't `while core.get_rtio_counter() < now_mu(): pass` prevent seamless handover (without dead time) between different experiment from ever working, even if the other issues are fixed?

sbourdeauducq, 2016-04-09:
Yes, this would have to be revisited (e.g. made optional) if/when cross-experiment smooth handover is implemented.

jordens, 2016-04-11:
It also eats up all the slack that you would have otherwise. I don't think this is the best answer to Raghu's problem. If you e.g. want to leave your cooling lasers on at the end of an experiment "for at least some time", this slack now becomes much longer and you reduce the rate of experiments.

sbourdeauducq, 2016-04-11:
We could look into seamless handover across experiments. The reason it's not done right now is that a rogue experiment could otherwise set ``now`` far in the future and break the core device until it is rebooted or a "fix-up" experiment that calls e.g. ``break_realtime`` is executed.

Maybe seamless handover can be automatically disabled (and the runtime falls back on the current resetting behavior) if the value of ``now`` at experiment startup is unreasonable (now - RTIO counter > some user-defined value).

sbourdeauducq, 2016-04-11:
The reset also clears the RTIO FIFOs, which solves the question of a rogue experiment clogging FIFOs with events years in the future.

jordens, 2016-04-11:
Ack. IMHO risking rogue experiments is accepable and we should not clear the FIFOs by default. A cleanup experiments that evicts these events by clearing the FIFOs and resets `now` would be desirable for me anyway. But yes, this is a bit of a steep learning curve for those not used to keeping the FIFOs and the difference in RTIO time and CPU time in mind.

jordens, 2016-05-05:
Let's handle inter-experiment seamless handover in another ticket and leave this one for documenting proper coding and interlocking of experiments like the pair above.

jordens, 2016-05-05:
... whether RTIO state is reset, FIFOs, pending IO events, now, rtio_counter etc.

sbourdeauducq, 2016-05-06:
and when the idle kernel is run (in 1.0: every time there is no active session with the control PC, and it is terminated by resetting the kernel CPU when the control PC connects).

sbourdeauducq, 2016-06-14:
Moving to 2.0:
 * let's get 1.1 out soon
 * this will be changed (inter-experiment seamless handover)
 * there are my slides in the meantime

jordens, 2016-08-24:
Covered in 69ea3d3cc0ade74d8e484cdeca45d6b6522a35e5

closed 2016-08-24 by jordens