m-labs/artiq #392: OSError when trying to build bitstream

r-srinivas, 2016-04-12:
I'm trying to run this command,

` python3.5 -m artiq.gateware.targets.kc705 -H qc1  # or qc2`

but get the following error,

```
Traceback (most recent call last):
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 391, in <module>
    main()
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 387, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/soc.py", line 61, in build_artiq_soc
    builder.build()
  File "/home/rabi/artiq-dev/misoc/misoc/integration/builder.py", line 156, in build
    run=self.compile_gateware, **kwargs)
  File "/home/rabi/artiq-dev/misoc/misoc/integration/soc_core.py", line 195, in build
    self.platform.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/platform.py", line 28, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/ise.py", line 188, in build
    self.map_opt, self.par_opt)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/ise.py", line 97, in _run_ise
    settings = common.settings(ise_path, ver, "ISE_DS")
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/common.py", line 20, in settings
    + path)
OSError: no version directory for Xilinx tools found in /opt/Xilinx
```

It used to work fine but I'm unable to generate the bitstream. It seems like the `/opt/Xilinx` folder exists and I've been able to generate bitstreams before. I've updated migen, misoc and artiq. Any idea what could be wrong?

cr1901, 2016-04-12:
This is probably my fault. Yesterday, I committed a patch to [unconditionally set the `source` argument](https://github.com/m-labs/migen/commit/c44664d294308dfba3ad3f5764c8b4c12f0d13e2) to the Xilinx ISE builder in Migen; building MiSoC bitstreams on Windows was broken without it.

I assume you're using a *nix; my patch shouldn't have broken the *nix version of the ISE builder. I have Xilinx and Linux installed on my netbook. I'll see if I can duplicate/fix later tonight.

r-srinivas, 2016-04-12:
I'm not sure if that was it. I tried checking out an older version of migen but that still gave me the same message. I'll try reinstalling my Xilinx stuff.

jordens, 2016-04-13:
Try the vivado toolchain Option.

sbourdeauducq, 2016-04-13:
@r-srinivas Did you install ISE or Vivado?

r-srinivas, 2016-04-13:
Vivado. It was working before for generating the bitstream so I'm not sure what changed. I guess it was maybe when I uninstalled artiq from #361.

sbourdeauducq, 2016-04-13:
@r-srinivas ``ls /opt/Xilinx`` please.

sbourdeauducq, 2016-04-13:
And yes, if you want to build with Vivado, you need to add ``--toolchain vivado``. ISE is the default, and if it is not installed, you get this error message that simply tells you it was not found.

r-srinivas, 2016-04-13:
```
rabi@68810MAGTRAPVM:/opt$ ls /opt/Xilinx
DocNav  Downloads  Vivado  Vivado_HLS  xic
```
This was after the reinstall. 

If I try `--toolchain vivado` I get,

```
Traceback (most recent call last):
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 391, in <module>
    main()
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 387, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/soc.py", line 61, in build_artiq_soc
    builder.build()
  File "/home/rabi/artiq-dev/misoc/misoc/integration/builder.py", line 156, in build
    run=self.compile_gateware, **kwargs)
  File "/home/rabi/artiq-dev/misoc/misoc/integration/soc_core.py", line 195, in build
    self.platform.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/platform.py", line 28, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/vivado.py", line 131, in build
    _run_vivado(build_name, toolchain_path, source)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/vivado.py", line 70, in _run_vivado
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

sbourdeauducq, 2016-04-13:
No error before the Python traceback?

r-srinivas, 2016-04-13:
```
rabi@68810MAGTRAPVM:~/artiq-dev$ python3.5 -m artiq.gateware.targets.kc705 -H nist_qc2 --toolchain vivado
make: Entering directory `/home/rabi/artiq-dev/misoc_nist_qc2_kc705/software/libcompiler_rt'
 CC       divsi3.o
 CC       modsi3.o
 CC       comparesf2.o
 CC       comparedf2.o
 CC       negsf2.o
 CC       negdf2.o
 CC       addsf3.o
 CC       subsf3.o
 CC       mulsf3.o
 CC       divsf3.o
 CC       lshrdi3.o
 CC       muldi3.o
 CC       divdi3.o
 CC       ashldi3.o
 CC       ashrdi3.o
 CC       udivmoddi4.o
 CC       floatsisf.o
 CC       floatunsisf.o
 CC       fixsfsi.o
 CC       fixdfdi.o
 CC       fixunssfsi.o
 CC       fixunsdfdi.o
 CC       adddf3.o
 CC       subdf3.o
 CC       muldf3.o
 CC       divdf3.o
 CC       floatsidf.o
 CC       floatunsidf.o
 CC       floatdidf.o
 CC       fixdfsi.o
 CC       fixunsdfsi.o
 CC       clzsi2.o
 CC       ctzsi2.o
 CC       udivdi3.o
 CC       umoddi3.o
 CC       moddi3.o
 CC       ucmpdi2.o
 CC       powidf2.o
 AR       libcompiler_rt.a
make: Leaving directory `/home/rabi/artiq-dev/misoc_nist_qc2_kc705/software/libcompiler_rt'
make: Entering directory `/home/rabi/artiq-dev/misoc_nist_qc2_kc705/software/libbase'
 CC       crt0-or1k.o
 CC       exception.o
 CC       libc.o
 CC       errno.o
 CC       crc16.o
 CC       crc32.o
 CC       console.o
 CC       system.o
 CC       id.o
 CC       uart.o
 CC       time.o
 CC       qsort.o
 CC       strtod.o
 CC       spiflash.o
 CC       vsnprintf.o
 AR       libbase.a
 CC       vsnprintf-nofloat.o
 AR       libbase-nofloat.a
make: Leaving directory `/home/rabi/artiq-dev/misoc_nist_qc2_kc705/software/libbase'
make: Entering directory `/home/rabi/artiq-dev/misoc_nist_qc2_kc705/software/libnet'
 CC       microudp.o
 CC       tftp.o
 AR       libnet.a
make: Leaving directory `/home/rabi/artiq-dev/misoc_nist_qc2_kc705/software/libnet'
make: Entering directory `/home/rabi/artiq-dev/misoc_nist_qc2_kc705/software/bios'
 CC       isr.o
[... 614 lines cut ...]
Traceback (most recent call last):
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 391, in <module>
    main()
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 387, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/soc.py", line 61, in build_artiq_soc
    builder.build()
  File "/home/rabi/artiq-dev/misoc/misoc/integration/builder.py", line 156, in build
    run=self.compile_gateware, **kwargs)
  File "/home/rabi/artiq-dev/misoc/misoc/integration/soc_core.py", line 195, in build
    self.platform.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/platform.py", line 28, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/vivado.py", line 131, in build
    _run_vivado(build_name, toolchain_path, source)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/vivado.py", line 70, in _run_vivado
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

There were some warnings before but nothing stopped it.

sbourdeauducq, 2016-04-13:
The relevant error is:
```
ERROR: [Synth 8-439] module 'top' not found
```

What does the generated Verilog (``misoc_nist_qc1_kc705/gateware/top.v``) look like? It should have ``module top(`` as second line.

r-srinivas, 2016-04-13:
```
/* Machine-generated using Migen */
module user_sma_clock_n(
	input serial_cts,
	input serial_rts,
	output reg serial_tx,
	input serial_rx,
	input clk200_p,
	input clk200_n,
	input cpu_reset,
	output [15:0] ddram_a,
	output [2:0] ddram_ba,
	output ddram_ras_n,
	output ddram_cas_n,
	output ddram_we_n,
	output ddram_cs_n,
	output [7:0] ddram_dm,
	inout [63:0] ddram_dq,
	output [7:0] ddram_dqs_p,
	output [7:0] ddram_dqs_n,
	output ddram_clk_p,
	output ddram_clk_n,
	output ddram_cke,
	output ddram_odt,
	output ddram_reset_n,
```

sbourdeauducq, 2016-04-13:
Uhm, there is obviously something wrong with ``module user_sma_clock_n(``. Did you change migen, misoc or artiq in any way, or is this the original code?

r-srinivas, 2016-04-13:
I had checked out an earlier version of `migen`. I tried with the latest version of migen and get this error,

```
Traceback (most recent call last):
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 411, in <module>
    main()
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 407, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/rabi/artiq-dev/artiq/artiq/gateware/soc.py", line 61, in build_artiq_soc
    builder.build()
  File "/home/rabi/artiq-dev/misoc/misoc/integration/builder.py", line 156, in build
    run=self.compile_gateware, **kwargs)
  File "/home/rabi/artiq-dev/misoc/misoc/integration/soc_core.py", line 195, in build
    self.platform.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/platform.py", line 28, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev/migen/migen/build/xilinx/vivado.py", line 124, in build
    named_sc, named_pc = platform.resolve_signals(v_output.ns)
  File "/home/rabi/artiq-dev/migen/migen/build/generic_platform.py", line 324, in resolve_signals
    sc = self.constraint_manager.get_sig_constraints()
  File "/home/rabi/artiq-dev/migen/migen/build/generic_platform.py", line 215, in get_sig_constraints
    pins = self.connector_manager.resolve_identifiers(pins)
  File "/home/rabi/artiq-dev/migen/migen/build/generic_platform.py", line 128, in resolve_identifiers
    r.append(self.connector_table[conn][pn])
KeyError: 'FMC'
```

sbourdeauducq, 2016-04-13:
Right, @dhslichter did not test his patch. Let me fix it...

sbourdeauducq, 2016-04-13:
In the meantime, QC1 and Clock should be unaffected by the KeyError.

sbourdeauducq, 2016-04-13:
KeyError with QC2 fixed in 212ee8ca35ec36800898cdb0390868a66c35ba51.

r-srinivas, 2016-04-13:
Looks like it worked for nist_qc1, thanks!

```
INFO: [Vivado 12-3199] DRC finished with 0 Errors, 42 Warnings
INFO: [Vivado 12-3200] Please refer to the DRC report (report_drc) for more information.
Loading data files...
Loading site data...
Loading route data...
Processing options...
Creating bitmap...
Creating bitstream...
Bitstream compression saved 63200768 bits.
Writing bitstream ./top.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-118] WebTalk data collection is enabled (User setting is ON. Install Setting is ON.).
INFO: [Common 17-186] '/home/rabi/artiq-dev/artiq/misoc_nist_qc1_kc705/gateware/usage_statistics_webtalk.xml' has been successfully sent to Xilinx on Wed Apr 13 11:12:48 2016. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2015.4/doc/webtalk_introduction.html.
INFO: [Common 17-83] Releasing license: Implementation
write_bitstream: Time (s): cpu = 00:01:30 ; elapsed = 00:01:29 . Memory (MB): peak = 2707.996 ; gain = 324.340 ; free physical = 164 ; free virtual = 5685
# write_cfgmem -force -format bin -interface spix4 -size 16 -loadbit "up 0x0 top.bit" -file top.bin
Creating config memory files...
Creating bitstream load up from address 0x00000000
Loading bitfile top.bit
Writing file ./top.bin
Writing log file ./top.prm
===================================
Configuration Memory information
===================================
File Format        BIN
Interface          SPIX4
Size               16M
Start Address      0x00000000
End Address        0x00FFFFFF

Addr1         Addr2         Date                    File(s)
0x00000000    0x003611D7    Apr 13 11:12:46 2016    top.bit
# quit
INFO: [Common 17-206] Exiting Vivado at Wed Apr 13 11:12:51 2016...
rabi@68810MAGTRAPVM:~/artiq-dev/artiq$ 
```

I'll try for qc2 now.

Just for the manual,

Could

> For KC705:
> 
> $ python3.5 -m artiq.gateware.targets.kc705 -H qc1  # or qc2

be changed to nist_qc1 or nist_qc2? 

And in the

>Preparing the core device FPGA board

section, could the `--toolchain vivavo` option be added in the instructions? Thanks!

sbourdeauducq, 2016-04-13:
Done.

closed 2016-04-13 by sbourdeauducq