m-labs/artiq #393: OSError: Connection Closed even though experiment runs fine

r-srinivas, 2016-04-14:
I'm running the following experiment,

```Python
from artiq.experiment import *

class LED(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")

    @kernel
    def run(self):
        self.led.on()
        delay(1*s)
        self.led.off()
        delay(1*s)
        self.led.on()
        delay(1*s)
        self.led.off()
```

And I can see the LED flashing twice but get the following error,

```
WARNING:worker(1146,led.py):artiq.coredevice.comm_generic:Mismatch between gatew
are (1.0rc1+204.gfaa268e) and software (1.0rc2) versions
ERROR:worker(1146,led.py):root:Terminating with exception (OSError: Connection c
losed)
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\master\worker
_impl.py", line 231, in main
    exp_inst.run()
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\language\core
.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
re.py", line 108, in run
    self.comm.serve(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 523, in serve
    self._read_header()
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 104, in _read_header
    (sync_byte, ) = struct.unpack("B", self.read(1))
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
mm_tcp.py", line 60, in read
    raise IOError("Connection closed")
OSError: Connection closed
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1146
```

I can change the experiment to just having the LED flash once and that works as well, so it seems like the experiment running but I'mnot sure what the error means.

sbourdeauducq, 2016-04-14:
Can you run artiq_corelog in the device DB folder after you have got this error?

Are you using master/2.x runtime/bitstream with 1.0rc2 compiler? This may cause problems.

r-srinivas, 2016-04-14:
Do you mean open a separate terminal and run `artiq_corelog`?

If so, I get,

Linux
```
ARNING:artiq.coredevice.comm_generic:Mismatch between gateware (1.0rc1+204.gfaa268e) and software (1.0rc2) versions
Startup RTIO clock: internal
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
send_rpc_value: unknown tag ffffffff
Failed to send RPC request (service 0, tag ����������������������������)
```

Windows
```
[artiq-2016-04-04] C:\Anaconda3\artiq_test>artiq_corelog
WARNING:artiq.coredevice.comm_generic:Mismatch between gateware (1.0rc1+204.gfaa
268e) and software (1.0rc2) versions
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-04-04\Scripts\artiq_corelog-script.py", lin
e 5, in <module>
    sys.exit(artiq.frontend.artiq_corelog.main())
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\frontend\arti
q_corelog.py", line 26, in main
    print(comm.get_log(), end="")
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\encodings\cp437.py", line 19, in
encode
    return codecs.charmap_encode(input,self.errors,encoding_map)[0]
UnicodeEncodeError: 'charmap' codec can't encode characters in position 1428-145
5: character maps to <undefined>
```

sbourdeauducq, 2016-04-14:
Do not mix major versions.

sbourdeauducq, 2016-04-14:
As for the other error, http://stackoverflow.com/questions/30539882/whats-the-deal-with-python-3-4-unicode-different-languages-and-windows/30551552

sbourdeauducq, 2016-04-14:
Reproduced when mixing 1.x/2.x, works find otherwise. Again don't do that.

closed 2016-04-14 by sbourdeauducq

r-srinivas, 2016-04-14:
Okay, thanks. I was generating the bitstream on the virtual device and trying to run it on Windows using the version in conda but I guess I should just install it from github.

sbourdeauducq, 2016-04-14:
There are two conda channels/labels, "dev" with 2.0.dev/master and "main" with 1.0rc2/release-1.

Why do you want to use the development version?