m-labs/artiq #394: Different error messages on windows and linux when DDS bus channel incorrect

r-srinivas, 2016-04-14:
My device_db contains DDS entries like,

```
    "dds0": {
        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "AD9914",
        "arguments": {"bus_channel": 58, "channel": 0},
        "comment": "Comments work in DDS panel as well"
    },
```

And I'm running the following experiment,

```Python
from artiq.experiment import *

class DDS_Set(EnvExperiment):
    def build(self):
     
        self.setattr_device("core")
        self.setattr_device("core_dds")
        self.setattr_device("dds0")
        self.setattr_device("dds2")
        self.setattr_device("dds3")
        self.setattr_device("dds4")
        self.setattr_device("dds5")

    @kernel
    def run(self):
        with self.core_dds.batch:
            self.dds0.set(100*MHz)
            self.dds2.set(200*MHz)
            self.dds3.set(200*MHz)
            self.dds4.set(200*MHz)
            self.dds5.set(200*MHz)
```

The DDS bus channel is actually 50, not 58 as it was previously after the recent changes. I built a new bitstream and flashed the FPGA. If I run the experiment with the incorrect bus channel on windows, I get the following error message,

```
WARNING:worker(1149,dds_set.py):artiq.coredevice.comm_generic:Mismatch between g
ateware (1.0rc1+204.gfaa268e) and software (1.0rc2) versions
ERROR:worker(1149,dds_set.py):root:Terminating with exception (UnicodeDecodeErro
r: 'utf-8' codec can't decode byte 0xdd in position 324: invalid continuation by
te)
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\master\worker
_impl.py", line 231, in main
    exp_inst.run()
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\language\core
.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
re.py", line 108, in run
    self.comm.serve(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 527, in serve
    self._serve_exception(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 508, in _serve_exception
    traceback = list(reversed(symbolizer(backtrace))) + \
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
re.py", line 89, in <lambda>
    lambda addresses: target.symbolize(library, addresses)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\compiler\targ
ets.py", line 166, in symbolize
    library=library) \
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\compiler\targ
ets.py", line 39, in __enter__
    tempfiles["__stdout__"] = stdout.decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xdd in position 324: invali
d continuation byte
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1149
```

On linux, I get a more explicit error message,

```
WARNING:worker(3084,dds_set.py):artiq.coredevice.comm_generic:Mismatch between gateware (1.0rc1+204.gfaa268e) and software (1.0rc2) versions
ERROR:worker(3084,dds_set.py):root:Terminating with exception (DDSError: Attempted to use invalid DDS bus)
Core Device Traceback (most recent call last):
  File "/home/rabi/artiq-master/repository/dds_set.py", line 27, in artiq_worker_dds_set.DDS_Set.run (RA=0x16e8)
    with self.core_dds.batch:
  File "<artiq>/coredevice/dds.py", line 42, in artiq.coredevice.dds._BatchContextManager.__exit__ (RA=0xed8)
    self.core_dds.dds_batch_exit()
  File "<artiq>/coredevice/dds.py", line 71, in artiq.coredevice.dds.CoreDDS.dds_batch_exit (inlined)
    dds_batch_exit()
  File "/home/rabi/artiq-dev/artiq/artiq/runtime/dds.c", line 97, in dds_set_one
    artiq_raise_from_c("DDSError", "Attempted to use invalid DDS bus", 0, 0, 0);
artiq.coredevice.exceptions.DDSError(0): Attempted to use invalid DDS bus
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-2016-04-13/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 231, in main
    exp_inst.run()
  File "/home/rabi/anaconda3/envs/artiq-2016-04-13/lib/python3.5/site-packages/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-13/lib/python3.5/site-packages/artiq/coredevice/core.py", line 108, in run
    self.comm.serve(object_map, symbolizer)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-13/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 527, in serve
    self._serve_exception(object_map, symbolizer)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-13/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 519, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.DDSError: Attempted to use invalid DDS bus
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 3084
```

If I correct the bus channel the experiment runs but I get the OSError on both systems as in #393.

sbourdeauducq, 2016-04-14:
If your bitstream is built from master, please use the dev packages for the software as well.

sbourdeauducq, 2016-04-14:
Or better, build the bitstream from release-1.

whitequark, 2016-04-14:
That probably won't help, that code didn't change since branching...

r-srinivas, 2016-04-14:
So I can just install artiq from `artiq-dev/artiq` with `python3.5 setup.py develop --user` from the manual? 

I tried that on the virtual machine and can get `artiq_master` running but `artiq_gui` gives me 

```
rabi@68810MAGTRAPVM:~/artiq-dev/artiq$ artiq_gui
Traceback (most recent call last):
  File "/home/rabi/.local/bin/artiq_gui", line 9, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_gui')()
  File "/home/rabi/anaconda3/lib/python3.5/site-packages/setuptools-20.2.2-py3.5.egg/pkg_resources/__init__.py", line 549, in load_entry_point
  File "/home/rabi/anaconda3/lib/python3.5/site-packages/setuptools-20.2.2-py3.5.egg/pkg_resources/__init__.py", line 2541, in load_entry_point
ImportError: Entry point ('console_scripts', 'artiq_gui') not found
```

r-srinivas, 2016-04-14:
`conda list`

```
rabi@68810MAGTRAPVM:~/artiq-dev/artiq$ conda list
# packages in environment at /home/rabi/anaconda3:
#
_license                  1.1                      py35_1    defaults
abstract-rendering        0.5.1               np110py35_0    defaults
aiohttp                   0.21.2                    <pip>
alabaster                 0.7.7                    py35_0    defaults
anaconda                  2.4.1               np110py35_0    defaults
anaconda-client           1.3.1                    py35_0    defaults
argcomplete               1.0.0                    py35_1    defaults
artiq (/home/rabi/artiq-dev/artiq) 1.0rc1+204.gfaa268e           <pip>
astropy                   1.1.2               np110py35_0    defaults
asyncserial               0.1             py_3+gitc51e345    http://conda.anaconda.org/m-labs/channel/main/noarch/asyncserial-0.1-py_3+gitc51e345.tar.bz2
babel                     2.2.0                    py35_0    defaults
beautifulsoup4            4.4.1                    py35_0    defaults
bitarray                  0.8.1                    py35_0    defaults
blaze                     0.9.0                     <pip>
blaze-core                0.9.0                    py35_0    defaults
bokeh                     0.11.1                   py35_0    defaults
boto                      2.39.0                   py35_0    defaults
bottleneck                1.0.0               np110py35_0    defaults
cffi                      1.5.2                    py35_0    defaults
chardet                   2.3.0                     <pip>
clyent                    1.2.1                    py35_0    defaults
colorama                  0.3.6                    py35_0    defaults
conda                     4.0.5                    py35_0    defaults
conda-build               1.19.2                   py35_0    defaults
conda-env                 2.4.5                    py35_0    defaults
configobj                 5.0.6                    py35_0    defaults
cryptography              1.2.3                    py35_0    defaults
curl                      7.45.0                        0    defaults
cycler                    0.10.0                   py35_0    defaults
cython                    0.23.4                   py35_0    defaults
cytoolz                   0.7.5                    py35_0    defaults
datashape                 0.5.1                    py35_0    defaults
decorator                 4.0.9                    py35_0    defaults
docutils                  0.12                     py35_0    defaults
dynd                      f641248                   <pip>
dynd-python               0.7.1                    py35_0    defaults
et-xmlfile                1.0.1                     <pip>
et_xmlfile                1.0.1                    py35_0    defaults
fastcache                 1.0.2                    py35_0    defaults
flask                     0.10.1                   py35_1    defaults
fontconfig                2.11.1                        5    defaults
freetype                  2.5.5                         0    defaults
greenlet                  0.4.9                    py35_0    defaults
h5py                      2.5.0               np110py35_4    defaults
hdf5                      1.8.15.1                      2    defaults
idna                      2.0                      py35_0    defaults
ipykernel                 4.3.1                    py35_0    defaults
ipython                   4.1.2                    py35_0    defaults
ipython-genutils          0.1.0                     <pip>
ipython-notebook          4.0.4                    py35_0    defaults
ipython-qtconsole         4.0.1                    py35_0    defaults
ipython_genutils          0.1.0                    py35_0    defaults
ipywidgets                4.1.1                    py35_0    defaults
itsdangerous              0.24                     py35_0    defaults
jbig                      2.1                           0    defaults
jdcal                     1.2                      py35_0    defaults
jedi                      0.9.0                    py35_0    defaults
jinja2                    2.8                      py35_0    defaults
jpeg                      8d                            0    defaults
jsonschema                2.4.0                    py35_0    defaults
jupyter                   1.0.0                    py35_1    defaults
jupyter-client            4.2.1                     <pip>
jupyter-console           4.1.1                     <pip>
jupyter-core              4.1.0                     <pip>
[... 101 lines cut ...]
sqlite                    3.9.2                         0    defaults
statsmodels               0.6.1               np110py35_0    defaults
sympy                     1.0                      py35_0    defaults
system                    5.8                           2    defaults
tables                    3.2.2                     <pip>
terminado                 0.5                      py35_1    defaults
theano                    0.7.0               np110py35_0    defaults
tk                        8.5.18                        0    defaults
toolz                     0.7.4                    py35_0    defaults
tornado                   4.3                      py35_0    defaults
traitlets                 4.1.0                    py35_0    defaults
ujson                     1.35                     py35_0    defaults
unicodecsv                0.14.1                   py35_0    defaults
[... cut ...]
xlrd                      0.9.4                    py35_0    defaults
xlsxwriter                0.8.4                    py35_0    defaults
xlwt                      1.0.0                    py35_0    defaults
xz                        5.0.5                         1    defaults
yaml                      0.1.6                         0    defaults
zeromq                    4.1.3                         0    defaults
zlib                      1.2.8                         0    defaults
rabi@68810MAGTRAPVM:~/artiq-dev/artiq$ 
```

sbourdeauducq, 2016-04-14:
I meant the dev conda channel (which tracks master) but it's probably better that you build the bitstream from release-1 instead.

sbourdeauducq, 2016-04-14:
https://github.com/m-labs/artiq/tree/release-1

sbourdeauducq, 2016-04-14:
Reproduced when mixing gateware 1.x and software 2.x. Please don't do that (this is why it gives you a warning).

closed 2016-04-14 by sbourdeauducq

whitequark, 2016-04-14:
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xdd in position 324: invalid continuation byte

But why is it doing this? I don't think it should.

sbourdeauducq, 2016-04-14:
If that helps, the string that fails decoding is:

```
b'0x00001747\nartiq.coredevice.dds._DDSGeneric.set_mu\nC:\\Anaconda64\\envs\\sta ble\\lib\\site-packages\\artiq\\coredevice/dds.py:179\n0x00001377\nartiq.coredev ice.dds._DDSGeneric.set\nC:\\Anaconda64\\envs\\stable\\lib\\site-packages\\artiq \\coredevice/dds.py:186\n0x00001de7\nartiq_run_repository\\coredevice_examples\\ simple\xdds_test.DDSTest.run\nrepository\\coredevice_examples\\simple/dds_test.p y:32\n0x00001093\n__modinit__\n<synthesized>:1\n'
```

The problem is fully reproducible.

sbourdeauducq, 2016-04-14:
May be a bug with the old (1.0rc2) compiler and not due to mixing.

sbourdeauducq, 2016-04-14:
With master, the string is:

```
b'0x00000dd3\n__modinit__\nC:\\Anaconda64\\envs\\python\\lib\\site-packages\\artiq\\coredevice/dds.py:189\nartiq.coredevice.dds._DDSGeneric.set\nC:\\Anaconda64\\envs\\python\\lib\\site-packages\\artiq\\coredevice/dds.py:196\nartiq_run_repository\\coredevice_examples\\simple\\dds_test.DDSTest.run\nrepository\\coredevice_examples\\simple/dds_test.py:32\n__modinit__\n<synthesized>:1\n'
```

whitequark, 2016-04-14:
Looks like the problem is here: `artiq_run_repository\\coredevice_examples\\simple\xdds_test.DDSTest.run`. See that `\x`? Looks like some kind of very stupid escaping failure.

r-srinivas, 2016-04-15:
I don't think it's a mixing issue. I get the same error with the software and gateware from 2016-04-04.

With the wrong dds bus channel I get the following error message.

```
C:\Anaconda3\artiq_test>call activate artiq-2016-04-04
Deactivating environment "C:\Anaconda3"...
Activating environment "C:\Anaconda3\envs\artiq-2016-04-04"...
ERROR:worker(1209,dds_start.py):root:Terminating with exception (UnicodeDecodeEr
ror: 'utf-8' codec can't decode byte 0xdd in position 324: invalid continuation
byte)
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\master\worker
_impl.py", line 231, in main
    exp_inst.run()
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\language\core
.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
re.py", line 108, in run
    self.comm.serve(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 527, in serve
    self._serve_exception(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 508, in _serve_exception
    traceback = list(reversed(symbolizer(backtrace))) + \
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\coredevice\co
re.py", line 89, in <lambda>
    lambda addresses: target.symbolize(library, addresses)
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\compiler\targ
ets.py", line 166, in symbolize
    library=library) \
  File "C:\Anaconda3\envs\artiq-2016-04-04\lib\site-packages\artiq\compiler\targ
ets.py", line 39, in __enter__
    tempfiles["__stdout__"] = stdout.decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xdd in position 324: invali
d continuation byte
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1209
```

sbourdeauducq, 2016-04-15:
> with the software and gateware updated 04-15.

From main or dev? What is the output of ``python -c "import artiq; print(artiq.__version__)"``?

r-srinivas, 2016-04-15:
From main. 

```

C:\Anaconda3\artiq_test>call activate artiq-2016-04-15
Deactivating environment "C:\Anaconda3"...
Activating environment "C:\Anaconda3\envs\artiq-2016-04-15"...
[artiq-2016-04-15] C:\Anaconda3\artiq_test>python -c "import artiq; print(artiq.
__version__)"
1.0rc2
```

sbourdeauducq, 2016-04-15:
Yes, the problem is in 1.0rc2, and is fixed in git.