m-labs/artiq #395: Problems with artiq importing comment below class(EnvExperiment) as experiment name

r-srinivas, 2016-04-15:
I was trying to test out the TTLs on the transport code so I created a separate file in the same subdirectory `repository/examples` which contains `transport.py` and `transport_ttl_test.py`.

`transport.py` is from the examples folder, `transport_ttl_test.py` looks like this,

 ```Python
import numpy as np

from artiq.experiment import *
from artiq.wavesynth.coefficients import SplineSource

class ttl_test(EnvExperiment):
    """Transport"""
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("bd_sw")
        self.setattr_device("pmt")
        self.setattr_device("electrodes")
        
    def run(self):
        self.bd_sw.pulse(1*ms)
```

In the gui, I get an error saying that there is a duplicate experiment to transport, I see two entries in the GUI called Transport and Transport1. Strangely Transport1 calls all the variables from transport like wait_at_stop, speed, repeats and bins even though I haven't set those arguments in ttl_test. 

I get the same problem if I call it,

```Python
class ttl_test(EnvExperiment):
    """Transport1"""
```

but if I call it,

```Python
class ttl_test(EnvExperiment):
    """Transport 1"""
```

with the space it's fine and I don't see the arguments from Transport. Changing that comment and scanning the repository head in the gui results in the GUI crashing as in issue #388.

jordens, 2016-04-15:
I would generally consider it a bad idea to use the docstring to identify experiments.

r-srinivas, 2016-04-15:
Yes, I wasn't aware that artiq took it as the experiment name until today. I thought it just from the name of the subclass.

sbourdeauducq, 2016-04-16:
> Strangely Transport1 calls all the variables from transport like wait_at_stop, speed, repeats and bins even though I haven't set those arguments in ttl_test.

I'm not sure what you mean. Are you saying that the GUI kept the arguments from ``Transport1`` after you had removed them and scanned the repository? Yes, it is supposed to do that: after you open an experiment for the first time, it memorizes the values from the user and does not reset them when the repository is rescanned. This has nothing to do with the resolution of experiment name conflicts. Click the "Recompute all arguments" button if you want to reset arguments to the repository.

If you don't like experiment names in docstrings, don't use them, then ARTIQ will take the class names instead.

The rest is also normal behavior AFAICT. There has been no release that fixes the crash yet.

r-srinivas, 2016-04-16:
I mean if I call it Transport1, it loads all the arguments from
transport.py not this experiment that is just a bunch of TTL experiments.
It seems to treat it as another instance of the transport.py experiment.

sbourdeauducq, 2016-04-16:
The GUI takes all the arguments from anything that has been called ``Transport1`` before, yes. And it does so until you click "Recompute all arguments".

r-srinivas, 2016-04-16:
Okay, that seems fine then.

sbourdeauducq, 2016-04-16:
Obviously, the GUI has to show a copy of the arguments, so it can be edited by the user. So you need to tell it when you want to reset it to the defaults retrieved by running the experiment's ``build``.

closed 2016-04-16 by sbourdeauducq