m-labs/artiq #396: Specify device/port in connection refused error

r-srinivas, 2016-04-15:
I'm testing out the transport code and have 4 pdqs hooked up. I closed the terminal of one of the pdq_controllers and got this error,

```
ERROR:worker(3224,transport_ttl_test.py):root:Terminating with exception (ConnectionRefusedError: [Errno 111] Connection refused)
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 225, in main
    **expid["arguments"])
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/language/environment.py", line 136, in __init__
    self.build()
  File "/home/rabi/artiq-master/repository/examples/transport_ttl_test.py", line 20, in build
    self.setattr_device("electrodes")
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/language/environment.py", line 218, in setattr_device
    setattr(self, key, self.get_device(key))
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/language/environment.py", line 213, in get_device
    return self.__device_mgr.get(key)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/master/worker_db.py", line 133, in get
    dev = _create_device(desc, self)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/master/worker_db.py", line 93, in _create_device
    return device_class(device_mgr, **desc.get("arguments", {}))
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/devices/pdq2/mediator.py", line 160, in __init__
    self.pdq2s = [dmgr.get(d) for d in pdq2_devices]
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/devices/pdq2/mediator.py", line 160, in <listcomp>
    self.pdq2s = [dmgr.get(d) for d in pdq2_devices]
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/master/worker_db.py", line 133, in get
    dev = _create_device(desc, self)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/master/worker_db.py", line 104, in _create_device
    return cls(desc["host"], desc["port"], target_name)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/site-packages/artiq/protocols/pc_rpc.py", line 103, in __init__
    self.__socket = socket.create_connection((host, port), timeout)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/socket.py", line 711, in create_connection
    raise err
  File "/home/rabi/anaconda3/envs/artiq-2016-04-15/lib/python3.5/socket.py", line 702, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
ERROR:master:artiq.master.scheduler:got worker exception in prepare stage, deleting RID 3224
```

From the traceback I can see it's due to the electrodes entry in device_db

```
  File "/home/rabi/artiq-master/repository/examples/transport_ttl_test.py", line 20, in build
    self.setattr_device("electrodes")
```

But it would be useful if the error message specified that, along with which specific pdq failed.

sbourdeauducq, 2016-04-16:
Modifying the exception (which is generated by Python) sounds inappropriate. What can be done is adding ``logger.debug`` calls that trace what devices are requested and what connections are attempted, and then, when you get the error, you can run the experiment again in debug mode to determine where it failed. Sounds OK?

jordens, 2016-04-16:
Having the device manager rethrow exceptions with added information would be fine here IMHO.

commit b74aa123 (2016-04-16): master/worker_db: rethrow DeviceError exception with device name on device creation failure. Closes #396
closed 2016-04-16 by sbourdeauducq