m-labs/artiq #400: No error if trying to pulse or turn on ttl input

r-srinivas, 2016-04-15:
I tried setting one of my ttl channels, ttl1, to be an input. If I try to pulse or turn it on in an experiment, I don't get a signal but the experiment still runs. It could be useful to get a warning if you're trying to turn on a ttl input.

sbourdeauducq, 2016-04-16:
What should the electrical state be after calling ``output()`` on a TTL line? 1 or 0?

sbourdeauducq, 2016-04-16:
* The TTL module gives an interface to the FPGA I/O pin. Setting level and direction independently makes sense electrically.
* Such a warning is difficult to implement (needs to be fast, needs to be forwarded to the user in a meaningful way). And modifying the internal level of a pin currently set as input is a valid operation anyway.
* There can be other similar issues with e.g. I2C direction switches so a warning here is of little help.
* Direction of FPGA I/O pins is easy to see in the TTL monitoring panel.

closed 2016-04-16 by sbourdeauducq