m-labs/artiq #401: Modify GUI widget based on current selection of another GUI widget

dleibrandt, 2016-04-15:
I have an experiment with many arguments, and I'd like to be able to perform a 1d scan of a single argument.  Rather than use a Scannable object for each argument, I think it would be nicer to use a single Scannable object and a EnumerationValue object to select the variable, as in the example below.

What I'd like is to be able to set the options for the `scan_variable` argument and the units of the `scan_range` argument based on the current selection of the `scan_type` argument.  Is there a way to do this?

```
import numpy as np
from artiq.experiment import *

class SelectScanVariable(EnvExperiment):
    """Select scan variable"""

    def build(self):
        # Get the hardware devices
        self.setattr_device("scheduler")
        
        self.setattr_device("core")
        self.setattr_device("core_dds")
        
        pulse_times = []
        pulse_frequencies = []
        for i in range(10):
            pulse_times.append("t" + str(i))
            pulse_frequencies.append("f" + str(i))
            
            self.setattr_argument("t" + str(i), NumberValue(1000*us, unit="us", scale=1e-6, min=0))
            self.setattr_argument("f" + str(i), NumberValue(200*MHz, unit="MHz", scale=1e6, min=0))

        self.setattr_argument("scan_type", EnumerationValue(["time", "frequency"], "frequency"))
        if self.scan_type == "time":
            self.setattr_argument("scan_variable", EnumerationValue(pulse_times))
            self.setattr_argument("scan_range", Scannable(global_min=0,
                global_max=1, default=LinearScan(0, 1000e-6, 100), scale=1e-6, ndecimals=3, unit="us"))
        if self.scan_type == "frequency":
            self.setattr_argument("scan_variable", EnumerationValue(pulse_frequencies))
            self.setattr_argument("scan_range", Scannable(global_min=0,
                global_max=1e9, default=LinearScan(200e6, 201e6, 100), scale=1e6, ndecimals=6, unit="MHz"))
                
        self.set_dataset("detect_scan_x", np.full(self.scan_range.npoints, np.nan),
                         broadcast=True, save=True)
        self.set_dataset("detect_scan_y", np.full(self.scan_range.npoints, np.nan),
                         broadcast=True, save=True)
        
    def run(self):
        for i, x in enumerate(self.scan_range):
            print(x)
            setattr(self, self.scan_variable, x)
            
            # do the experiment
            
            self.mutate_dataset("detect_scan_x", i, x)
            self.mutate_dataset("detect_scan_y", i, np.random.poisson(20))
```

sbourdeauducq, 2016-04-16:
Not currently.

sbourdeauducq, 2016-06-29:
Use one ``Scannable`` per variable, the new "disable other scans" feature of the GUI, and perhaps the new ``MultiScanManager`` if you have a lot of variables.

closed 2016-06-29 by sbourdeauducq