m-labs/artiq #402: Question about self.core.comm.close()

r-srinivas, 2016-04-15:
I want to run the following experiment,

```Python
from artiq.experiment import *
import numpy as np

class detect_test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl1")
        self.setattr_device("ttl6")
        self.points = 2
        self.input = np.random.rand(self.points)
        self.results = np.zeros((self.points, 1))

    def run(self):
        for i in range(self.points):
            ct = int(20*self.input[i])
            self.results[i] = self.detect(ct)
            self.core.comm.close()
        print(self.results)

    @kernel
    def detect(self, rand_input):
        with parallel:
            self.ttl6.gate_rising(100*us)
            with sequential:
                for i in range(rand_input):                
                    self.ttl1.pulse(1*us)
                    delay(1*us)
        counts = self.ttl6.count()
        return counts
```

Where I want to store the counts in an array that I print later. If I run it for one points it's fine, for more than one point I get,

I get the following error,

```
ERROR:worker(1476,detect_test.py):root:While compiling <repository>\test_experim
ents\detect_test.py
<synthesized>:1:1-1:63: error: host object has an attribute 'i_previous_timestam
p' of type int(width=32), which is different from previously inferred type int(w
idth=64) for the same attribute
`<artiq.coredevice.ttl.TTLInOut object at 0x000000000473EF98>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\test_experiments\detect_test.py:29:5: note: expanded from here
    def detect(self, rand_input):
    ^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1476
ERROR:worker(1477,detect_test.py):root:While compiling <repository>\test_experim
ents\detect_test.py
<synthesized>:1:1-1:63: error: host object has an attribute 'i_previous_timestam
p' of type int(width=32), which is different from previously inferred type int(w
idth=64) for the same attribute
`<artiq.coredevice.ttl.TTLInOut object at 0x000000000473EEF0>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\test_experiments\detect_test.py:29:5: note: expanded from here
    def detect(self, rand_input):
    ^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1477
```

I get the same error if I have, `self.core.break.real_time()`

```
from artiq.experiment import *
import numpy as np

class detect_test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl1")
        self.setattr_device("ttl6")
        self.points = 2
        self.input = np.random.rand(self.points)
        self.results = np.zeros((self.points, 1))

    def run(self):
        for i in range(self.points):
            ct = int(20*self.input[i])
            self.results[i] = self.detect(ct)
        print(self.results)

    @kernel
    def detect(self, rand_input):
        self.core.break_realtime()
        with parallel:
            self.ttl6.gate_rising(100*us)
            with sequential:
                for i in range(rand_input):                
                    self.ttl1.pulse(1*us)
                    delay(1*us)
        counts = self.ttl6.count()
        return counts
```

With error message,

```
ERROR:worker(1478,detect_test.py):root:While compiling <repository>\test_experim
ents\detect_test.py
<synthesized>:1:1-1:63: error: host object has an attribute 'i_previous_timestam
p' of type int(width=32), which is different from previously inferred type int(w
idth=64) for the same attribute
`<artiq.coredevice.ttl.TTLInOut object at 0x000000000473EF98>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\test_experiments\detect_test.py:29:5: note: expanded from here
    def detect(self, rand_input):
    ^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1478
ERROR:worker(1479,detect_test.py):root:While compiling <repository>\test_experim
ents\detect_test.py
<synthesized>:1:1-1:63: error: host object has an attribute 'i_previous_timestam
p' of type int(width=32), which is different from previously inferred type int(w
idth=64) for the same attribute
`<artiq.coredevice.ttl.TTLInOut object at 0x000000000473EF60>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\test_experiments\detect_test.py:28:5: note: expanded from here
    def detect(self, rand_input):
    ^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1479
```

How should I run this experiment, then?

r-srinivas, 2016-04-16:
Maybe unrelated but trying to measure how long this step takes,

`self.results[i] = self.detect(ct)`

using `time.time` gives me a value of about 3.5s. That seems a little long to me. Is that normal?

sbourdeauducq, 2016-04-16:
As demonstrated by:

```python
from artiq.experiment import *

class BlinkForever(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")

    @kernel
    def k(self):
        self.led.pulse(1000*ms)
        delay(1000*ms)

    def run(self):
        #self.led.o_previous_timestamp = 0
        print(type(self.led.o_previous_timestamp))
        self.k()
        print(type(self.led.o_previous_timestamp))
        self.k()
```

```
<class 'artiq.language.core.int'>
WARNING:artiq.coredevice.comm_generic:Mismatch between gateware (2.0.dev+210.gd4ba525) and software (2.0.dev+210.gd4ba525.dirty) versions
<class 'int'>
```

Another odd problem is the second compilation of ``k`` is not rejected in my example.

whitequark, 2016-04-21:
The second compilation is not rejected because the embedding code does the equivalent of guessing the width for int literals when typechecking. No mystery here.

closed 2016-04-21 by whitequark

r-srinivas, 2016-04-26:
Would this fix be in 1.0rc3 ? I'm running the following experiment on windows,

```python
from artiq.experiment import *
import numpy as np

class detect_test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl1")
        self.setattr_device("ttl2")
        self.setattr_device("ttl6")
        self.points = 1
        self.input = np.random.randint(0, 20, self.points)
        self.counts = []

    def run(self):
        for i in range(self.points):
            self.counts.append(self.detect(int(self.input[i])))
        print(self.counts)

    @kernel
    def detect(self, pulses):
        with parallel:
            self.ttl6.gate_rising(100*us)
            with sequential:
                for i in range(pulses):
                    with parallel:
                        self.ttl1.pulse(1*us)
                        self.ttl2.pulse(1*us)
                    delay(1*us)
        return self.ttl6.count()
```

If `self.points = 1`, it's fine, if `self.points=2` I get this error,

```
ERROR:worker(1652,detect_test.py):root:While compiling <repository>\test_experim
ents\detect_test.py
<synthesized>:1:1-1:63: error: host object has an attribute 'i_previous_timestam
p' of type int(width=32), which is different from previously inferred type int(w
idth=64) for the same attribute
`<artiq.coredevice.ttl.TTLInOut object at 0x000000000475CB00>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\test_experiments\detect_test.py:29:5: note: expanded from here
    def detect(self, pulses):
    ^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1652
```

whitequark, 2016-04-26:
The fix is not in `1.0rc3`.

commit 5cccdcaa (2016-04-21): coredevice: deserialize int64(width=64) as int(width=64), not host_int.

Fixes #402.
commit 8a28039b (2016-04-22): coredevice: deserialize int64(width=64) as int(width=64), not host_int.

Fixes #402.