m-labs/artiq #403: Can't access numpy.ints in kernel

r-srinivas, 2016-04-16:
For the following experiment,

```Python
from artiq.experiment import *
import numpy as np

class array_test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.points = np.array([int(2)])
        
    @kernel
    def run(self):
        for i in range(self.points[0]):
            print(i)
```

I get the error message,

```
ERROR:worker(1505,array_test.py):root:While compiling <repository>\tests\array_t
est.py
<repository>\tests\array_test.py:19:24-19:35: error: type <instance numpy.ndarra
y {
                __objectid__: int(width=32)
        }> is not iterable
        for i in range(self.points[0]):
                       ^^^^^^^^^^^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1505
```

Is there a way to use elements of numpy arrays in the kernel? Could be useful to have that functionality.

r-srinivas, 2016-04-16:
I guess I could have a separate function that takes `int(self.points[0])` as an input on the array and call that in run, with run outside the kernel. The issue I had was if I wanted to quickly append an element of array to store results of counts, but I suppose that would work on a list instead?

sbourdeauducq, 2016-04-16:
Yes, numpy is not supported in kernels.

closed 2016-04-16 by sbourdeauducq