m-labs/artiq #404: Terminating experiment that runs forever

r-srinivas, 2016-04-16:
I have the following experiment that I want to run in the background while no experiments are running. My first attempt was,

```Python
from artiq.experiment import *

class IonMonitor(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl2")

    @kernel
    def run(self):
        while True:
            self.ttl2.pulse(10*ms)
            delay(10*ms)
```

Which gives me the pulse sequences I want. However, I can't terminate an instance of this experiment and if I schedule another experiment with a higher priority it doesn't run. 

My second attempt was,

```Python
from artiq.experiment import *

class IonMonitor(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl2")
        self.setattr_device("scheduler")

    def run(self):
        while True:
            self.scheduler.pause()
            self.pulse()

    @kernel
    def pulse(self):
        self.ttl2.pulse(10*ms)
        delay(10*ms)
```

Which generated an underflow error.
```
RID 1562
ERROR:worker(1563,ion_monitor.py):root:Terminating with exception (RTIOUnderflow
: RTIO underflow at 1718678108032 mu, channel 2, slack -968181408 mu)
Core Device Traceback (most recent call last):
  File "C:\Anaconda3\artiq_test\repository\test_experiments/ion_monitor.py", lin
e 24, in artiq_worker_C:\Anaconda3\artiq_test\repository\test_experiments\ion_mo
nitor.IonMonitor.pulse (RA=0x638)
    self.ttl2.pulse(10*ms)
  File "<artiq>\coredevice/ttl.py", line 151, in artiq.coredevice.ttl.TTLInOut.p
ulse (RA=0x7a0)
    self.on()
  File "<artiq>\coredevice/ttl.py", line 130, in artiq.coredevice.ttl.TTLInOut.o
n (inlined)
    self.set_o(True)
  File "<artiq>\coredevice/ttl.py", line 115, in artiq.coredevice.ttl.TTLInOut.s
et_o (inlined)
    rtio_output(now_mu(), self.channel, 0, 1 if o else 0)
  File "/var/lib/buildbot/slaves/debian-stretch-amd64-1/miniconda/conda-bld/work
/artiq/runtime/rtio.c", line 28, in rtio_process_exceptional_status
    <unknown>
artiq.coredevice.exceptions.RTIOUnderflow(0): RTIO underflow at 1718678108032 mu
, channel 2, slack -968181408 mu
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\master\worker
_impl.py", line 231, in main
    exp_inst.run()
  File "C:\Anaconda3\artiq_test\repository\test_experiments\ion_monitor.py", lin
e 20, in run
    self.pulse()
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\language\core
.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\coredevice\co
re.py", line 108, in run
    self.comm.serve(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 527, in serve
    self._serve_exception(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 519, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.RTIOUnderflow: RTIO underflow at 1718678108032 mu, c
hannel 2, slack -968181408 mu
```

Third attempt was,

```Python
from artiq.experiment import *

class IonMonitor(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl2")
        self.setattr_device("scheduler")

    def run(self):
        while True:
            self.scheduler.pause()
            self.pulse()

    @kernel
    def pulse(self):
        self.core.break_realtime()
        self.ttl2.pulse(10*ms)
        delay(10*ms)
```

Which seems to work. It runs forever, and I can terminate it or schedule other experiments above it. However, if I look on the scope, the pulses are about 1.5 s apart, which leaves a significant dead time between loops.

Furthermore, if I schedule another experiment ahead of it, it terminates and runs that other experiment, but doesn't resume Ion Monitor. I instead get this error message.

```
ERROR:worker(1574,ion_monitor.py):root:Terminating with exception (OSError: Conn
ection closed)
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\master\worker
_impl.py", line 231, in main
    exp_inst.run()
  File "C:\Anaconda3\artiq_test\repository\test_experiments\ion_monitor.py", lin
e 20, in run
    self.pulse()
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\language\core
.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\coredevice\co
re.py", line 106, in run
    self.comm.load(kernel_library)
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 293, in load
    self._read_empty(_D2HMsgType.LOAD_COMPLETED)
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 132, in _read_empty
    self._read_header()
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 104, in _read_header
    (sync_byte, ) = struct.unpack("B", self.read(1))
  File "C:\Anaconda3\envs\artiq-2016-04-15\lib\site-packages\artiq\coredevice\co
mm_tcp.py", line 60, in read
    raise IOError("Connection closed")
OSError: Connection closed
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1574
```

 What's the best way of writing this experiment?

jordens, 2016-04-16:
You can also write an experiment that reschedules itself. But that will also have the slow startup time as this one. Alternatively run 100 tight loops of your background experiment before pausing.

Also please try to keep separate things separate. File a separate issue for the slow kernel compilation, loading and starting (if there is not one already). I pointed out that this is a weak point. Windows seems to be able to exacerbate it quite a bit.

What exactly are you doing in your last segment there? What/who terminates what and which one of those is "Ion Monitor"? Also probably a different issue.

r-srinivas, 2016-04-16:
I have another experiment,

```Python
from artiq.experiment import *
import numpy as np

class TTL_test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        #  self.setattr_device("ttl1")
        self.setattr_device("ttl2")
        #  self.setattr_device("ttl6")

    @kernel
    def run(self):
        self.ttl2.pulse(10*us)
```

That I'm running at a higher priority. IonMonitor is the name of the experiment I want to run in the background. TTL_test terminates IonMonitor and I want IonMonitor to resume once TTL_test is done.

jordens, 2016-04-16:
I know you do but it is not helpful. If there are multiple issues, file multiple issues.

What do you mean by "TTL_test terminates IonMonitor"? Actively or is that what you hypothesize from the traceback above?

jordens, 2016-04-16:
Also, which ARTIQ version is this? There are a bunch of changes in the pipeline which should improve compilation time.

If you file an issue, there is a request to read the issue writing guidelines which also asks you this information.

r-srinivas, 2016-04-16:
I can see it pauses ion monitor from the TTL signals on the scope. It's the
artiq version I had installed from conda yesterday.

r-srinivas, 2016-04-16:
This was on windows I should add.

jordens, 2016-04-16:
Again: which version? From the dev label or from main?

r-srinivas, 2016-04-16:
Main.

jordens, 2016-04-16:
Then regarding the compilation and uploading time, wait for the next rc or try dev. File a separate concise issue for the other problem if it is actually distinct from https://github.com/m-labs/artiq/issues/393. Please close this one if there are not other issues you bundled here.

sbourdeauducq, 2016-04-17:
> Furthermore, if I schedule another experiment ahead of it, it terminates and runs that other experiment, but doesn't resume Ion Monitor. I instead get this error message.

This is why I told you to do ``self.core.comm.close()`` before ``self.scheduler.pause()``
(2.0 will do that automatically).

r-srinivas, 2016-04-25:
Okay, `self.core.comm.close()` solves the issue with the experiment runs in the background, though there's still a large deadtime between loops, of about 1.1 ms or so.

closed 2016-04-26 by sbourdeauducq