m-labs/artiq #406: set_dataset in build causes error

dleibrandt, 2016-04-25:
I'm running 1.0rc3.  When I uncomment the commented line in build() below, I get the following error:
```
root:Terminating with exception (AttributeError: type object 'ParentDatasetDB' has no attribute 'set')
```

```
class DatasetDemo(EnvExperiment):
    """Dataset demo"""

    def build(self):
        self.setattr_device("core")
        # self.set_dataset("test_dataset_build", 7, broadcast=True, persist=True, save=True)
        
    def prepare(self):
        self.set_dataset("test_dataset_prepare", 8, broadcast=True, persist=True, save=True)
        
    def run(self):
        self.set_dataset("test_dataset_run", 9, broadcast=True, persist=True, save=True)
```

sbourdeauducq, 2016-04-26:
Why do you want to set a dataset in build?

dleibrandt, 2016-04-26:
I had been creating empty dataset arrays in build that get populated during run.  This worked in 1.0rc2, but doesn't work in 1.0rc3.

This certainly isn't a necessary capability.  I can easily create the arrays in prepare or even in run, but I would suggest adding something to the documentation saying that datasets can only be set in xxx methods.

closed 2016-04-26 by sbourdeauducq

sbourdeauducq, 2016-04-27:
Modifying datasets in ``build`` is not a good idea because this function is executed when scanning the repository and recomputing arguments.