m-labs/artiq #407: Long dead time between experiments on Windows

r-srinivas, 2016-04-25:
I'm using 1.0 rc3 on Windows. I have the following experiment  `IonMonitor` in the background,

```python
from artiq.experiment import *

class IonMonitor(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl2")
        self.setattr_device("scheduler")

    def run(self):
        while True:
            self.core.comm.close()
            self.scheduler.pause()
            self.pulse()

    @kernel
    def pulse(self):
        self.core.break_realtime()
        for i in range(1):
            self.ttl2.pulse(10*ms)
            delay(10*ms)
```

And then I run another experiment `TTL_test` with a higher priority,

```python
from artiq.experiment import *

class TTL_test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl1")

    @kernel
    def run(self):
        self.ttl1.pulse(10*us)
```

This experiment executes and the previous experiment `IonMonitor` turns on again. However, between the last `IonMonitor` pulse and the `TTL_test` pulse, there's a gap of about 1.2s, and 1.1s between the `TTL_test` pulse and the next `IonMonitor` pulse. Is there a way to shorten this gap?

Using 1.0rc3 on linux this gap is about 140 ms and 170 ms for before and after TTL test respectively.

whitequark, 2016-04-25:
I would expect this to result from an additional constant latency of running the linker on Windows. 

Generally, you should expect a relatively high latency every time kernel mode is entered. Type checking and compilation take a while, especially once you move past trivial experiments.

r-srinivas, 2016-04-25:
Is there a better way to code the background experiment then?

whitequark, 2016-04-25:
Yes. The loop that performs the pulses should not preempt itself after every pulse. Instead, it should just run forever in the background, and be preempted once there is a higher priority experiment scheduled.

However, I think this is not currently possible. @sbourdeauducq ?

r-srinivas, 2016-04-26:
Could programming the idle kernel solve this? Would that turn on when there's a dead time between experiments or only when the ethernet cable is disconnected?

whitequark, 2016-04-26:
Yes, I think so. There's very little dead time between stopping the idle kernel and starting the newly uploaded one.

sbourdeauducq, 2016-04-26:
What about adding a function ``self.scheduler.has_higher_priorities()`` that returns whether a higher priority experiment is present, and ``self.scheduler.pause()`` will preempt the current one?

This way ``IonMonitor`` can RPC it in its kernel and the kernel will only return when needed, and there will be a smaller gap (~millisecond) when there is no preemption.

There will still be a large gap when it *is* preempted, though.

sbourdeauducq, 2016-04-26:
And the problem with the idle kernel is it cannot modify datasets.

whitequark, 2016-04-26:
Sounds reasonable. And we can remove *that* gap easily by only preempting after compilation is done.

sbourdeauducq, 2016-04-26:
How exactly? Add a feature to pre-compile kernels, which can be invoked in the prepare stage?

r-srinivas, 2016-04-26:
That seems reasonable. How long could the gap be reduced to if the preempting is only done after compilation? But this wouldn't reduce the dead time for two non-looping experiments sequenced one after another, right?

whitequark, 2016-04-26:
@sbourdeauducq I thought about some kind of hack to return from the `has_higher_priorities` RPC early from the kernel's perspective, let it do a few more loops while we're compiling, and then finally request termination. But this sounded much better in my head, we probably shouldn't actually do that.

jordens, 2016-04-30:
200 ms of the dead time are due to #398

jordens, 2016-05-03:
And the rest of the delay is likely due to windows generally being slow with new processes and especially python that opens many files.

I would shape the behavior of yielding from a kernel by checking for higher priorities a bit differently. I would handle both `TerminationRequested` and a new `YieldRequested` through that channel. And have `preempt()` from a kernel would RPC, potentially raise either of those, potentially return, and then the host side can handle them or convert them, e.g. into `pause()`. (Suggestions for better names welcome).

sbourdeauducq, 2016-05-03:
> windows generally being slow with new processes and especially python that opens many files

The pipelining should absorb the Python process creation. Could be the linker as @whitequark suggests.

whitequark, 2016-05-03:
Now that we're on LLVM 3.8 we can use the LLVM linker, and rip out a whole bunch of cruft at the same time as making it all faster.

whitequark, 2016-05-03:
That will also:
  * remove the binutils dependency;
  * make runtime smaller as we can make use of LTO;
  * improve error messages as LLVM's symbolizer works better than GNU's. (this is not strictly related to the linker but it's an adjacent API that would be convenient to make use of at the same time.)

jordens, 2016-05-03:
Pipelining does not absorb anything significant for the experiments in question. They have a few us/ms of duration in `run()`. It is not a secret that Python startup is dead slow on Windows.

jordens, 2016-05-03:
Maybe the pipelining can play a role in some `pause()` cases. But in general, for a queue of experiments lasting a few ms each, it won't help.

sbourdeauducq, 2016-05-05:
Pipelining does not help if the desired experiment rate is higher than the process creation rate. But for pause, the scheduler only preempts the low-priority experiment when the high-priority one is ready to execute run (and build and prepare have been executed), so it does absorb process creation.

r-srinivas, 2016-05-26:
We tried loading an idle_kernel but pulses a ttl but it doesn't seem to load in between kernels of the IonMonitor loop. Is that something that should happen?

sbourdeauducq, 2016-05-26:
Is the IonMonitor code still exactly as above?

r-srinivas, 2016-05-26:
No, but similar.

```python
from artiq.experiment import *

class Ion_Monitor_Test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")
        self.setattr_device("ttl_pmt")
        self.setattr_device("scheduler")

    def run(self):
        while True:
            self.scheduler.pause()
            self.pulse()
                
    @kernel
    def pulse(self):
        self.core.break_realtime()
        self.ttl_bdd.pulse(100*ms)
        with parallel:
            self.ttl_pmt.gate_rising(100*us)
            self.ttl_bdd.pulse(100*us)
        counts = self.ttl_pmt.count()
        self.set_dataset("counts", counts, broadcast=True)
```

Our idle_kernel is,

```Python
from artiq.experiment import *

class IdleKernel(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")

    @kernel
    def run(self):
        while True:
            self.ttl_bdd.pulse(990*us)
            delay(10*us)
```

If you have a look at the BDD on the scope, there's about 1.5 s dead time on the scope between ttl_bdd pulses.

r-srinivas, 2016-05-26:
Okay, looks like putting in `self.core.comm.close()` is still required for the idle_kernel to run. I thought scheduler.pause called that automatically.

sbourdeauducq, 2016-05-26:
With this code then it is expected that the idle kernel is not run. The idle kernel is only run when there is no valid TCP connection with the control PC, and on ARTIQ 1.x ``scheduler.pause`` does not implicitly disconnect.

sbourdeauducq, 2016-05-26:
@whitequark 1.5s is particularly long. Could you do a quick profiling of the compiler on Windows to see if there could be any low-hanging fruits?

r-srinivas, 2016-05-26:
The desired behavior of the idle kernel would be to have it run when there is no other kernel running, not only at times where there is no TCP connection.  This is less of an issue if the delay time between kernels were ~100 ms, but at 1-2 seconds this is an issue for keeping ions in surface traps between experiments.

With `self.core.comm.close()` the delay is still 750 ms before the idle kernel starts.

sbourdeauducq, 2016-05-26:
No, use seamless handover. That will even cover your ~140ms delay on Linux.

r-srinivas, 2016-05-26:
How does seamless handover work?

sbourdeauducq, 2016-05-26:
When a kernel exits and you are not disconnecting, the RTIO core is not reset and it keeps executing what has been programmed into its output FIFOs. Additionally, when you run the next kernel, the initial value of ``now`` is what it was when the previous kernel exited. This is demonstrated by this example that blinks a LED with a precise and steady frequency despite the constant compiling and reloading of kernels:
https://github.com/m-labs/artiq/blob/master/artiq/examples/master/repository/coredevice_examples/simple/handover.py

r-srinivas, 2016-05-26:
That example doesn't seem to work on Windows as is. I had to increase the delay to 750*ms to avoid underflows.

r-srinivas, 2016-05-26:
If I understand it right, the compilation and reloading of the kernels has to be done quicker than the execution of the kernel for the seamless handover, to work, right?

What's happening in this example here, https://github.com/m-labs/artiq/blob/master/artiq/examples/master/repository/coredevice_examples/simple/blink_forever.py
?

sbourdeauducq, 2016-05-26:
You need to have sufficient timeline delays in the kernels to allow for compilation and loading of a second kernel. It a similar situation as RPCs.

r-srinivas, 2016-05-26:
Calling `self.core.comm.close()`, should we still see a 750ms dead time between ttl_bdd pulses?

sbourdeauducq, 2016-05-27:
No. ``comm.close`` (which is required by ``scheduler.pause``) will disconnect and cause RTIO reset + startup of any idle kernel.

What about leaving bdd on at the end of the IonMonitor.pulse? Something like:
```python
self.bdd.on()
delay(100*us) # guarantees a minimum time with bdd on (you probably don't need this line)
# make sure that this kernel doesn't exit before all programmed RTIO events have been
# executed, so that a possible RTIO reset won't swallow bdd.on()
while self.core.get_rtio_counter() < now():
    pass
``` 
See also the discussion in #425 that proposes a different way of handling handovers and the idle kernel, that could make such things easier.

r-srinivas, 2016-06-13:
Just to update, I had a look at the delays between scan points for this experiment, looks like between the kernels it is 3.5s. 

Using 1.0 on windows,

```python
from artiq.experiment import *
from repository.magtrap_subcomponents import Detection
import numpy as np

class BD_Scan_Test(EnvExperiment):
    '''BD Scan Test Pause'''
    
    def build(self):
        #  build hardware for experiment
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")
        self.setattr_device("ttl_bd1")
        self.setattr_device("ttl_bd2")
        self.setattr_device("ttl_bd_ne")
        self.setattr_device("dds_bd1")
        self.setattr_device("scheduler")
        #  instance of detection subcomponent
        self.detect = Detection(parent=self)
        #  variable to be scanned
        self.setattr_argument("frequency_scan", Scannable(
            default=LinearScan(300, 340, 100)))
        self.setattr_argument("repetitions", NumberValue(100, min=10, max=500, 
                                                         ndecimals=0))
        self.setattr_argument("nbins", NumberValue(100, ndecimals=0))
        #  call value for dds from dataset database        
        self.f0 = self.get_dataset("dds_f.bd1")

    def run(self):
        l = len(self.frequency_scan)
        #  set up size of arrays to hold data
        self.set_dataset("dds_frequency", np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset("counts", np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset("counts_hist", np.empty((l, int(self.nbins))),
                         broadcast=True, save=True)
        self.set_dataset("counts_hist_bins", np.linspace(0, 100, self.nbins+1),
                         broadcast=True, save=False)
        self.counts = [0]*int(self.repetitions)
        for i, f in enumerate(self.frequency_scan):
            # need to call core.comm.close for scheduler.pause()
            self.core.comm.close()
            self.scheduler.pause()
            self.dds_scan(i, f)
        #  resets dds_bd1 to default value at the end of all scan points
        self.dds_bd1.set(self.f0)

    @kernel
    def dds_scan(self, index, frequency):
        hist = [0]*int(self.repetitions)
        total = 0
        #  need to break real time between experiments due to delay from
        #  mutate dataset
        self.core.break_realtime()
        self.dds_bd1.set(frequency*MHz)
        #  turn bdd off from previous loop
        self.ttl_bdd.off()
        for i in range(int(self.repetitions)):
            with parallel:
                #  pulse sequence for each experiment
                self.ttl_bd1.pulse(50*us)
                self.ttl_bd2.pulse(50*us)
                self.ttl_bd_ne.pulse(50*us)
            self.ttl_bdd.pulse(500*us)
            n = self.detect.bd_ne(100*us)
            #  delay from gate rising in detect
            delay(10*us)
            if n >= int(self.nbins):
                n = int(self.nbins)-1
            hist[n] += 1
            total += n
        #  need this pulse to keep bdd on till next scan point
        self.ttl_bdd.on()
        #  upload data from scan point to dataset for plotting, incurs delay
        self.mutate_dataset("counts_hist", index, hist)
        self.mutate_dataset("counts", index, total)
        self.mutate_dataset("dds_frequency", index, frequency)
```

Looking at the ttl_bd on the scope,
[image]

So about 3.5 s between kernels.

whitequark, 2016-07-08:
@sbourdeauducq  I'm using the following code with `python -m artiq.compiler.testbench.perf_embedding`:

```python
from artiq.experiment import *

class Benchmark(EnvExperiment):

    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")

    @kernel
    def run(self):
        while True:
            self.ttl0.pulse(990*us)
            delay(10*us)
```

On Linux:
  * embedding takes 40ms/run (mostly parser)
  * transforms take 13ms/run
  * optimization and linking takes 101ms/run

On Windows, on a hardware-assisted VM on the same machine:
  * embedding takes 195ms/run
  * transforms take 62ms/run
  * optimization and linking takes 182ms/run

On Linux, on the same hardware-assisted VM on the same machine, with paravirtualization disabled:
  * embedding takes 192ms/run
  * transforms take 70ms/run
  * optimization and linking takes 170ms/run

(Enabling paravirtualization makes it significantly faster.)

So... I would conclude that either 1) the cause of slowdown is not the compiler, since there are no differences between Linux and Windows except noise, or 2) the slowdown is present on bare hardware but vanishes when the OSes are virtualized.

sbourdeauducq, 2016-07-08:
Bah, what happens with the Raghu's test case?

whitequark, 2016-07-08:
On the host Linux system (note that I had to hack it to remove the `self.detect.bd_ne` call as I've no source available):

* embedding takes 190ms/run (mostly parser)
* transforms take 64ms/run
* optimization and linking takes 642ms/run

The last part is high because I only have a debug LLVM locally, which is slow. In any case 3.5s seems excessive.

sbourdeauducq, 2016-07-08:
@r-srinivas Please post your full source including the ``bd_ne`` method.
@whitequark And on Windows?

whitequark, 2016-07-08:
On the Windows VM:

* embedding takes 946ms/run (mostly parser)
* transforms take 322ms/run
* optimization and linking takes 823ms/run

Curiously, native-code linker is penalized much less than interpreted Python by being inside a VM.

r-srinivas, 2016-07-08:
The detection class was,

```python
from artiq.experiment import *

class Detection(HasEnvironment):

    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_pmt")
        self.setattr_device("ttl_bd1")
        self.setattr_device("ttl_bd2")
        self.setattr_device("ttl_bdd")
        self.setattr_device("ttl_bd_ne")

    @kernel
    def bd_ne(self, time):
        with parallel:
            self.ttl_pmt.gate_rising(time)
            self.ttl_bd1.pulse(time)
            self.ttl_bd2.pulse(time)
            self.ttl_bd_ne.pulse(time)
        return self.ttl_pmt.count()

    @kernel
    def pulse_ne(self, time):
        with parallel:
            self.ttl_bd1.pulse(time)
            self.ttl_bd2.pulse(time)
            self.ttl_bd_ne.pulse(time)

    @kernel
    def bdd(self, time):
        with parallel:
            self.ttl_pmt.gate_rising(time)
            self.ttl_bdd.pulse(time)
        return self.ttl_pmt.count()

    @kernel
    def no_laser(self, time):
        self.ttl_pmt.gate_rising(time)
        return self.ttl_pmt.count()
```

r-srinivas, 2016-07-08:
Using 1.1 and gateware (1.1+11.g6a1706b), with the following code (similar to the previous bd scan test but we had cleaned it up a little)

```python
from artiq.experiment import *
from repository.magtrap_subcomponents import Detection
import numpy as np

class BD_Scan(EnvExperiment):
    '''BD Scan Pause Test'''

    def build(self):
        #  build hardware for experiment
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")
        self.setattr_device("dds_bd1")
        self.setattr_device("scheduler")
        #  instance of detection subcomponent
        self.detect = Detection(parent=self)
        #  arguments in gui
        self.setattr_argument("frequency_scan", Scannable(
            default=LinearScan(300, 330, 100)))
        self.setattr_argument("repetitions", NumberValue(100, min=10, max=500, 
                                                         ndecimals=0))
        self.setattr_argument("nbins", NumberValue(100, ndecimals=0))
        self.setattr_argument("t_detect", NumberValue(100e-6, scale=1e-6,
                                                      unit='us', ndecimals=0))
        #  call value for dds from dataset database        
        self.f0 = self.get_dataset("dds_f.bd1")

#==============================================================================
#  prepare stage for initialising datasets
#==============================================================================

    def prepare(self):
        self.xvar = "data.dds_bd1_f"
        self.xlabel = self.xvar + ".x"
        self.counts_label = self.xvar + ".counts"
        self.counts_hist_label = self.xvar + ".counts_hist"
        self.counts_hist_bins_label = self.xvar + ".counts_hist_bins"
        l = len(self.frequency_scan)
        #  set up size of arrays to hold data
        self.set_dataset(self.xlabel, np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_label, np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_hist_label, np.empty((l, int(self.repetitions))),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_hist_bins_label, np.linspace(0, 100, self.nbins+1),
                         broadcast=True, save=False)
        self.counts = [0]*int(self.repetitions)
        #  generate list of frequencies as numpy arrays not supported
        #  on core device
        self.frequencies = list(self.frequency_scan)
    
#==============================================================================
#  run
#==============================================================================

    def run(self):
        self.experiment(self.frequencies)

    #  using this function to have all scan points on kernel
    def experiment(self, frequencies):
        i = 0
        for f in frequencies:
            self.core.comm.close()
            self.scheduler.pause()
            self.dds_scan(i, f)
            i += 1
        #  resets dds_bd1 to default value at the end of all scan points
        self.dds_bd1.set(self.f0)

    @kernel
    def dds_scan(self, index, frequency):
        hist = [0]*int(self.repetitions)
        total = 0
        #  need to break real time between experiments due to delay from
        #  mutate dataset
        self.core.break_realtime()
        self.dds_bd1.set(frequency*MHz)
        #  turn bdd off from previous loop
        self.ttl_bdd.off()
        for i in range(int(self.repetitions)):
            #  pulse sequence for each experiment
            self.detect.pulse_ne(50*us)
            self.ttl_bdd.pulse(500*us)
            n = self.detect.bd_ne(self.t_detect)
            #  delay from gate rising in detect
            delay(10*us)
            if n >= int(self.nbins):
                n = int(self.nbins)-1
            hist[n] += 1
            total += n
        #  need this pulse to keep bdd on till next scan point
        self.ttl_bdd.on()
        #  upload data from scan point to dataset for plotting, incurs delay
        self.mutate_dataset(self.counts_hist_label, index, hist)
        self.mutate_dataset(self.counts_label, index, total/self.repetitions)
        self.mutate_dataset(self.xlabel, index, frequency)
```

Looking at ttl_bdd on the scope,

[image]

Still about 3.5 - 4 seconds between kernels.

jordens, 2016-07-09:
At NIST at least there is a massive penalty for opening files/execution due to virus scanners etc. on Windows.

whitequark, 2016-07-11:
@r-srinivas I've determined that the differences between Linux and Windows, or anything that differs in ARTIQ between Linux and Windows, are not the cause of your problem.

I could not get your test case to run intact (it immediately causes an underflow), so I am running it with one difference: `n = self.detect.bd_ne(self.t_detect)` is replaced with `n = 0`.

I'm running the test case on two VMs, with Windows 7 and Ubuntu 16.04, configured with equivalent virtualizer settings (specifically with paravirtualization turned off for the Linux VM), running stock ARTIQ from conda. In both cases I get virtually identical delays between the pulse trains, somewhere between 300 and 400ms.

As a point of reference, running the following trivial benchmark incurs delays of about 100ms between pulses, representing the baseline time taken by the compiler pipeline.

```python
from artiq.experiment import *

class Benchmark(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("loop_out")

    def run(self):
        while True:
            self.pulse()

    @kernel
    def pulse(self):
        self.core.break_realtime()
        self.loop_out.pulse(10*ms)
```

Could you also run this benchmark and report the delay? Also, can you run `python -m artiq.compiler.testbench.perf_embedding benchmark.py` (using the code above) and report its results?

> Could importing the Detection class from another file caused an additional delay?

No. The file is read only once and cached by Python (the `inspect.getsource` function takes care of that).

@jordens That might be the culprit here. Every time a kernel is compiled, ARTIQ calls the linker, which spawns an external process.

r-srinivas, 2016-07-11:
The delay seems to be about 500 - 750 ms between pulses.

[image]

I tried running the code but ran into this error
```
[artiq-2016-06-24] c:\artiq-magtrap>python -m artiq.compiler.testbench.perf_embe
dding benchmark.py
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-06-24\lib\runpy.py", line 170, in _run_modu
le_as_main
    "__main__", mod_spec)
  File "C:\Anaconda3\envs\artiq-2016-06-24\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Anaconda3\envs\artiq-2016-06-24\lib\site-packages\artiq\compiler\test
bench\perf_embedding.py", line 52, in <module>
    main()
  File "C:\Anaconda3\envs\artiq-2016-06-24\lib\site-packages\artiq\compiler\test
bench\perf_embedding.py", line 39, in main
    stitcher = embed()
  File "C:\Anaconda3\envs\artiq-2016-06-24\lib\site-packages\artiq\compiler\test
bench\perf_embedding.py", line 35, in embed
    stitcher.stitch_call(experiment.run, (experiment,), {})
  File "C:\Anaconda3\envs\artiq-2016-06-24\lib\site-packages\artiq\compiler\embe
dding.py", line 542, in stitch_call
    synthesizer = self._synthesizer(self._function_loc(function.artiq_embedded.f
unction))
AttributeError: 'function' object has no attribute 'artiq_embedded'
```

whitequark, 2016-07-11:
> I tried running the code but ran into this error

Hm, right, that was broken in 1.1. Can you set up another conda switch that has the `dev` repositories and run the same command there? You don't need to touch any hardware as perf_embedding is a software-only command.

sbourdeauducq, 2016-07-11:
Do you confirm it is fixed in the release-1 branch? We should release 1.2 soon.

whitequark, 2016-07-11:
@sbourdeauducq I was talking only about the `perf_embedding` tool.

sbourdeauducq, 2016-07-11:
Yes, do you confirm perf_embedding is fixed in release-1?

whitequark, 2016-07-11:
Hm, there was a different problem with it in `release-1` but I just fixed it.

@r-srinivas A correction: you need to run this code in `perf_embedding`, not what I initially asked you to. (`perf_embedding` expects the run method to be a kernel function).

```python
from artiq.experiment import *

class Benchmark(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("loop_out")

    @kernel
    def run(self):
        self.core.break_realtime()
        self.loop_out.pulse(10*ms)
```

This will still not work in 1.1, but it will with development packages or in upcoming 1.2.

r-srinivas, 2016-07-11:
If I just modify the file according to the changes in b5901db2655628acf95d4208e20f24e58a7800db would that be sufficient? I don't need to flash the coredevice, right?

whitequark, 2016-07-11:
Yes. You don't.

r-srinivas, 2016-07-11:
The output I get is

```
[artiq-2016-06-24] c:\artiq-magtrap>python -m artiq.compiler.testbench.perf_embe
dding benchmark.py
128 ARTIQ embedding runs: 5.03s, 39.29ms/run
         9572099 function calls (7241603 primitive calls) in 5.018 seconds

   Ordered by: internal time
   List reduced from 381 to 10 due to restriction <10>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
156288/1536    0.330    0.000    2.453    0.002 parser.py:206(rule)
   507648    0.318    0.000    0.354    0.000 parser.py:369(_restore)
558976/768    0.266    0.000    2.465    0.003 parser.py:160(rule)
569856/768    0.257    0.000    2.472    0.003 parser.py:87(outer_rule)
277888/1536    0.177    0.000    4.446    0.003 algorithm.py:36(visit)
    23296    0.177    0.000    0.405    0.000 lexer.py:238(_refill)
322560/1536    0.162    0.000    2.447    0.002 parser.py:125(rule)
84224/7808    0.129    0.000    4.135    0.001 algorithm.py:24(generic_visit)
   671872    0.122    0.000    0.122    0.000 {built-in method builtins.getattr}

    21760    0.121    0.000    0.121    0.000 {method 'match' of '_regex.Pattern
' objects}

342 ARTIQ transforms and validators runs: 5.01s, 14.65ms/run
         9829767 function calls (7912173 primitive calls) in 5.008 seconds

   Ordered by: internal time
   List reduced from 344 to 10 due to restriction <10>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
915534/2736    0.475    0.000    2.899    0.001 algorithm.py:36(visit)
300960/9576    0.382    0.000    4.299    0.000 algorithm.py:24(generic_visit)
370044/3078    0.321    0.000    4.501    0.001 algorithm.py:29(_visit_one)
  3024990    0.313    0.000    0.313    0.000 {built-in method builtins.isinstan
ce}
   374148    0.201    0.000    0.201    0.000 {built-in method builtins.hasattr}

   213066    0.179    0.000    0.179    0.000 types.py:59(find)
   839952    0.130    0.000    0.130    0.000 {built-in method builtins.getattr}

    41382    0.115    0.000    0.169    0.000 types.py:119(__init__)
    50958    0.098    0.000    0.106    0.000 ir.py:138(set_operands)
     2394    0.095    0.000    0.185    0.000 dead_code_eliminator.py:17(process
_function)

31 LLVM optimization and linking runs: 5.03s, 162.29ms/run
         1681688 function calls (1495036 primitive calls) in 5.031 seconds

   Ordered by: internal time
   List reduced from 596 to 10 due to restriction <10>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
       31    2.209    0.071    2.209    0.071 {built-in method _winapi.CreatePro
cess}
      312    1.452    0.005    1.452    0.005 {method 'acquire' of '_thread.lock
' objects}
      775    0.081    0.000    0.088    0.000 module.py:11(parse_assembly)
89033/50655    0.075    0.000    0.214    0.000 {method 'format' of 'str' object
s}
8432/3658    0.058    0.000    0.196    0.000 llvm_ir_generator.py:176(llty_of_t
ype)
       31    0.050    0.002    0.051    0.002 targets.py:267(_emit_to_memory)
       63    0.049    0.001    0.049    0.001 {built-in method nt.open}
89932/58745    0.045    0.000    0.253    0.000 _utils.py:40(__str__)
       31    0.038    0.001    0.038    0.001 passmanagers.py:77(run)
91357/21390    0.032    0.000    0.060    0.000 {built-in method builtins.hash}

```

whitequark, 2016-07-11:
@r-srinivas OK, this is definitely the AV in NIST. But that's not the whole story. Even with the AV slowdown it should give you not much more than 300ms (I estimate maybe 100ms more taken by stripping, not accounted for in perf_embedding) delay between pulses, whereas you observe 500-700ms. Something else is afoot.

Could it be that the AV also penalizes network connections?

r-srinivas, 2016-07-11:
Just to rule out the switch I connected our coredevice directly to the computer and the delays are still 500-750 ms between pulses. Is there an easy way to check whether an AV penalises network connections? I'll have a check with the IT assistance here anyway.

whitequark, 2016-07-11:
I don't have any code to do that on hand but I'll write some tomorrow.

whitequark, 2016-07-11:
BTW I've just added another commit that alters perf_embedding to drill down more into the LLVM optimization, compilation and linking. Can you also cherry-pick that and re-run `perf_embedding`?

r-srinivas, 2016-07-11:
This is what I got.

```
[artiq-2016-06-24] c:\artiq-magtrap>python -m artiq.compiler.testbench.perf_embe
dding benchmark.py
130 ARTIQ embedding runs: 5.01s, 38.55ms/run
         9721663 function calls (7354753 primitive calls) in 5.001 seconds

   Ordered by: internal time
   List reduced from 381 to 10 due to restriction <10>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
158730/1560    0.378    0.000    2.448    0.002 parser.py:206(rule)
   515580    0.318    0.000    0.353    0.000 parser.py:369(_restore)
567710/780    0.277    0.000    2.460    0.003 parser.py:160(rule)
578760/780    0.255    0.000    2.467    0.003 parser.py:87(outer_rule)
282230/1560    0.164    0.000    4.448    0.003 algorithm.py:36(visit)
    23660    0.162    0.000    0.382    0.000 lexer.py:238(_refill)
327600/1560    0.159    0.000    2.442    0.002 parser.py:125(rule)
85540/7930    0.129    0.000    4.134    0.001 algorithm.py:24(generic_visit)
   682370    0.124    0.000    0.124    0.000 {built-in method builtins.getattr}

  1161940    0.118    0.000    0.118    0.000 {built-in method builtins.isinstan
ce}

352 ARTIQ transforms and validators runs: 5.01s, 14.25ms/run
         10117187 function calls (8143523 primitive calls) in 5.014 seconds

   Ordered by: internal time
   List reduced from 344 to 10 due to restriction <10>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
942304/2816    0.483    0.000    2.915    0.001 algorithm.py:36(visit)
309760/9856    0.383    0.000    4.288    0.000 algorithm.py:24(generic_visit)
380864/3168    0.321    0.000    4.494    0.001 algorithm.py:29(_visit_one)
  3113440    0.314    0.000    0.314    0.000 {built-in method builtins.isinstan
ce}
   385088    0.205    0.000    0.205    0.000 {built-in method builtins.hasattr}

   219296    0.179    0.000    0.179    0.000 types.py:59(find)
   864512    0.148    0.000    0.148    0.000 {built-in method builtins.getattr}

    42592    0.115    0.000    0.157    0.000 types.py:119(__init__)
     2464    0.107    0.000    0.197    0.000 dead_code_eliminator.py:17(process
_function)
    52448    0.099    0.000    0.107    0.000 ir.py:138(set_operands)

138 LLVM optimizations runs: 5.02s, 36.40ms/run
         7056219 function calls (6242295 primitive calls) in 5.022 seconds

   Ordered by: internal time
   List reduced from 433 to 10 due to restriction <10>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
     3450    0.358    0.000    0.389    0.000 module.py:11(parse_assembly)
390540/221904    0.320    0.000    0.913    0.000 {method 'format' of 'str' obje
cts}
31878/16284    0.213    0.000    0.700    0.000 llvm_ir_generator.py:176(llty_of
_type)
416760/284832    0.199    0.000    1.073    0.000 _utils.py:40(__str__)
      138    0.163    0.001    0.163    0.001 passmanagers.py:77(run)
406686/95220    0.142    0.000    0.262    0.000 {built-in method builtins.hash}

  1170792    0.119    0.000    0.119    0.000 {built-in method builtins.isinstan
ce}
    47610    0.115    0.000    0.904    0.000 values.py:194(_to_string)
   175260    0.106    0.000    0.135    0.000 types.py:537(is_mono)
163944/158010    0.105    0.000    0.249    0.000 _utils.py:50(get_reference)

3138 LLVM machine code emission runs: 5.00s, 1.59ms/run
[... 43 lines cut ...]
58 Stripping debug information runs: 5.05s, 87.10ms/run
         35621 function calls in 5.051 seconds

   Ordered by: internal time
   List reduced from 116 to 10 due to restriction <10>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
       58    3.487    0.060    3.487    0.060 {built-in method _winapi.CreatePro
cess}
      580    1.366    0.002    1.366    0.002 {method 'acquire' of '_thread.lock
' objects}
      116    0.085    0.001    0.085    0.001 {built-in method nt.open}
       58    0.030    0.001    0.030    0.001 {built-in method _winapi.WaitForSi
ngleObject}
      116    0.016    0.000    0.016    0.000 {method 'close' of '_io.BufferedRa
ndom' objects}
      116    0.007    0.000    0.007    0.000 {built-in method _thread.start_new
_thread}
      116    0.005    0.000    0.005    0.000 {method 'flush' of '_io.BufferedRa
ndom' objects}
      232    0.003    0.000    0.003    0.000 {built-in method io.open}
      406    0.003    0.000    0.003    0.000 {built-in method _winapi.CloseHand
le}
       58    0.002    0.000    5.051    0.087 targets.py:183(strip)
```

whitequark, 2016-07-11:
OK, yeah, you should be able to shave at least 150ms of delay just by getting rid of the AV. The last two steps take no more than 15ms on my machine.

whitequark, 2016-07-13:
@r-srinivas This code should help diagnose some networking problems:
```python
import time
from artiq.coredevice.comm_tcp import Comm
host = 'kc705.lab.m-labs.hk'

print('new connection')
for i in range(5):
    start = time.perf_counter()
    comm = Comm(None, host)
    comm.check_ident()
    end = time.perf_counter()
    print(end - start)

print('roundtrip on existing connection')
comm = Comm(None, host)
for i in range(5):
    start = time.perf_counter()
    comm.check_ident()
    end = time.perf_counter()
    print(end - start)
```

[31 more replies]

closed 2018-02-01 by sbourdeauducq