m-labs/artiq #412: Optional hardware devices in experiment classes

dleibrandt, 2016-04-29:
We'll want to write general purpose experiment classes that can run on several different physical experimental setups.  Sometimes, the different setups will have optional hardware.  I'd like to do something like the following:

```
B_HAS_TTL13 = False

class OptionalDevice(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl12")
        if (B_HAS_TTL13):
            self.setattr_device("ttl13")
        
    @kernel
    def run(self):
        with parallel:
            self.ttl12.pulse(100*us)
            if (B_HAS_TTL13):
                self.ttl13.pulse(100*us)
```

But the compiler spits an error because `self.ttl13` does not exist.  Is there a way to do this, perhaps similar to the preprocessor directives in c++?

whitequark, 2016-04-30:
There is. You can do something like:

``` python
class OptionalDevice(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl12")
        if (B_HAS_TTL13):
            self.setattr_device("ttl13")
            self.ttl13_pulse = lambda t: self.ttl13.pulse(t)
        else:
            self.ttl13_pulse = lambda t: None

    @kernel
    def run(self):
        with parallel:
            self.ttl12.pulse(100*us)
            self.ttl13_pulse(100*us)
```

sbourdeauducq, 2016-04-30:
> self.ttl13_pulse = lambda t: self.ttl13.pulse(t)

Doesn't this need a kernel decorator?

whitequark, 2016-04-30:
Oh, yeah. `kernel(lambda t: ...)` or even better use a proper `def`.

sbourdeauducq, 2016-04-30:
I would rather suggest:

```python
class FakeTTL:
    @kernel
    def pulse(self, t):
        pass

...

if has_ttl13:
    self.ttl13 = self.get_device("ttl13")
else:
    self.ttl13 = FakeTTL()
self.kernel_invariants.add("ttl13")  # for optimization, optional
```

sbourdeauducq, 2016-05-03:
@dleibrandt Does this solve your problem?

dleibrandt, 2016-05-03:
Yup, thanks.

closed 2016-05-03 by dleibrandt