m-labs/artiq #414: Core device object inheritance error

dleibrandt, 2016-04-29:
I have two different DAC driver classes for my two types of DACs.  They both are subclasses of this class:

```
class DACcal():
    """
    Calibration info for 16 bit DACs.
    """
    
    def __init__(self, n_chan, v_min, v_max):
        self.v_min = v_min
        self.v_max = v_max
        
        self.gain = [0] * n_chan
        self.offset = [0] * n_chan

        for i in range(n_chan):
            self.cal(i, 0, v_min, 0xffff, v_max)
    
    def cal(self, i, word0, v0, word1, v1):
        self.gain[i] = (word1 - word0) / (v1 - v0);
        self.offset[i] = v0 - word0 / self.gain[i];
    
    @portable
    def voltage_to_dac_word(self, i, voltage):
        """Returns DAC word corresponding to given voltage."""
        
        word = int((voltage - self.offset[i]) * self.gain[i])
        
        if (word < 0):
            word = 0
        elif (word > 0xffff):
            word = 0xffff
        
        return word
```

I can use either DAC in separate experiments without problems.  However, if I try to use them in the same experiment I get an error.  Here is the simplest experiment in which I could get the error to occur:
```
class TestAlternatingSpiDevices(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl12")

        self.AD5370 = self.get_device("dac0")
        self.DAC8814 = self.get_device("dac1")
        
        print("build:")
        print(self.AD5370)
        print(self.AD5370.voltage_to_dac_word(0, 0))
        print(self.DAC8814)
        print(self.DAC8814.voltage_to_dac_word(0, 0))
        
        
    @kernel
    def run(self):
        print("core device:")
        print(self.AD5370)
        print(self.AD5370.voltage_to_dac_word(0, 0))
        print(self.DAC8814)
        print(self.DAC8814.voltage_to_dac_word(0, 0))
```

The calls to `voltage_to_dac_word` in build work fine, but the calls in run result in the following error:
```
root:While compiling /home/rabi/artiq-work/test_alternating_spi_devices.py
/home/rabi/artiq-work/test_alternating_spi_devices.py:25:15-25:27: error: cannot unify <instance artiq.coredevice.DAC8814.DAC8814> with <instance artiq.coredevice.AD5370.AD5370 {
		__objectid__: int(width=32)
	}> while inferring the type for self argument
        print(self.DAC8814.voltage_to_dac_word(0, 0))
              ^^^^^^^^^^^^~~~~~~~~~~~~~~~~~~~~       
/home/rabi/artiq-work/test_alternating_spi_devices.py:25:15-25:27: note: expression of type <instance artiq.coredevice.DAC8814.DAC8814 {
		__objectid__: int(width=32)
	}>
        print(self.DAC8814.voltage_to_dac_word(0, 0))
              ^^^^^^^^^^^^                           
/home/rabi/artiq-work/test_alternating_spi_devices.py:25:15-25:47: note: reference to a method 'voltage_to_dac_word(self:<instance artiq.coredevice.AD5370.AD5370>, i:int(width='a), voltage:int(width='b))->int(width='c) delay('d)'
        print(self.DAC8814.voltage_to_dac_word(0, 0))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^       
```

The problem goes away if I get rid of the `DACcal` class and put its functionality directly in the two DAC driver classes.

jordens, 2016-04-29:
@whitequark

whitequark, 2016-04-30:
Yes. There is currently no support for inheritance at all. @sbourdeauducq, which idioms we cannot live with?

sbourdeauducq, 2016-04-30:
> There is currently no support for inheritance at all.

There is some, the DDS uses it...

whitequark, 2016-04-30:
That works by accident.

sbourdeauducq, 2016-04-30:
Simply supporting ``class C: def x(); class A(C); class B(C);`` with code duplication done by the compiler sounds simple enough and would solve this problem (and most of them I guess).

dleibrandt, 2016-05-02:
If it's not too much more work, supporting multiple inheritance would be useful also.  The AD5370 driver that I wrote inherits from both @jordens 's AD5360 class and my DACcal class.

whitequark, 2016-05-02:
There's no difference to whether single or multiple inheritance is used, just what method is resolved after all.

commit 355af3e5 (2016-05-16): embedding: specialize inherited functions.

Fixes #414.
closed 2016-05-16 by whitequark