m-labs/artiq #415: Multiple SPI buses

dleibrandt, 2016-04-29:
I'm running the KC705 clock hardware.

I'd like to be able to write commands to two SPI devices (connected to different TTL lines, of course) simultaneously.  This seems to be impossible, as far as I can tell, because they currently use the same RTIO SPI bus.  Would it be possible to have three RTIO SPI buses, one for each of the RJ45 connectors on the breakout board?

(This would also eliminate the annoyance of having to call the equivalent of `ad5360.setup_bus()` before every SPI command is written, since I don't always know if another SPI device has used the bus in the meantime.)

jordens, 2016-04-29:
This is precisely what the `"spi_device": "spi0"` in the device database is for. What's the problem?

The "annoyance" of having to setup the bus every time (if you use a single bus for multiple devices) is inherent to using a single bus for multiple devices.

dleibrandt, 2016-04-29:
Whoops, I see my problem.  I was using different SPI buses, but had a coding bug that prevented simultaneous updates.  Never mind.

closed 2016-04-29 by dleibrandt