m-labs/artiq #417: Default value when no scan selected

r-srinivas, 2016-04-29:
Using 1.0 rc3 on Windows. If I have a linear scan and switch to no scan, the default value for no scan is 0. I think it would be useful if that took the default value of the linear scan instead.

sbourdeauducq, 2016-04-30:
If you want any interaction between the widgets of the different scan modes, this needs a full specification.

r-srinivas, 2016-05-05:
Sorry, what do you mean by full specification?

dhslichter, 2016-05-11:
There is currently no single "default value" for a linear scan, simply a range and a number of points to be scanned.  The problem is that for a `Scannable()` object, one can provide default behavior only for one type of scan, for example `LinearScan()`.  If in the GUI one then clicks to change the scan type, the default behavior of those other types of scans cannot be pre-set from the experiment file.  

For example, one often wants to scan around a particular frequency of interest, call it `fres`, which might come from a dataset or be an argument.  You have another argument (or dataset value) `width` in your experiment to describe the range of frequencies to scan by default.  So you would have the argument:
 
`self.setattr_argument("fscan", Scannable(default=LinearScan(fres-width/2, fres+width/2, 100))`

This comes up in the fancy scan widget.  However, in your experiment you may instead want to scan a different variable (say a pulse duration).  In this instance, you turn `fscan` from Linear Scan to No Scan.  

The desired behavior would be that one can specify a default value for the `NoScan()` object as well, which in this case would probably be `fres`.  I can think of two ways one might handle this:

1. Redesign `Scannable()`.  Add the optional parameters `noscan=value`, `lin=(start, stop, npoints)`, `rand=(start, stop, npoints, seed=0)`, and `explicit=sequence`.  This enables one to set default values for each of these other types of scan besides just the default scan type, which will be passed in as appropriate if the scan widget selects these other types of scan.  

2. Give `LinearScan` an additional optional parameter `noscan`, which consists of a desired default value if the scan is turned to No Scan.  When the `Scannable()` argument is changed from a linear scan to No Scan, this should somehow be passed to the `NoScan()` as its value.  This seems less natural to implement than #1.

dhslichter, 2016-05-23:
ping @sbourdeauducq

commit 43081b4f (2016-05-23): language,gui: support setting unselected scan default values. Closes #417
closed 2016-05-23 by sbourdeauducq